zephyrproject-rtos/zephyr #42361: OpenOCD flashing not working on cc1352r1_launchxl/cc26x2r1_launchxl

gmarull, 2022-02-01:
**Describe the bug**

`cc1352r1_launchxl` and `cc26x2r1_launchxl` boards are flashed using OpenOCD in Zephyr, however, it seems to be broken. Latest XDS110 firmware is used (3.0.0.13).

**To Reproduce**

```
west build -b cc26x2r1_launchxl samples/hello_world
west flash
```

**Expected behavior**

`west flash` works.

**Impact**

Not able to load Zephyr firmware.

**Logs and console output**

```
Open On-Chip Debugger 0.11.0+dev-00244-g7e3dbbbe2 (2021-11-18-07:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
adapter speed: 1500 kHz

Info : XDS110: connected
Info : XDS110: vid/pid = 0451/bef3
Info : XDS110: firmware version = 3.0.0.13
Info : XDS110: hardware version = 0x0023
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 1500 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: cc26x2.jrc: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : starting gdb server for cc26x2.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc26x2.cpu         cortex_m   little cc26x2.cpu         tap-disabled

Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: cc26x2.jrc: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Target not examined yet
Error: auto_probe failed
auto erase enabled
Info : XDS110: disconnected
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.2
 - Commit SHA or Version used fdc25cd44c2b693aecff96d3dfcd0ee088d48d5b

**Additional context**

#21372

gmarull, 2022-02-01:
Closing, despite the errors board seems to be flashed.

Fermi-4, 2022-10-19:
I am also seeing all these messages indicating some kind of failure but the board does flash:

`(.venv) fermi-4@fermi4-MS-7A67:~/zephyrproject/zephyr$ west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/fermi-4/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.11.0+dev-00725-gc5c47943d-dirty (2022-10-03-06:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter srst delay: 100

Info : XDS110: connected
Info : XDS110: vid/pid = 0451/bef3
Info : XDS110: firmware version = 3.0.0.15
Info : XDS110: hardware version = 0x0023
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 5500 kHz
Info : JTAG tap: cc13x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc13x2.cpu enabled
Info : [cc13x2.cpu] Cortex-M4 r0p1 processor detected
Info : [cc13x2.cpu] target has 6 breakpoints, 4 watchpoints
Info : [cc13x2.cpu] external reset detected
Info : starting gdb server for cc13x2.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc13x2.cpu         cortex_m   little cc13x2.cpu         running

Info : JTAG tap: cc13x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc13x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: [cc13x2.cpu] Could not find MEM-AP to control the core
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x1000118e msp: 0x11001ff0
auto erase enabled
wrote 360448 bytes from file /home/fermi-4/zephyrproject/zephyr/build/zephyr/zephyr.hex in 3.445470s (102.163 KiB/s)

Info : JTAG tap: cc13x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc13x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: [cc13x2.cpu] Could not find MEM-AP to control the core
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x1000118e msp: 0x11001ff0
verified 360448 bytes in 1.080734s (325.705 KiB/s)

Info : JTAG tap: cc13x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc13x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: [cc13x2.cpu] Could not find MEM-AP to control the core
shutdown command invoked
Info : XDS110: disconnected
`