zephyrproject-rtos/zephyr #42374: STM32L5: Entropy : Power Management not working due to entropy driver & stop mode

wcappelle, 2022-02-01:
**Describe the bug**
When enabling power management with stop mode(any of the three) enabled for the STM32L5 soc, the stop mode will stop all clocks in the mcu going to that lp mode. when returning from stop mode, not all clocks are being restored, like the HSI48 clock which is being used by USB & RNG hardware blocks. When this happends, and the entropy driver needs to get new samples, it enables interrupts for the block and this starts a constant entry to the ISR due to a clock error detected int he RNG hardware block.
This endless loop in the stm32_rng_isr never gets resolved.

The problem could be resolved in the power.c of the soc to re-enable the HSI48 clock (or just use MSI as a clock source if USB is not present, the power consumption will like that much more :-) ).

Next to that, a pm constraint needs to be added to the entropy driver, preventing the mcu to go to deep sleep when the entropy driver is filling the random number pool.

**To Reproduce**
enable lo p

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
no low power for the L5 and probably other families which uses the rng & stop mode (L4,...)

**Environment (please complete the following information):**
 - OS:  Linux
 -  Zephyr SDK, 13.1
 - zephyr v2.7.99-3760-g4a0e5e9958

**Additional context**
I've done the work on a local branch, but it's deviating a bit from the mainstream clock implementations & clock selections (using MSI, with own overriding of the weak power functions), but I could share at least the power constraints in the entropy driver.

wcappelle, 2022-02-01:
I think this might need first some discussion on how to handle the diversity in clocks. I'm wondering why this clock configuration is scattered into the usb driver & entropy driver for the same clock source (HSI48).
With some direction, I could try to fix this (if not too much deviating from what I need to work ofc).

erwango, 2022-02-01:
> I think this might need first some discussion on how to handle the diversity in clocks.

About this point, I've started to work on this here: https://github.com/zephyrproject-rtos/zephyr/pull/42097.
H7 only for now, but that's the direction.

wcappelle, 2022-02-01:
> > I think this might need first some discussion on how to handle the diversity in clocks.
> 
> About this point, I've started to work on this here: #42097. H7 only for now, but that's the direction.

look nice in point of view to the devicetree, however, looks like a lot of work to rework this in the clock_stm32_ll_common.
It also doesn't completely fit to a design where 1 clock mux is needed for more than 1 hw resource. For this mcu in particular there is 1 switch for 2 blocks being rng and USB:
[image]
I don't know how you can set this in your devicetree without creating another virtual clockdomain?

erwango, 2022-02-02:
> @erwango look nice in point of view to the devicetree, however, looks like a lot of work to rework this in the clock_stm32_ll_common.

Sure, this will require some driver rework, but I don't how to achieve a scalable multi clock solution otherwise.

> It also doesn't completely fit to a design where 1 clock mux is needed for more than 1 hw resource. For this mcu in particular there is 1 switch for 2 blocks being rng and USB:

Indeed, for this kind of domain, we'll need to add new dedicated nodes. But this shouldn't be quite complex. Anyway, thanks for the pointer. As this is also present on H7 devices, I'll take some time to add one and see how it goes.

erwango, 2022-02-03:
> I don't know how you can set this in your devicetree without creating another virtual clockdomain?

So I've had a try here: https://github.com/erwango/zephyr/commit/57ea427333d0979d631185931af1e58146c25167
This adds stm32h7 per_ck as a virtual clock device:
```
&perck {
	clocks = <&rcc STM32_SRC_HSI_KER CKPER_SEL(0)>;
	status = "okay";
};
```
 Works fine (even if not fully tested yet), main drawback is the cost (~220 bytes of flash)

A much simpler alternative would be to add clock configuration directly to the clock consumer end node:
```
	clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00001000>,
		 <&rcc STM32_SRC_HSI_KER CKPER_SEL(0)>,
		 <&rcc STM32_SRC_CKPER SPI123_SEL(4)>;
```
Though, this proposal has 2 main drawbacks:
- If board dts is poorly configured, 2 consumers of the same internal clock (per_ck here) could set conflicting configurations
- This adds complexity on the consumers driver implementation side that should be able to differentiate between all these clocks 

In the end the virtual clock device feels saner.

wcappelle, 2022-02-04:
> t to a design where 1 clock mux is needed for more than 1 hw resource. For this mcu in particular there is 1 switch for 2 blocks being rng and USB

I'm personally also in favor of the first approach. 
The direction of your PR looks good for me. I think you're also solving the splitoff of all the source clocks (HSI/MSI/HSE/HSI48/..), since in the current clock intialization routines all is very tightly coupled: if you have MSI as clock source, it will disable HSI or HSE or any other clock it thinks it might not use. As far as I could see you can with your changes select HSI in combination with MSI for different hardware clock domains, and both will be enabled right?
This change would also make it possible to move the USB & RNG clocks outside of the drivers,and make these a user of the clock driver, and let the clock driver initialize the clock instead of usb & rng driver based on the need.

I've seen there is a split off for the clock drivers for u5,h7, is it more the idea to create different clock driver files per series (split off from the common file )?

erwango, 2022-02-04:
Thanks for having had a look and the feedback.

> I'm personally also in favor of the first approach. 

:+1: 

> The direction of your PR looks good for me. I think you're also solving the splitoff of all the source clocks (HSI/MSI/HSE/HSI48/..), since in the current clock intialization routines all is very [...]

Yes

> This change would also make it possible to move the USB & RNG clocks outside of the drivers,and make these a user of the clock driver, and let the clock driver initialize the clock instead of usb & rng driver based on the need.

That was one of the main goals driving the change indeed.

> I've seen there is a split off for the clock drivers for u5,h7, is it more the idea to create different clock driver files per series (split off from the common file )?

U5 and H7 had their own drivers already before this change. Main reason is that they were a bit specific compared to other series (H7 much more complex and U5 due to a renaming of some LL APIs).
Though, as much as possible, I'd like to keep a common ground or driver structure for the other series. Implementing a clock_control driver is not the easiest thing, so I'd like we keep the existing "simplicity" in adding new series.
I didn't had a close look to that yet, as for now I wanted to explore at the maximum the new "concepts" on the one that's supposed to be the most complex one (H7). One idea for a refactoring would be to take more advantage of the compatibles.
This will be my next task once the H7 part is merged.

erwango, 2022-06-29:
@wcappelle Would you mind have a quick check on what is left required on this PR after the merge of #45053 and #42375 ?
Thanks

wcappelle, 2022-06-30:
@erwango thanks for the changes. Your PR seems a nice improvement
However, I didn't see dts include for the stm32L5 family :include/zephyr/dt-bindings/clock/stm32l5_clock.h
Since i'm working on an L5, this means this would need to be added I guess?

I have little time to spare atm, but when release 3.2 is finished, I plan to move to that release, and will re-analyze this.

erwango, 2022-06-30:
> However, I didn't see dts include for the stm32L5 family :include/zephyr/dt-bindings/clock/stm32l5_clock.h

Actually l5 shares the same registers as l4 on that matter, so stm32l5.dtsi includes available stm32l4_clock.h. Nothing to do on this front.

> I have little time to spare atm, but when release 3.2 is finished, I plan to move to that release, and will re-analyze this.

Great, no rush.

wcappelle, 2022-10-21:
@erwango , sorry for the delay, but I finally found some time to move to 3.2 release. 
After looking to the changed done, I thing the entropy_stm32_rng_init should be updated? It doesn't look to the clock configurations set up in the devicetree, and starts (default) clocks & clock selections:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/entropy/entropy_stm32.c#L631

erwango, 2022-12-19:
https://github.com/zephyrproject-rtos/zephyr/pull/50643 merge.
Domain clock selection remains to be implemented

erwango, 2023-02-06:
Entropy domain clock selection implemented in https://github.com/zephyrproject-rtos/zephyr/pull/53719

closed 2023-02-06 by erwango