zephyrproject-rtos/zephyr #42396: ztest: week test_main() is promoted over given testsuite's test_main() if the testsuite uses own library

nvlsianpu, 2022-02-02:
**Describe the bug**

When I build and run affected tests for, for instance  "system.settings.functional.nvs" for native_posix board, I only obtain this ouput : 
```
*** Booting Zephyr OS build v2.7.99-2256-g8aeed7eaba71  ***
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

It looks like no tests are run.
While debugging this unexpected behaviour, I found that the `test_main` function from `settings_basic_test.c` is not called but it calls the weaked one from `ztest.c` : `void __weak test_main(void)`. That is why nothing is done when I run these tests.
Settings tests seems to be the only ones implementing a zephyr_library which call the ztest framework. It could be the reason why the function `test_main` is not called but I did not dig further.

And indeed the exapmle testsuite uses own library which includes the test_main(); The function is not linked in as the week symbol is found already (it is how the linker works).

This issue is caused by introduction of the week test_main implementation: https://github.com/zephyrproject-rtos/zephyr/pull/37192

**To Reproduce**
Steps to reproduce the behavior:
`west build -b native_posix tests/subsys/settings/functional/nvs/ -p -t run`

**Expected behavior**
Correctly run all the tests compiled. For instance, it can generate this output : 

```
*** Booting Zephyr OS build v2.7.99-2256-g8aeed7eaba71  ***
Running test suite settings_test_suite
===================================================================
START - test_clear_settings
 PASS - test_clear_settings in 0.0 seconds
===================================================================
START - test_support_rtn
 PASS - test_support_rtn in 0.0 seconds
===================================================================
START - test_register_and_loading
[00:00:00.000,000] <inf> fs_nvs: 4 Sectors of 4096 bytes

[...]

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

**Impact**
While running the tests with twister, it could consider that tests are all passed but nothing is done.

**Environment (please complete the following information):**
- zephyr-rtos 3.0.0-rc1

nvlsianpu, 2022-02-02:
cc @PerMac @carlescufi

PerMac, 2022-02-02:
One o twisters weak points is that it does not verify what was in fact run on a hardware and is easily satisfy with seeing `PROJECT EXECUTION SUCCESSFUL` line. I believe we have to add a check in twister which will compare the "test suite name" which is printed in the output with the one in the source code. If we'd have this, this issue would've be caught earlier, since the "weak suite" doesn't print the matching name. I believe adding the check matching printed name vs the one in the code is a high priority for us. @gopiotr, would you be able to implement this?

gopiotr, 2022-02-02:
>  @gopiotr, would you be able to implement this?

Yes, I work on it right know and it should help solve this issue. I will make PR today or tomorrow morning.

nashif, 2022-02-03:
Before we do anything in twister, we need to fix in ztest which is where the issue is. ztest is not running any tests and reports success, it should be possible to in ztest to identify that no tests have run and report failure, which will then expose the issue as it happens.

yperess, 2022-02-03:
One thing to add to this. These issues should be resolved also by the new ztest API I've introduced. There's another check I need to add to the verification step that's currently missing (if you misspell the test suite name when declaring a new test you'll never run), but other than that, you should never run into this problem (of tests reporting as passing when they never ran) if we move to the new API.

yperess, 2022-02-03:
https://github.com/zephyrproject-rtos/zephyr/pull/42467

nashif, 2022-02-03:
> I'm working on feature, which will help us to check if printed test suite name in output is compatible with those declared in source code.

this is a known problem and we had a solution for it that worked on the test runner that preceded twister. See `Run ID` in subsys/testsuite/include/tc_util.h.

gopiotr, 2022-02-04:
> this is a known problem and we had a solution for it that worked on the test runner that preceded twister. See Run ID in subsys/testsuite/include/tc_util.h.

@nashif I do not understand your suggestion. How existence of this `Run ID` feature could currently help in proper marking test results? So far there is only possibility to add `TC_PRINT_RUNID` macro to test source code. This cause only printing of this `Run ID` (only if it is defined). And that's all. To be useful, this feature should be "handled" somehow during test verification and so far it isn't (or probably I didn't find it). Moreover currently `TC_PRINT_RUNID` is used only in two tests:
`tests/benchmarks/app_kernel/src/master.c` and `tests/benchmarks/sys_kernel/src/syskernel.c` (and to be clear it is used wrongly - `TC_RUNID` variable is not created, so `TC_PRINT_RUNID` will not print `RunID` info). 

So what is your proposition? Should I add handling of `RunID` to ztest or Twister and next add `TC_PRINT_RUNID` to every test in Zephyr project? I think that more rationale is to work with something what occur in every ztest test - test suite name printing occurs in every one. So that is why I proposed this way, but if you have other opinion, please explain it better.

gopiotr, 2022-02-09:
During works on feature with adding checking of ztest test suite name I discovered, that exactly the same problem occur in following tests:
```
tests/subsys/settings/fcb/raw
tests/subsys/settings/functional/nvs
tests/subsys/settings/functional/fcb
tests/subsys/settings/littlefs/base64
tests/subsys/settings/littlefs/raw
tests/subsys/settings/nvs/raw
```
The results of run this tests (and intercepted failures) can be found here:
https://github.com/zephyrproject-rtos/zephyr/runs/5098822186?check_suite_focus=true
https://github.com/zephyrproject-rtos/zephyr/runs/5098822546?check_suite_focus=true

gregshue, 2022-02-09:
Curiosity question.  Why doesn't the weak `test_main()` report failure?  It shouldn't ever run, correct?

yperess, 2022-02-09:
Thoughts about just migrating these tests to the new ztest API?

gregshue, 2022-02-09:
@yperess Not really.  Just related thoughts about testing in general.