zephyrproject-rtos/zephyr #42397: Direction finding nrf5340 uninitialized memory pass to callback

vladimirleontiev, 2022-02-02:
**Bug description**

During tests I notice errors in logs and unexpected mac address in CTE event handler:

```
[00699596] <err> bt_df.hci_df_prepare_connectionless_iq_report: Received PA CTE report when CTE receive disabled
[00683245] <err> bt_df.hci_df_prepare_connectionless_iq_report: Unknown handle 0x0000 for iq samples report
```

The function hci_df_prepare_connectionless_iq_report does not set per_adv_sync_to_report when error occurs.

subsys/bluetooth/host/direction.c
```c
void hci_df_prepare_connectionless_iq_report(struct net_buf *buf,
					     struct bt_df_per_adv_sync_iq_samples_report *report,
					     struct bt_le_per_adv_sync **per_adv_sync_to_report)
{
	struct bt_hci_evt_le_connectionless_iq_report *evt;
	struct bt_le_per_adv_sync *per_adv_sync;

	if (buf->len < sizeof(*evt)) {
		BT_ERR("Unexpected end of buffer");
		return;
	}

	evt = net_buf_pull_mem(buf, sizeof(*evt));

	per_adv_sync = bt_hci_get_per_adv_sync(sys_le16_to_cpu(evt->sync_handle));

	if (!per_adv_sync) {
		BT_ERR("Unknown handle 0x%04X for iq samples report",
		       sys_le16_to_cpu(evt->sync_handle));
		return;
	}

	if (!atomic_test_bit(per_adv_sync->flags, BT_PER_ADV_SYNC_CTE_ENABLED)) {
		BT_ERR("Received PA CTE report when CTE receive disabled");
		return;
	}

	if (!(per_adv_sync->cte_type & BIT(evt->cte_type))) {
		BT_DBG("CTE filtered out by cte_type: %u", evt->cte_type);
		return;
	}

	report->chan_idx = evt->chan_idx;
	report->rssi = evt->rssi;
	report->rssi_ant_id = evt->rssi_ant_id;
	report->cte_type = BIT(evt->cte_type);
	report->packet_status = evt->packet_status;
	report->slot_durations = evt->slot_durations;
	report->sample_count = evt->sample_count;
	report->sample = &evt->sample[0];

	*per_adv_sync_to_report = per_adv_sync;
}
```

In function bt_hci_le_df_connectionless_iq_report  pointer per_adv_sync is uninitialized. The hci_df_prepare_connectionless_iq_report keeps the per_adv_sync uninitialized (if sync is unknown). So, listener will get sync that points to unexpected memory address.

```c
void bt_hci_le_df_connectionless_iq_report(struct net_buf *buf)
{
	struct bt_df_per_adv_sync_iq_samples_report cte_report;
	struct bt_le_per_adv_sync *per_adv_sync;
	struct bt_le_per_adv_sync_cb *listener;

	hci_df_prepare_connectionless_iq_report(buf, &cte_report, &per_adv_sync);

	SYS_SLIST_FOR_EACH_CONTAINER(&pa_sync_cbs, listener, node) {
		if (listener->cte_report_cb) {
			listener->cte_report_cb(per_adv_sync, &cte_report);
		}
	}
}
```

**To Reproduce**

1. Create project based on `direction_finding_connectionless_rx` example
2. Replace example files with files from the archive [project_files.zip](https://github.com/zephyrproject-rtos/zephyr/files/7985165/project_files.zip)
3. Run firmware on nrf5340 dev board with many CTE transmitters (I used 50 tags)

**Expected behavior**

The function bt_hci_le_df_connectionless_iq_report expected to close unknown synchronization instead of passing random address to callback.

**Impact**

Uninitialized memory passed to callback.

Use of uninitialized pointer corrupts memory.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain ncs 1.8.0
 - Number of tags with CTE used: 50
 - Commit SHA c932ad1952970987301719d927fc0232116e908f (v2.7.0-ncs1) https://github.com/nrfconnect/sdk-zephyr

ppryga, 2022-02-08:
Function `hci_df_prepare_connectionless_iq_report` should return and error and `bt_hci_le_df_connectionless_iq_report` should not call `cte_report_cb` in such situation. Required changes are added here: https://github.com/zephyrproject-rtos/zephyr/pull/42575/files.

Nevertheless it should not happen that controller reports IQ samples for periodic advertising sync, that is unknown to Host.
@cvinayak is it something that should be fixed as part of https://github.com/zephyrproject-rtos/zephyr/issues/40317?

cvinayak, 2022-02-08:
@ppryga I had this fixed as part of the consolidated peroidic sync bug fix: https://github.com/zephyrproject-rtos/zephyr/pull/42569, but https://github.com/zephyrproject-rtos/zephyr/pull/42575 is better fix.

closed 2022-02-08 by carlescufi