zephyrproject-rtos/zephyr #42403: `crc16_ansi()` isn't CRC-16-ANSI

attie-argentum, 2022-02-02:
**Describe the bug**
It appears that the [`crc16_ansi()`](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/sys/crc.h#L136) function does not actually implement CRC-16-ANSI.
According to Wikipedia, CRC-16-IBM is the same thing as (aka) ANSI, Modbus, etc... `poly: 0x8005`, `init: 0xffff`, `reversed: yes`.
Online calculators (e.g: [crccalc.com](https://crccalc.com/)) and other implementations (e.g: [crc16-ansi](https://github.com/bd2357/crc16-ansi/blob/master/src/crc16.c#L140)) agree with each other.

Within Zephyr, this function is only used by `subsys/net/lib/dns/resolve.c` and tests (which appear to use incorrect constants)... the DNS subsys only actually uses the CRC for internal identification of a request/response, so its value is not validated.

Note also that the Modbus subsys implements its own (correct) CRC function: [`modbus_rtu_crc16()`](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/modbus/modbus_serial.c#L241).

Am I misunderstanding here? (i.e: is this function correct? misnamed?)

**To Reproduce**
Use the attached files - [crc-test.zip](https://github.com/zephyrproject-rtos/zephyr/files/7986973/crc-test.zip).

1. `unzip crc-test.zip`
2. `cd crc-test`
3. `gcc crc-test-zephyr.c -o crc-test-zephyr`
4. `gcc crc-test-other.c -o crc-test-other`
5. `./crc-test-zephyr`
6. `./crc-test-other`

**Expected behavior**
Output of `./crc-test-zephyr` and `./crc-test-other` should match.

**Impact**
I was planning to use `crc16_ansi()` in my application, but it provides incorrect values... I've re-implemented `modbus_rtu_crc16()` instead.

As noted above, this isn't actually used in the Zephyr code base in any meaningful way, but it could be used by applications. Similarly to the DNS use-case above, internal integrity checking would be mostly unaffected (though for example, existing non-volatile storage may appear to be corrupt after a firmware update)... I suspect that anyone using this function with other systems will have issues.

**Logs and console output**
```
$ git clone --depth 1 https://github.com/bd2357/crc16-ansi.git
Cloning into 'crc16-ansi'...
remote: Enumerating objects: 307, done.
remote: Counting objects: 100% (307/307), done.
remote: Compressing objects: 100% (257/257), done.
remote: Total 307 (delta 16), reused 306 (delta 16), pack-reused 0
Receiving objects: 100% (307/307), 1.79 MiB | 0 bytes/s, done.
Resolving deltas: 100% (16/16), done.
Checking connectivity... done.
$ gcc crc-test-zephyr.c -o crc-test-zephyr
$ gcc crc-test-other.c -o crc-test-other
$ ./crc-test-zephyr
[hello world] --> 0x971b
68 65 6c 6c 6f 20 77 6f 72 6c 64
$ ./crc-test-other
[hello world] --> 0xddc7
68 65 6c 6c 6f 20 77 6f 72 6c 64
```

carlescufi, 2022-02-02:
Thanks for reporting this. You are completely right, I will submit a fix.

attie-argentum, 2022-02-03:
@carlescufi - no problem, thanks for confirming and taking it on.

PR #42457 (2022-02-04): lib: os: crc: Fix the CRC16 implementation
See commit messages for details.

Fixes #42403.
commit 7f652494 (2022-02-04): lib: os: crc: Rework the crc16() implementation

As described in #42403, there was an issue with the existing crc16_ansi()
implementation, since it was not calculating the CRC-16-ANSI (aka
CRC-16-MODBUS). This is because the  existing crc16() function only
supported non-reflected input and output (and the CRC-16-ANSI requires
reflection on both) and also it did not seem to support correctly inial
seeds different from 0x0000 (and, again, the CRC-16-ANSI requires 0xffff
as an initial seed).

This commit replaces the existing crc16() with a functional pair,
crc16() and crc16_reflect(), that also  [... cut]
closed 2022-02-04 by nashif