zephyrproject-rtos/zephyr #42404: Is it possible to have arbitrary data in a zephyr device tree node?

cmorganBE, 2022-02-02:
I'd like to use the device tree to store some system level constants.

Is it possible to store and retrieve arbitrary values from the device tree?

Attempting to load these values fails to compile as build/zephyr/include/generated/devicetree_unfixed.h lacks the values for 'custom-num' or 'another-value'.

```
...
    int custom_num = DT_PROP(DT_PATH(settings), custom_num);
    printf("custom_num %d\n", custom_num);
...
```
```
...
zephyr/include/devicetree.h:81:17: error: 'DT_N_S_settings_P_custom_num' undeclared (first use in this function)
   81 | #define DT_ROOT DT_N
      |                 ^~~~
...
```

The devicetree overlay file:

```
/* SPDX-License-Identifier: Apache-2.0 */
/ {
        aliases {
                someuart-uart = &uart7;
        };

        settings {
                custom-num = < 29992 >;
                another-value = "some string";
        };
};
```

I posted on SO the same question in case someone wants to get rep points there, https://stackoverflow.com/questions/70947522/is-it-possible-to-have-arbitrary-data-in-a-zephyr-device-tree-node

mbolivar-nordic, 2022-02-02:
Hi, please read https://docs.zephyrproject.org/latest/guides/dts/bindings.html#dt-bindings

closed 2022-02-02 by mbolivar-nordic

cmorganBE, 2022-02-03:
@mbolivar-nordic this helped a ton. There weren't any build outputs that indicated the mapping wasn't possible due to missing bindings. I did read through all of this a few times through and it wasn't obvious at all that bindings were necessary.