zephyrproject-rtos/zephyr #42420: mgmt/mcumgr/lib: Async image erase command with status check

de-nordic, 2022-02-02:
**Is your feature request related to a problem? Please describe.**
Currently when issues image erase command, the application will wait for the image to be erased before responding to
the host that send the command. The erase may take some time, depending on flash size, erase command implementation and so on; because host gets response from image erase command only after it finishes, it may take long time for the host to wait for the response without really knowing when it will end - which causes problem with distinguishing from command taking long and not responding at all.

**Describe the solution you'd like**
Asynchronous erase.
After receiving image erase command (write) with additional parameter, for example "async=true", the mcumgr immediately responds with rc=0 (OK), if it can accept command, and "async" with value "doing" and then stars erase.
Then host can periodically ask application with image erase read command for current status of erase, to figure out when it ends.

**Describe alternatives you've considered**
Completely changing current image erase to async, but this would break compatibility.

**Additional context**
TBD

philips77, 2022-02-04:
> ...and "async" with value "doing" and...

Better just to return rc. The "async" field could be of boolean type.

de-nordic, 2022-02-07:
FYI @MarekPieta @carlescufi

nordicjm, 2022-08-31:
Have had a look at this and it would require a very big restructuring of the whole of mcumgr/smp in zephyr. SMP would need to run from it's own dedicated work queue, for functions like the erase function, the stream object would need to be passed in and stored so it could be used from work queue items i.e. to respond that operations have been completed, or have failed. The SMP protocol itself would also need to be changed so that it would allow asynchronous responses from server side, i.e. zephyr, to let the client know that an operation has been completed, which would necessitate new libraries for all clients.
A test WIP branch is located here https://github.com/nordicjm/zephyr/tree/mcuasyncerase which supports sending an sync erase command, getting a response if the task has started or in progress, but does not notify the client once complete.

nordicjm, 2022-09-22:
After having another play with this I do not think it is worth the effort. Putting the erase in a separate work queue works, the status can then be queried exactly 0 or 1 times, after which mcumgr will drop any incoming data and not send any output or perform any processing until the erase process has completely finished, despite the erase process running in a low priority (7) vs mcumgr (3) - I am unable to find the cause for this issue. And for this to work, every img_mgmt function would then need to have mutex/semaphore protection, which would bloat the code and slow things like writing data down.

nordicjm, 2022-09-22:
OK so the limitation with this is when using UART, because the erase process (for internal flash) is a busy-wait with interrupts disabled, meaning that any UART data arriving will be silently ignored, if using a bluetooth transport then this issue is not apparent. Because this would break the UART transport or set up different standards for the protocols I vote to NACK this idea.

de-nordic, 2022-09-22:
> OK so the limitation with this is when using UART, because the erase process (for internal flash) is a busy-wait with interrupts disabled, meaning that any UART data arriving will be silently ignored (...)

@nordicjm  Are we able to cut erase into smaller portions so that we could respond to requests?

nordicjm, 2022-09-22:
> @nordicjm Are we able to cut erase into smaller portions so that we could respond to requests?

No, nothing I do is able to resolve it, probably because the overhead of a work queue item is small but the actual erase process is what blocks for the longest time. I've tried yielding with loops, using delayed work queue items whereby it erases on sector at a time with 150ms delays between each and different variations of everything combined and it will not work via the UART. It actually makes it worse, because the shell is receiving **some** of the data sent from the host, then when the whole erase process is finished, if you send a mcumgr command over the UART, it has some data in the buffer and then things it has a corrupt request so drops the whole buffer and you need to send the command again to get it to execute.

de-nordic, 2022-09-22:
Bummer...

@nordicjm  Thanks for investing your time into researching the issue. Taking your analysis, of the problem, into account I am closing the issue, as it is  not possible to implement.

closed 2022-09-22 by de-nordic

nordicjm, 2022-09-22:
Test code (if needed in future) uploaded to https://github.com/nordicjm/zephyr/tree/mcuasyncerasetest