zephyrproject-rtos/zephyr #42431: [backport v2.7-branch] backport of #41962 failed

zephyrbot, 2022-02-02:
This issue was created automatically because the backport of #41962 to `v2.7-branch` failed.

The backport to `v2.7-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v2.7-branch v2.7-branch
# Navigate to the new working tree
cd .worktrees/backport-v2.7-branch
# Create a new branch
git switch --create backport-41962-to-v2.7-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick 731241f8d049924591cf4f46dda26ddd0782d8c7
# Push it to GitHub
git push --set-upstream origin backport-41962-to-v2.7-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v2.7-branch
```
Then, create a pull request where the `base` branch is `v2.7-branch` and the `compare`/`head` branch is `backport-41962-to-v2.7-branch`.

ycsin, 2022-02-06:
@cfriedt how should I continue with this? Cherry pick and resolve conflict?

cfriedt, 2022-02-06:
@ycsin exactly

cfriedt, 2022-04-12:
closing as the backport has been merged.

closed 2022-04-12 by cfriedt