zephyrproject-rtos/zephyr #42476: Get rid of OpenOCD thread info aliases

julien-massot, 2022-02-04:
**Is your enhancement proposal related to a problem? Please describe.**
Upstream OpenOCD Zephyr thread awareness should use
_kernel_thread_info_{offsets,num_offsets,size_t_size) symbols
and not the OpenOCD specific one.

Please see:
https://repo.or.cz/openocd.git/blob/dbbac5f11d662ab0827a997aa5d173fd62c51708:/src/rtos/zephyr.c#l372

We should modify OpenOCD Zephyr thread awareness in that way, so that we
can revert this PR #42393

julien-massot, 2022-02-04:
https://review.openocd.org/c/openocd/+/6844

carlescufi, 2022-02-04:
@SeppoTakalo could you please verify if this will have any impact on the JLink integration?

SeppoTakalo, 2022-02-07:
Yes. Looks like we are using `_kernel_openocd_offsets` in the JLink plugin.