zephyrproject-rtos/zephyr #42478: Unable to build mcuboot for b_u585i_iot02a

JiiZR, 2022-02-04:
**Describe the bug**
Building mcuboot for ST's B-U585I-IOT02A fails.

**To Reproduce**
Steps to reproduce the behavior:
1.add to \bootloader\mcuboot\boot\zephyr\dts.overlay

&flash0 {
        /*
         * For more information, see:
         * http://docs.zephyrproject.org/latest/guides/dts/index.html#flash-partitions
         */
        partitions {
                compatible = "fixed-partitions";
                #address-cells = <1>;
                #size-cells = <1>;

                boot_partition: partition@0 {
                        label = "mcuboot";
                        reg = <0x00000000 0x10000>;
                };
                slot0_partition: partition@10000 {
                        label = "image-0";
                        reg = <0x00010000 0x20000>;
                };
                slot1_partition: partition@30000 {
                        label = "image-1";
                        reg = <0x00030000 0x50000>;
                };
                scratch_partition: partition@70000 {
                        label = "image-scratch";
                        reg = <0x00070000 0x10000>;
                };
        };
};

2. \zephyrproject\bootloader\mcuboot>west build -p auto -b b_u585i_iot02a boot\zephyr

**Expected behavior**
Build should be OK.

**Impact**
Not able to proceed

**Logs and console output**

-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): C:/OWN/Work/ZEP/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/OWN/Work/ZEP/zephyrproject/bootloader/mcuboot/boot/zephyr
-- Zephyr version: 3.0.0-rc1 (C:/OWN/Work/ZEP/zephyrproject/zephyr), build: v3.0.0-rc1-247-gf86f6e002510
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: b_u585i_iot02a
-- Cache files will be written to: C:/OWN/Work/ZEP/zephyrproject/zephyr/.cache
-- Found toolchain: gnuarmemb (C:/gnu_arm_embedded)
-- Found BOARD.dts: C:/OWN/Work/ZEP/zephyrproject/zephyr/boards/arm/b_u585i_iot02a/b_u585i_iot02a.dts
-- Found devicetree overlay: C:/OWN/Work/ZEP/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
-- Found devicetree overlay: C:/OWN/Work/ZEP/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
-- Generated zephyr.dts: C:/OWN/Work/ZEP/zephyrproject/bootloader/mcuboot/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/OWN/Work/ZEP/zephyrproject/bootloader/mcuboot/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: C:/OWN/Work/ZEP/zephyrproject/bootloader/mcuboot/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: C:/OWN/Work/ZEP/zephyrproject/bootloader/mcuboot/build/zephyr/dts.cmake

warning: FLASH_MAP (defined at subsys/storage/flash_map/Kconfig:10) has direct dependencies FLASH_HAS_DRIVER_ENABLED with value n, but is currently being y-selected by the following symbols:
 - MCUBOOT_DEVICE_SETTINGS (defined at C:/OWN/Work/ZEP/zephyrproject/bootloader/mcuboot/boot/zephyr/Kconfig:656), with value y, direct dependencies y (value: y)Parsing C:/OWN/Work/ZEP/zephyrproject/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration 'C:/OWN/Work/ZEP/zephyrproject/zephyr/boards/arm/b_u585i_iot02a/b_u585i_iot02a_defconfig'
Merged configuration 'C:/OWN/Work/ZEP/zephyrproject/bootloader/mcuboot/boot/zephyr/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at C:/OWN/Work/ZEP/zephyrproject/zephyr/cmake/kconfig.cmake:272 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  C:/OWN/Work/ZEP/zephyrproject/zephyr/cmake/app/boilerplate.cmake:544 (include)
  C:/OWN/Work/ZEP/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  C:/OWN/Work/ZEP/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:44 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=C:\Python310\python.exe' '-BC:\OWN\Work\ZEP\zephyrproject\bootloader\mcuboot\build' '-SC:\OWN\Work\ZEP\zephyrproject\bootloader\mcuboot\boot\zephyr' -GNinja

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: GNU Arm Embedded Toolchain Version 10.3-2021.10
 -  commit f86f6e00251066815b0a7a7ad45047b525b3fab9 (HEAD -> main, origin/main, origin/HEAD)

erwango, 2022-02-04:
@JiiZR Thanks for raising this. Point is that we still miss stm32u5 flash driver implementation which is required to build MCUBoot.

JiiZR, 2022-02-04:
Ok, thanks for your information.

FRASTM, 2022-03-16:
@JiiZR can you try with the support of flash for stm32u5 https://github.com/zephyrproject-rtos/zephyr/pull/42538

&flash0 is already in `b_u585i_iot02a.dts`, maybe it could be deleted in the ../bootloader/mcuboot/boot/zephyr/dts.overlay

JiiZR, 2022-03-21:
Hi, here are results:

MCUboot can be compiled and if only primary partition is used then MCUboot jumps to the application blinky. But if swap is requested from second partition then panic is seen. Signing is done with imgtool 1.9.0 and command is: 
west sign -t imgtool -- --slot-size 0x20000 --version 1.2.3 --header-size 0x400 --key C:\OWN\zep_u5\zephyrproject\bootloader\mcuboot\root-rsa-2048.pem --pad --confirm --align 8

I also tried align 16 but then magic is generated wrongly to the trailer.

Prints from crash:
*** Booting Zephyr OS build zephyr-v3.0.0-1287-g08f4492d385c  ***
I: Starting bootloader
I: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
I: Swap type: test
D: erasing scratch area
D: initializing status; fa_id=3
D: writing swap_info; fa_id=3 off=0x3fd8 (0x63fd8), swap_type=0x2 image_num=0x0
D: writing swap_size; fa_id=3 off=0x3fd0 (0x63fd0)
D: writing magic; fa_id=3 off=0x3ff0 (0x63ff0)
D: erasing trailer; fa_id=1
D: initializing status; fa_id=1
D: writing swap_info; fa_id=1 off=0x1ffd8 (0x3ffd8), swap_type=0x2 image_num=0x0
D: writing swap_size; fa_id=1 off=0x1ffd0 (0x3ffd0)
D: writing magic; fa_id=1 off=0x1fff0 (0x3fff0)
D: erasing trailer; fa_id=2
D: erasing scratch area
D: writing copy_done; fa_id=1 off=0x1ffe0 (0x3ffe0)
E: panic!

JiiZR, 2022-04-08:
I took a closer look for the mcuboot code and changed this:

\zephyrproj_u5\bootloader\mcuboot\boot\bootutil\include\bootutil\bootutil_public.h

// hack
#define MCUBOOT_BOOT_MAX_ALIGN 16

#ifdef MCUBOOT_BOOT_MAX_ALIGN

....

and then signed apps to slot0 and slot1:
west sign -t imgtool -- --slot-size 0x20000 --version 4.5.6 --header-size 0x400 --key C:\OWN\Work\ZEP\zephyrproj_u5\bootloader\mcuboot\root-rsa-2048.pem --pad --confirm --align 16

And now mcuboot works fine, app from slot1 is updated to slot0 and app boots, so ST drivers are OK.

erwango, 2022-04-08:
Thanks @JiiZR for the heads up. I'm closing the ticket.

closed 2022-04-08 by erwango