zephyrproject-rtos/zephyr #42499: mec15xxevb_assy6853: boards.mec15xxevb_assy6853.i2c.i2c_pca95xx test failed.

yerabolu, 2022-02-04:
**Describe the bug**
test/boards/mec15xxevb_assy6853/i2c_api test fails on mec15xxevb_assy6853. 

**To Reproduce**
Steps to reproduce the behavior: 

- twister --hardware-map $HOME/mec1.file --device-testing -T tests/boards/mec15xxevb_assy6853/i2c_api/ -vv
- See the assertion failure in logs. 

**Logs and console output**
```
*** Booting Zephyr OS build v3.0.0-rc1-233-gf5f61ac605e1  ***
Running test suite i2c_test
===================================================================
START - test_i2c_pca95xx
E: Write error: -5
Assertion failed at WEST_TOPDIR/zephyr/tests/boards/mec15xxevb_assy6853/i2c_api/src/main.c:48: test_i2c_pca95xx: (ret == 0 is false)
Failed to write data to i2c device
FAIL - test_i2c_pca95xx in 0.18 seconds
===================================================================
Test suite i2c_test failed.
===================================================================
PROJECT EXECUTION FAILED
```
**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain (e.g Zephyr SDK, ...): SDK 13.2
 - Commit SHA: f5f61ac605e1

rljordan, 2022-02-08:
@enjiamai  Please work on this and collaborate with @jvasanth1

enjiamai, 2022-02-09:
Hi @rljordan , ok, got it.

yerabolu, 2022-02-09:
@enjiamai : Test is not failing on mec15xxevb_assy6853. Closing this issue as it is not reproducible.

closed 2022-02-09 by yerabolu

enjiamai, 2022-02-09:
Agree, I cannot reproduce it in specified commit and latest one. Let's watch if it comes out again.