zephyrproject-rtos/zephyr #42518: Bluetooth Ext Adv:Sync: While multipler advertiser are working, sync terminates repeatedly

saleh-unikie, 2022-02-07:
**Describe the bug**
Using the https://github.com/zephyrproject-rtos/zephyr/pull/41091 (force-pushed on 2 Feb), I tested 10 simultaneous advertiser (AoA tag) with one scanner (AoA locator). The source codes are based on "connection-less direction finding" examples, by adding simultaneous tag support to scanner code. The advertiser are working with extended adv interval = 80ms, legacy adv interval = 150ms. All devices are nRF52833DK board.

**Successful Scenario:** If at receiver side, **"skip" parameter is a non-zero value**, then "synchronization" is made successfully and the probability of losing the sync is low.
```
struct bt_le_per_adv_sync_param param;
param.skip = 1;
param.timeout = 2000;
...
bt_le_per_adv_sync_create(&param, synch);
```
The output log is something like this:
```
try to sync tag1
tag1 synced
try to sync tag2
tag2 synced
try to sync tag3
tag3 synced
try to sync tag4
tag4 synced
try to sync tag5
tag5 synced
try to sync tag6
tag6 synced
try to sync tag7
tag7 synced
try to sync tag8
tag8 synced
try to sync tag9
tag9 synced
try to sync tag10
tag10 synced

(after 30minutes no sync lost was seen)
```

**Failure Scenario:** But if, **"skip" parameter is set to zero**, at the beginning of the program, syncs are made normally like above condition, but **after a while it start to lose the sync** with tags, and it usually must **retries more than once to make the sync again**. in most retries, it terminate eventually exactly after calling the ``bt_le_per_adv_sync_create`` function.
```
struct bt_le_per_adv_sync_param param;
param.skip = 0;
param.timeout = 2000;
...
bt_le_per_adv_sync_create(&param, synch);
```
The output log could be:
```
try to sync tag1
tag1 synced
try to sync tag2
tag2 synced
try to sync tag3
tag3 synced
try to sync tag4
tag4 synced
try to sync tag5
tag5 synced
try to sync tag6
tag6 synced
try to sync tag7
tag7 synced
try to sync tag8
tag8 synced
try to sync tag9
tag9 synced
try to sync tag10
tag10 synced

tag3 sync lost
try to sync tag3

tag3 sync lost
try to sync tag3

tag3 sync lost
try to sync tag3

tag3 sync lost
try to sync tag3
tag3 synced

(the above retries occur for other tags too and it is very common to lose the sync, 
which makes the whole system functionality unstable) 
```

In both cases timeout value is same. I supposed that "skip" parameter can not affect on sync termination as I asked this question in nordic devzone, but seems it is not.
https://devzone.nordicsemi.com/f/nordic-q-a/84232/more-information-about-skip-and-timeout-parameters-of-ble-5-advertising

**To Reproduce**
Steps to reproduce the behavior:
1. use "connection-less direction finding" examples
2. Add simultaneous tag support to the scanner source code 
2. west build -b nrf52833dk_nrf52833
3. west flash
4. See error

**Expected behavior**
The termination is only dependent on "timoute" parameter, changing "skip" should not change the behavior of losing sync between scanner and advertiser.

**Impact**
System is not stable when skip=0 and multiple advertiser try to sync with the scanner.

**Logs and console output**
as you can see in the bug description (above)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/pull/41091/commits/9749514d83887d1dedde9966a0dc2a5d401aca97

**Additional context**
In addition to the above problem, sometime this assertion failure happens too
```
ASSERTION FAIL [next] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:649
```

cvinayak, 2022-02-09:
Could you let me know if all the 10 advertisers where powered up at the same time? (do all the periodic advertising interval overlap on every interval?) In such cases, 6 intervals to synchronized with a periodic advertiser may be facing issues, could you take air trace using a sniffer to confirm any errors due to collisions of these 10 periodic advertising?

> The advertiser are working with extended adv interval = 80ms, legacy adv interval = 150ms.

Do you have 20 advertisements in total from 10 nRF52833 DKs, one Extended and one Legacy advertising instances on each of the 10 DKs? What about the periodic advertising, what are the used parameters?

> 2. Add simultaneous tag support to the scanner source code

It would be best if a draft PR with samples to reproduce the issue be made available for debugging the issue.

cvinayak, 2022-02-10:
@saleh-unikie , thanks for the reply, I will try to reproduce at my end today and revert back with my findings. (Most of the consolidated commits have now been merged upstream, pending a new issue https://github.com/zephyrproject-rtos/zephyr/pull/42664). I will be used the latest upstream with your copy of the sample.

cvinayak, 2022-02-10:
@saleh-unikie could you please cherry-pick these two PRs (over latest upstream bf450552b85b23b6eaf563f314e8237c1d8cc613) and let me know if it helps with the assertion, #42664 and #42678.

Also, use these values (finger in the air, based on my experience that if Rx events are not processed fast enough by Host, then Controller will generate sync failed to be established if the Rx buffers have been used up by the other syncs that are receiving reports):
```
CONFIG_BT_BUF_EVT_RX_COUNT=15
CONFIG_BT_CTLR_RX_BUFFERS=10
```

saleh-unikie, 2022-02-10:
Thanks @cvinayak, if there was another changes let me know, then I will test them as soon as possible.

Well, I switched to the latest commit of main upstream and then those two commit was cherry picked. 
```
$ git log --oneline
bb85fe53f1 (HEAD -> test_sync_issue_42518_t1) Bluetooth: Controller: Fix to use non-discardable buffer
a9771c352e Bluetooth: Controller: Fix short prepare when many enqueued in pipeline
b68bdf20cf (origin/main, origin/HEAD, main) lib/os: fdtable: add locking to posix api
1f22a2bc3c net: tcp: Fix possible deadlock in tcp_conn_unref()
...
```

I ran [receiver code](https://github.com/saleh-unikie/direction_finding_connectionless_rx_multiple/commit/5699a047145663c6486e4b2061bfd80d132035e0) while 10 transmitter were advertising (randomly turned on) for two rounds, each time letting it go for ~30 minutes. I couldn't see any assertion fail but after a while sync started to terminate and didn't make again. Here you can see the log:
```
*** Booting Zephyr OS build v3.0.0-rc2-35-gbb85fe53f1b1  ***
Starting Connectionless Locator Demo
Bluetooth initialization...success
C       Creating Sync:[DEVICE]: 01:DA:D0:4B:79:49 (random) ...success.
S       PER_ADV_SYNC[0]: [DEVICE]: 01:DA:D0:4B:79:49 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
C       Creating Sync:[DEVICE]: 26:A0:DF:36:73:3D (random) ...success.                                    
S       PER_ADV_SYNC[1]: [DEVICE]: 26:A0:DF:36:73:3D (random) synced, Interval 0x0030 (60 ms), PHY LE 2M  
C       Creating Sync:[DEVICE]: 0A:82:4D:7C:70:37 (random) ...success.                                    
S       PER_ADV_SYNC[2]: [DEVICE]: 0A:82:4D:7C:70:37 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M  
C       Creating Sync:[DEVICE]: 3F:C4:3F:A1:E5:53 (random) ...success.                                    
S       PER_ADV_SYNC[3]: [DEVICE]: 3F:C4:3F:A1:E5:53 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M  
C       Creating Sync:[DEVICE]: 0B:63:98:30:4C:77 (random) ...success.                                    
S       PER_ADV_SYNC[4]: [DEVICE]: 0B:63:98:30:4C:77 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M  
C       Creating Sync:[DEVICE]: 3F:87:1B:06:DA:13 (random) ...success.                                    
T       PER_ADV_SYNC[5]: [DEVICE]: 3F:87:1B:06:DA:13 (random) sync terminated                             
Deleting Periodic Advertising Sync...success                                                              
C       Creating Sync:[DEVICE]: 32:8B:FA:5F:CC:48 (random) ...success.                                    
T       PER_ADV_SYNC[5]: [DEVICE]: 32:8B:FA:5F:CC:48 (random) sync terminated                             
Deleting Periodic Advertising Sync...success                                                              
C       Creating Sync:[DEVICE]: 3F:87:1B:06:DA:13 (random) ...success.                                    
T       PER_ADV_SYNC[5]: [DEVICE]: 3F:87:1B:06:DA:13 (random) sync terminated                             
Deleting Periodic Advertising Sync...success                                                              
C       Creating Sync:[DEVICE]: 24:6B:75:54:67:02 (random) ...success.                                    
S       PER_ADV_SYNC[5]: [DEVICE]: 24:6B:75:54:67:02 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M  
C       Creating Sync:[DEVICE]: 3F:87:1B:06:DA:13 (random) ...success.                                    
T       PER_ADV_SYNC[6]: [DEVICE]: 3F:87:1B:06:DA:13 (random) sync terminated                             
Deleting Periodic Advertising Sync...success                                                              
C       Creating Sync:[DEVICE]: 2F:73:D8:80:7C:2C (random) ...success.                                    
S       PER_ADV_SYNC[6]: [DEVICE]: 2F:73:D8:80:7C:2C (random) synced, Interval 0x0030 (60 ms), PHY LE 2M  
C       Creating Sync:[DEVICE]: 3F:87:1B:06:DA:13 (random) ...success.                                    
T       PER_ADV_SYNC[7]: [DEVICE]: 3F:87:1B:06:DA:13 (random) sync terminated                             
Deleting Periodic Advertising Sync...success                                                              
C       Creating Sync:[DEVICE]: 3F:87:1B:06:DA:13 (random) ...success.                                    
S       PER_ADV_SYNC[7]: [DEVICE]: 3F:87:1B:06:DA:13 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M  
C       Creating Sync:[DEVICE]: 36:61:42:F1:BB:53 (random) ...success.                                    
S       PER_ADV_SYNC[8]: [DEVICE]: 36:61:42:F1:BB:53 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M  
C       Creating Sync:[DEVICE]: 32:8B:FA:5F:CC:48 (random) ...success.                                    
S       PER_ADV_SYNC[9]: [DEVICE]: 32:8B:FA:5F:CC:48 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M  
T       PER_ADV_SYNC[7]: [DEVICE]: 3F:87:1B:06:DA:13 (random) sync terminated                             
T       PER_ADV_SYNC[6]: [DEVICE]: 2F:73:D8:80:7C:2C (random) sync terminated                             
T       PER_ADV_SYNC[4]: [DEVICE]: 0B:63:98:30:4C:77 (random) sync terminated                             
T       PER_ADV_SYNC[9]: [DEVICE]: 32:8B:FA:5F:CC:48 (random) sync terminated                             
T       PER_ADV_SYNC[0]: [DEVICE]: 01:DA:D0:4B:79:49 (random) sync terminated                             
T       PER_ADV_SYNC[8]: [DEVICE]: 36:61:42:F1:BB:53 (random) sync terminated
```
log files: 
[... cut ...]
XC      Creating sync timed out                                                                           
Deleting Periodic Advertising Sync...success                                                              
....................T   PER_ADV_SYNC[0]: [DEVICE]: 34:E6:17:0E:68:E8 (random) sync terminated             
..............ASSERTION FAIL [e] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_sy8
```

Another test is ongoing and after ~20min, I havn't seen the failure again. I will repeat it and if failure was seen, I will report it here.

cvinayak, 2022-02-11:
Two things:
1. Change the print message so that it is unique, https://github.com/saleh-unikie/direction_finding_connectionless_rx_multiple/commit/5782285efb32fcd6fc8800cf43ed7dc8183631b7#r66464587
2. Could try the Rx sample with CTE samples disabled. (Lets try to narrow down as to whether periodic sync implementation is stable).

saleh-unikie, 2022-02-11:
I made some changes in print logs to make sure all of them are unique and the output is more clear. The changes are pushed to repo.
Also I commented out these lines at receiver side, NOT enabling the CTE:
```
/*
err = enable_cte_rx(synch_info.synch_ptr);
if (err != 0) {
    printk("XC\tenable_cte_rx failed\n", err);
    delete_sync(synch_info.synch_ptr);
    scan_disable();
   continue;
}
*/
```
The devices are working and when a failure is seen, I will report it here (fortunately/unfortunately it doesn't happen too much)

cvinayak, 2022-02-11:
@saleh-unikie This is the modified sample I have been testing under simulation, and the one hour simulation log of 10 periodic advertisers to which the periodic sync sample makes upto 16 synchronizations (as advertiser change addresses, some repeat overlapping sync are established): https://github.com/zephyrproject-rtos/zephyr/files/8046922/periodic_sync_multiple_one_hour.log

There are few sync terminates that I have not analyzed yet, but there are no stall or assertions under simulation.

saleh-unikie, 2022-02-11:
Thanks @cvinayak 

I had similar results, after disabling reception of CTE. Here you can find the results:

**Test Condition:**
Receiver
Source Code: [direction_finding_connectionless_rx_multiple](https://github.com/saleh-unikie/direction_finding_connectionless_rx_multiple)
Zephyr: https://github.com/zephyrproject-rtos/zephyr/pull/42664 and https://github.com/zephyrproject-rtos/zephyr/pull/42678 on top of b68bdf20cf

Transmitter:
Source code: [direction_finding_connectionless_tx](https://github.com/saleh-unikie/direction_finding_connectionless_tx)
Number of transmitters: 10

**Results:**
Scenario A) CTE reception is NOT enabled. Test duration: 1h and 20min
```
*** Booting Zephyr OS build v3.0.0-rc2-35-gbb85fe53f1b1  ***
Starting Connectionless Locator Demo
Bluetooth initialization...success
C       Creating Sync:[DEVICE]: 25:10:38:9F:73:BA (random) ...success.
S       PER_ADV_SYNC[0]: [DEVICE]: 25:10:38:9F:73:BA (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
C       Creating Sync:[DEVICE]: 27:83:C5:5E:AF:FB (random) ...success.
S       PER_ADV_SYNC[1]: [DEVICE]: 27:83:C5:5E:AF:FB (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
C       Creating Sync:[DEVICE]: 11:84:5F:38:06:4C (random) ...success.
S       PER_ADV_SYNC[2]: [DEVICE]: 11:84:5F:38:06:4C (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
C       Creating Sync:[DEVICE]: 25:08:3C:63:47:08 (random) ...success.
S       PER_ADV_SYNC[3]: [DEVICE]: 25:08:3C:63:47:08 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
C       Creating Sync:[DEVICE]: 30:6E:A9:78:50:C1 (random) ...success.
S       PER_ADV_SYNC[4]: [DEVICE]: 30:6E:A9:78:50:C1 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
C       Creating Sync:[DEVICE]: 00:8B:2C:77:84:B9 (random) ...success.
S       PER_ADV_SYNC[5]: [DEVICE]: 00:8B:2C:77:84:B9 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
C       Creating Sync:[DEVICE]: 0D:11:A7:7B:D8:D3 (random) ...success.
S       PER_ADV_SYNC[6]: [DEVICE]: 0D:11:A7:7B:D8:D3 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
C       Creating Sync:[DEVICE]: 07:F3:51:6F:C3:69 (random) ...success.
S       PER_ADV_SYNC[7]: [DEVICE]: 07:F3:51:6F:C3:69 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
C       Creating Sync:[DEVICE]: 19:D6:38:93:F9:7F (random) ...success.
S       PER_ADV_SYNC[8]: [DEVICE]: 19:D6:38:93:F9:7F (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
C       Creating Sync:[DEVICE]: 1A:FE:BE:D8:46:2A (random) ...success.
S       PER_ADV_SYNC[9]: [DEVICE]: 1A:FE:BE:D8:46:2A (random) synced, Interval 0x0030 (60 ms), PHY LE 2M
....................
....................
....................
....................
....................
....................
....................
....................
....................
....................
....................
....................
....................
.............T  PER_ADV_SYNC[6]: [DEVICE]: 0D:11:A7:7B:D8:D3 (random) sync terminated                                                  
.......                                                                                                                                
........C       Creating Sync:[DEVICE]: 0D:11:A7:7B:D8:D3 (random) ...success.                                                         
S       PER_ADV_SYNC[6]: [DEVICE]: 0D:11:A7:7B:D8:D3 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M                               
....................
............
```
Obviously it is stable and only after 65min one of the sync has terminated which is established again after ~4min. (each ``.`` equal to 15second)

Scenario B) CTE reception is enabled. Test duration: 14min
```
*** Booting Zephyr OS build v3.0.0-rc2-41-g33c3d3db7cfb  ***                                                                           
Starting Connectionless Locator Demo                                                                                                   
Bluetooth initialization...success                                                                                                     
C       Creating Sync:[DEVICE]: 27:83:C5:5E:AF:FB (random) ...success.                                                                 
S       PER_ADV_SYNC[0]: [DEVICE]: 27:83:C5:5E:AF:FB (random) synced, Interval 0x0030 (60 ms), PHY LE 2M                               
C       Creating Sync:[DEVICE]: 11:84:5F:38:06:4C (random) ...success.                                                                 
S       PER_ADV_SYNC[1]: [DEVICE]: 11:84:5F:38:06:4C (random) synced, Interval 0x0030 (60 ms), PHY LE 2M                               
[... 14 lines cut ...]
T       PER_ADV_SYNC[0]: [DEVICE]: 27:83:C5:5E:AF:FB (random) sync terminated                                                          
.C      Creating Sync:[DEVICE]: 19:D6:38:93:F9:7F (random) ...success.                                                                 
S       PER_ADV_SYNC[0]: [DEVICE]: 19:D6:38:93:F9:7F (random) synced, Interval 0x0030 (60 ms), PHY LE 2M                               
C       Creating Sync:[DEVICE]: 27:83:C5:5E:AF:FB (random) ...success.                                                                 
T       PER_ADV_SYNC[6]: [DEVICE]: 0D:11:A7:7B:D8:D3 (random) sync terminated                                                          
T       PER_ADV_SYNC[9]: [DEVICE]: 27:83:C5:5E:AF:FB (random) sync terminated                                                          
[... cut ...]
S       PER_ADV_SYNC[9]: [DEVICE]: 30:6E:A9:78:50:C1 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M                               
.T      PER_ADV_SYNC[7]: [DEVICE]: 0D:11:A7:7B:D8:D3 (random) sync terminated 
(termination and resync continues)
```
The full log is attached: [8_cte.txt](https://github.com/zephyrproject-rtos/zephyr/files/8047406/8_cte.txt)

------------------------------------------
Well, obviously enabling the CTE reception affects on the syncs, what could be the next step?

cvinayak, 2022-02-11:
@saleh-unikie Thanks for your quick response. 

1. Could you please confirm that you are using the vanilla upstream/main without #42678? If using, does it make a difference?
2. In Scenario B, does it not stall (i.e. even after `sync terminated` it will continue to discover extended advertisers, and re-establish synchronization) ?

I will switch my focus towards review of code for possible bugs when periodic sync has CTE reporting enabled, will post here if I have some thing.

saleh-unikie, 2022-02-11:
> Could you please confirm that you are using the vanilla upstream/main

To ensure the results are true, I did the test again. One time with the latest upstream/main (5094a6e08c) and one time with https://github.com/zephyrproject-rtos/zephyr/pull/42678. In both cases, **CTE reception was disabled**. The result was similar, stable in both after running for ~1h. Here you can see the logs:
[9_cte_disabled_42678.txt](https://github.com/zephyrproject-rtos/zephyr/files/8048373/9_cte_disabled_42678.txt)
[9_cte_disabled_main.txt](https://github.com/zephyrproject-rtos/zephyr/files/8048375/9_cte_disabled_main.txt)

> In Scenario B, does it not stall (i.e. even after sync terminated it will continue to discover extended advertisers, and re-establish synchronization) ?

Yes it didn't stall at any point. The ``termination->creat sync->synced`` loop was ongoing. Anyway, I will try to execute this scenario for several times to ensure the behavior is same in all rounds. I'll update here if there was any new findings.

cvinayak, 2022-02-11:
Thank you, once again.

We are going in the right direction. (I believe the assertion could still happen, as I have done nothing to fix it, if it is a bug, it will recur). I have switch my focus to reviewing implementation when CTE enabled.

(I wish I had 11 nRF52833 DKs)

saleh-unikie, 2022-02-11:
Thanks for your time!

I tried my sample code several times and these "assertion failures" were seen:

Test Condition:
Receiver
Source Code: [direction_finding_connectionless_rx_multiple](https://github.com/saleh-unikie/direction_finding_connectionless_rx_multiple)
Zephyr: [64c2633c90](https://github.com/zephyrproject-rtos/zephyr/pull/42678) on top of 5094a6e08c

Transmitter:
Source code: [direction_finding_connectionless_tx](https://github.com/saleh-unikie/direction_finding_connectionless_tx)
Number of transmitters: 10

Failures:
```
C       Creating Sync:[DEVICE]: 27:83:C5:5E:AF:FB (random) ...success.                                                                 
S       PER_ADV_SYNC[7]: [DEVICE]: 27:83:C5:5E:AF:FB (random) synced, Interval 0x0030 (60 ms), PHY LE 2M                               
ASSERTION FAIL [(ret == 0) || (ret == 2)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:728
```
Full log: [10_cte_enabled_42678.txt](https://github.com/zephyrproject-rtos/zephyr/files/8049789/10_cte_enabled_42678.txt)

After the above log, I added a print line before when it tries to make enable reception of CTE.

```
XC      Creating sync timed out                                                                                                        
Deleting Periodic Advertising Sync...success                                                                                           
ASSERTION FAIL [e] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_sync.c:938
```
Full log: [12_cte_enabled_42678.txt](https://github.com/zephyrproject-rtos/zephyr/files/8049797/12_cte_enabled_42678.txt)

```
............T   PER_ADV_SYNC[7]: [DEVICE]: 0D:11:A7:7B:D8:D3 (random) sync terminated                                                  
........                                                                                                                               
ASSERTION FAIL [e] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_sync.c:938 
```
Full log: [13_cte_enabled_42678.txt](https://github.com/zephyrproject-rtos/zephyr/files/8049802/13_cte_enabled_42678.txt)

```
S       PER_ADV_SYNC[6]: [DEVICE]: 30:6E:A9:78:50:C1 (random) synced, Interval 0x0030 (60 ms), PHY LE 2M                               
E       Enabling reception of CTE                                                                                                      
...ASSEASSERTION FAIL [err >= 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:578
```
Full log: [14_cte_enabled_42678.txt](https://github.com/zephyrproject-rtos/zephyr/files/8049807/14_cte_enabled_42678.txt)

```
S       PER_ADV_SYNC[9]: [DEVICE]: 19:D6:38:93:F9:7F (random) synced, Interval 0x0030 (60 ms), PHY LE 2M                               
E       Enabling reception of CTE                                                                                                      
ASSERTION FAIL [next] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:653
```
Full log: [15_cte_enabled_42678.txt](https://github.com/zephyrproject-rtos/zephyr/files/8049846/15_cte_enabled_42678.txt)

cvinayak, 2022-02-12:
@saleh-unikie Based on code inspection, not sure if this is required, @ppryga can confirm whether these changes would help: https://github.com/zephyrproject-rtos/zephyr/pull/42757

ppryga, 2022-02-15:
After some time of running with 4DKs TX and single RX (based on code from @saleh-unikie repo) I start to observe this: 
[image]

Then RX starts to lose synchronization. I'm have to check it with DF functionality disabled. I'll do that tomorrow.
[unexpected-LE_BIS-while-per-adv-sync.zip](https://github.com/zephyrproject-rtos/zephyr/files/8069691/unexpected-LE_BIS-while-per-adv-sync.zip)

ppryga, 2022-02-18:
UPDATE! It was false alarm with LE-BIS. I've just run a sniffer in the same environment with DKs exercising periodic advertising and still observe LE-BIS PDUs. So there is no issues in regard of periodic advertising.

@saleh-unikie could you check if the issue is still available with code in current `main` branch?

carlescufi, 2022-03-02:
@saleh-unikie we will address this as soon as possible.

cvinayak, 2022-03-22:
@saleh-unikie and @jakkra after some delays, I now have 10 nRF52833 DKs.

Please confirm that I need to use these for debugging:
[direction_finding_connectionless_rx_multiple](https://github.com/saleh-unikie/direction_finding_connectionless_rx_multiple)
[direction_finding_connectionless_tx](https://github.com/saleh-unikie/direction_finding_connectionless_tx)

saleh-unikie, 2022-03-22:
> @saleh-unikie and @jakkra after some delays, I now have 10 nRF52833 DKs.
>
> Please confirm that I need to use these for debugging: [...]

Good news!
I've used these two projects for generating the above logs. Let me know if there was any problem or any help needed. Good luck!

cvinayak, 2022-03-24:
@saleh-unikie During code inspection, something that could be related to the ASSERTION FAIL is related to fixes in this PR: https://github.com/zephyrproject-rtos/zephyr/pull/44183. I will do more testing tomorrow to confirm if this PR fixes the assertions reported by you here: https://github.com/zephyrproject-rtos/zephyr/issues/42518#issuecomment-1044529868

Could please try your samples with ZEPHYR_BASE set to the changes in the PR #44183

saleh-unikie, 2022-03-24:
Sure, I will do it tomorrow and will report here.

saleh-unikie, 2022-03-25:
@cvinayak I've tested https://github.com/zephyrproject-rtos/zephyr/pull/44183 and the reported problems are not seen any more. Thanks a lot! 🙏

Anyway, there is another problem to receive IQ samples properly which has seen also at main repository of Zephyr (But I've not reported yet). So I think these are completely different and we can consider it as solved.

cvinayak, 2022-03-25:
@saleh-unikie Thank you and appreciate the quick responses in helping me slowly resolve a few memory leaks and implementation defects.

I have noticed that if you enable `CONFIG_LOG=y` in your rx multiple sample, the stack logs error preparing the CTE samples. We are looking in to possible uninitialized values in host. Will post PRs after further investigations.

Please close this PR,  you can re-open or create new issue if you discover issue being not resolved.

closed 2022-03-25 by saleh-unikie