zephyrproject-rtos/zephyr #42524: KSZ8863 SPI driver

paveldedourek-eaton, 2022-02-07:
Wrong implementation of SPI protocol:

### SPI write Command
https://github.com/zephyrproject-rtos/zephyr/blob/a775606ae95559daca40b65ba68509712fa5be65/drivers/ethernet/dsa_ksz8863.h#L11

https://github.com/zephyrproject-rtos/zephyr/blob/a775606ae95559daca40b65ba68509712fa5be65/drivers/ethernet/dsa_ksz8xxx.c#L60

[image]

Wrong bit are used in SPI write command.

### SPI read command
https://github.com/zephyrproject-rtos/zephyr/blob/a775606ae95559daca40b65ba68509712fa5be65/drivers/ethernet/dsa_ksz8863.h#L12

https://github.com/zephyrproject-rtos/zephyr/blob/a775606ae95559daca40b65ba68509712fa5be65/drivers/ethernet/dsa_ksz8xxx.c#L92

[image]
Wrong bit are used in SPI read command.

carlescufi, 2022-02-07:
@paveldedourek-eaton since you've analyzed the issue, is there a chance you can send a Pull Request with the fix? Thanks!

carlescufi, 2022-02-08:
> @carlescufi I can try, is there some procedure to follow for bugfix and PR?

You just need to open a Pull Request with the fix and then we can add @arvinf, the original author, to review it.

arvinf, 2022-02-08:
Please also tag @lmajewski. He is the original author, including the SPI support. I refactored the driver to add support for a variant.

lmajewski, 2022-02-09:
@paveldedourek-eaton @arvinf  The part for KSZ8794 is working - at least the configuration (and communication) via SPI for the `ip_k66f` board. Recently, I've added several fixes (now in upstream/main) which were correcting some errors after adding the KSZ8863 variant.

paveldedourek-eaton, 2022-02-21:
I made working I2C communication between KSZ8863 and my Nucleo-F429ZI. 
I am just struggling with correct assignment of ports: eth0, lan1, lan2... when this will be done I will concentrate to prepare Pull Request.

paveldedourek-eaton, 2022-03-16:
Issue is solved in following PR: https://github.com/zephyrproject-rtos/zephyr/pull/43069
Waiting for approvals .

closed 2022-05-30 by github-actions[bot]