zephyrproject-rtos/zephyr #42534: BLE Testing functions do not work properly

piotrParf, 2022-02-07:
I am using Nordic nRF5340DK with Zephyr 2.7.0 LTS SDK. I am using built in j-link and SES(Segger Embedded Studio) to debug. I also use emptyapp from nordic on the application core and example HCI uart from /samples/bluetooth/hci_uart and set CONFIG_BT_CTLR_DTM_HCI=y in prj.conf

I am using nrf testing applications like RSSI Viewer and DTM test demos on Windows and code for nRF5340DK is being build in linux Debian enviroment. 

**The issue:**
When I use HCI command LE Transmitter Test the test packets stop transmitting after certain time or packet count(around ~4000).
I cant find any workaround to this issue, yet it is needed for Bluetooth hardware testing. 
The DTM sample app from nRF Connect SDK works as expected but they are not compatible with Bluetooth stack.

**To Reproduce**
Build an Empty Application Core from Nordic, to give control for network core, from their repository. 
https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/nrf5340/empty_app_core/README.html

Build an HCI-UART application with setting:  **CONFIG_BT_CTLR_DTM_HCI=y** 

1.Run SES and load HCI-UART example application to the network core,
2.load empty application core to the network core,
3.connect the UART of nrf5340 to rtt-uart output of NRF5340DK to proceed with attaching HCI-BLE to linux
4. On linux(debian in my case) run as su : _btattach -B /dev/ttyACM1 -S 1000000 -P h4 &_
5. now open the RSSI Viewer application on another Nordic developement kit and start it to watch the BLE radio spectrum.
6. now on the linux run the hci tool and send command: _hcitool -i hci1 cmd 0x08 0x1e 0x02 0x20 0x00 0x00_ (HCI_LE_Transmitter_Test 
[v2] from specification BLE 5.2 Vol 4, Part E) It should start sending packets with PRBS9 at channel 2 with length of 32 bytes, until issuance of the command hcitool -i hci1 cmd 0x08 0x1f (LE_Test_End)

**Expected behavior**
The Bluetooth stack should transmit test packets continuously until receiving LE_Test_End command from HCI interface, yet it spontaneously stops transmitting sooner than the command LE_Test_End.

**Additional info**
The same issue happens when in network core app directly is being called ll_test_tx() function from subsys/bluetooth/controller/ll_sw/nordic/lll/lll_test.c - the packets suddenly stop transmitting after some time.

cvinayak, 2022-02-09:
Since you are using a linux distribution, ensure that Bluetooth Daemon is disabled, i.e. `bluetoothd` is not running. Do upload the full HCI logs using `hcidump` tool for futher analysis of the issue.

Also, there are some work in progress related to Direct Test Mode support in Zephyr, please follow https://github.com/zephyrproject-rtos/zephyr/pull/42300

carlescufi, 2022-02-09:
@piotrParf can you please enclose a full `btmon` trace of the sequence please?

piotrParf, 2022-02-09:
I see nothing specia at btmon at the moment. Will investigate tomorrow. Up to today it looks like this. 
`sudo btmon
[sudo] password for user: 
Bluetooth monitor ver 5.55
= Note: Linux version 5.10.0-9-amd64 (x86_64)                                            0.496297
= Note: Bluetooth subsystem version 2.22                                                 0.496298
@ MGMT Open: bluetoothd (privileged) version 1.18                               {0x0001} 0.496299
= New Index: 00:00:00:00:00:00 (Primary,UART,hci0)                               [hci0] 13.425551
= Open Index: 00:00:00:00:00:00                                                  [hci0] 13.425587
< HCI Command: Reset (0x03|0x0003) plen 0                                     #1 [hci0] 13.425785
> HCI Event: Command Complete (0x0e) plen 4                                   #2 [hci0] 13.427845
      Reset (0x03|0x0003) ncmd 1
        Status: Success (0x00)
< HCI Command: Read Local Supported Features (0x04|0x0003) plen 0             #3 [hci0] 13.431066
> HCI Event: Command Complete (0x0e) plen 12                                  #4 [hci0] 13.433043
      Read Local Supported Features (0x04|0x0003) ncmd 1
        Status: Success (0x00)
        Features: 0x00 0x00 0x00 0x00 0x60 0x00 0x00 0x00
          BR/EDR Not Supported
          LE Supported (Controller)
< HCI Command: Read Local Version Information (0x04|0x0001) plen 0            #5 [hci0] 13.433112
> HCI Event: Command Complete (0x0e) plen 12                                  #6 [hci0] 13.433958
      Read Local Version Information (0x04|0x0001) ncmd 1
        Status: Success (0x00)
        HCI version: Reserved (0x0c) - Revision 0 (0x0000)
        LMP version: Reserved (0x0c) - Subversion 65535 (0xffff)
        Manufacturer: The Linux Foundation (1521)
< HCI Command: Read BD ADDR (0x04|0x0009) plen 0                              #7 [hci0] 13.434044
> HCI Event: Command Complete (0x0e) plen 10                                  #8 [hci0] 13.434933
      Read BD ADDR (0x04|0x0009) ncmd 1
        Status: Success (0x00)
        Address: 00:00:00:00:00:00 (OUI 00-00-00)
< HCI Command: LE Read Buffer Size (0x08|0x0002) plen 0                       #9 [hci0] 13.435092
> HCI Event: Command Complete (0x0e) plen 7                                  #10 [hci0] 13.435819
      LE Read Buffer Size (0x08|0x0002) ncmd 1
        Status: Success (0x00)
        Data packet length: 27
        Num data packets: 7
< HCI Command: LE Read Local Supported Features (0x08|0x0003) plen 0         #11 [hci0] 13.435866
> HCI Event: Command Complete (0x0e) plen 12                                 #12 [hci0] 13.436787
      LE Read Local Supported Features (0x08|0x0003) ncmd 1
        Status: Success (0x00)
        Features: 0xff 0x49 0x01 0x00 0x00 0x00 0x00 0x00
          LE Encryption
          Connection Parameter Request Procedure
          Extended Reject Indication
          Slave-initiated Features Exchange
          LE Ping
          LE Data Packet Length Extension
          LL Privacy
          Extended Scanner Filter Policies
          LE 2M PHY
          LE Coded PHY
          Channel Selection Algorithm #2
          Minimum Number of Used Channels Procedure
< HCI Command: LE Read Supported States (0x08|0x001c) plen 0                 #13 [hci0] 13.436859
> HCI Event: Command Complete (0x0e) plen 12                                 #14 [hci0] 13.437731
      LE Read Supported States (0x08|0x001c) ncmd 1
        Status: Success (0x00)
        States: 0x000003ffff3fffff
          Non-connectable Advertising State
          Scannable Advertising State
          Connectable Advertising State
          High Duty Cycle Directed Advertising State
          Passive Scanning State
          Active Scanning State
          Initiating State
            and Connection State (Master Role)
          Connection State (Slave Role)
          Non-connectable Advertising State
[... 794 lines cut ...]
@ RAW Open: hcitool (privileged) version 2.22                                  {0x0002} 63.933710
@ RAW Close: hcitool                                                           {0x0002} 63.933748
@ RAW Open: hcitool (privileged) version 2.22                                  {0x0002} 63.933762
@ RAW Close: hcitool                                                           {0x0002} 63.933770
@ RAW Open: hcitool (privileged) version 2.22                           {0x0002} [hci0] 63.933784
< HCI Command: LE Transmitter Test (0x08|0x001e) plen 4                      #89 [hci0] 63.933928
        invalid packet size
        02 20 00 00                                      . ..            
> HCI Event: Command Complete (0x0e) plen 4                                  #90 [hci0] 63.935645
      LE Transmitter Test (0x08|0x001e) ncmd 1
        Status: Success (0x00)
@ RAW Close: hcitool                                                    {0x0002} [hci0] 63.935768
@ RAW Open: hcitool (privileged) version 2.22                                  {0x0002} 77.997643
@ RAW Close: hcitool                                                           {0x0002} 77.997652
@ RAW Open: hcitool (privileged) version 2.22                                  {0x0002} 77.997656
@ RAW Close: hcitool                                                           {0x0002} 77.997657
@ RAW Open: hcitool (privileged) version 2.22                           {0x0002} [hci0] 77.997663
< HCI Command: LE Test End (0x08|0x001f) plen 0                              #91 [hci0] 78.008578
> HCI Event: Command Complete (0x0e) plen 6                                  #92 [hci0] 84.156464
      LE Test End (0x08|0x001f) ncmd 1
        Status: Success (0x00)
        Number of packets: 0
@ RAW Close: hcitool                                                    {0x0002} [hci0] 84.156669`

The problem also occurs when i call ll_test_tx() from firmware on my own not just while connected to hci or so. It was showed made so it could be replicated using HCI- UART example.

piotrParf, 2022-02-15:
From what I found up to date:
When running the ll_test_tx(1,20,0,0x01) function from firmware code the packets start transmitting and I am receiving the test packets at other side of the testing enviroment. But at some point(about 2500-4000 packets elapsed) the transmission timer is transmitting with very big period of time. When I measure the period beetwen the interrupts when this "long period starts happening" I receive value of 549762.
I am trying out how the timer is being used to trigger the radio sending packets in ll_test and i do not figured it out to the moment...

cvinayak, 2022-02-15:
It is an old implementation, the difference should not be large and the while loop should not iterate too many times such that it is not able to meet the next 625us. But I do see that under rollover of the timer count, the difference will be incorrectly very high.

Your suggestion is good, but do add implementation to handle rollover scenario. Please send a pull request with your changes, we can continue to refine the changes in that PR.

piotrParf, 2022-02-16:
should I do pool request on current dev branch or some 2.7 lts ? I am not familiar with working with such a huge system.

cvinayak, 2022-02-16:
You should send the pull request to the latest `main` branch, backports will be generated by github actions when we add labels.
For contributing guide, refer to https://docs.zephyrproject.org/latest/contribute/index.html#contribution-workflow

dhananjaygj, 2022-03-15:
We are also trying to use the DTM functionality of zephyr BLE controller with nRF53. We are facing problem where `ll_test_tx` works only for maximum 1 second and after that i don't see any RF signals in the analyser. 
@piotrParf 
Have you faced a problem like this too? 
Does your PR (https://github.com/zephyrproject-rtos/zephyr/pull/42866) fix DTM related timer issues? 
From your last comment, do you mean to say we should enable `CONFIG_BT_CTLR_TIFS_HW` for nRF53?

piotrParf, 2022-03-17:
@dhananjaygj 
From what I found enabling CONFIG_BT_CTLR_TIFS_HW will work but to make it work you need also to set BT_CTLR_ADVANCED_FEATURES=y otherwise it is not possible to set BT_CTLR_RADIO_ENABLE_FAST=n. 
But the timers can still rollover and wait a long time without fixing it(it is rare but can).
For me that PR fixed the issue because the radio sampling timer handlers are failing(functions radio_tmr_end_capture is written otherway in never versions and radio_tmr_end_get reads zero) with 2.7 on nrf5340 on default configuration with CONFIG_BT_CTLR_TIFS_HW disabled.

closed 2022-04-22 by carlescufi