zephyrproject-rtos/zephyr #42536: Nominate Seppo Takalo for LwM2M maintainer

SeppoTakalo, 2022-02-07:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for an elevated permission level in
the Contributor role.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Seppo Takalo
1. GitHub username SeppoTakalo
1. Organization (optional) nrfconnect

## Supporting Documents

Add links to 3-5 GitHub pull requests, in the Zephyr project, authored or
reviewed by the GitHub user that demonstrate the user's dedication to the
Zephyr project.

Submitted PRs:
* https://github.com/zephyrproject-rtos/zephyr/pull/42169
* https://github.com/zephyrproject-rtos/zephyr/pull/30711
* https://github.com/zephyrproject-rtos/zephyr/pull/32451
* https://github.com/nrfconnect/sdk-nrf/pull/5667

Reviews:
* https://github.com/zephyrproject-rtos/zephyr/pull/41718
* https://github.com/zephyrproject-rtos/zephyr/pull/41158

In reality, I don't submit so much code, but I do review lots of stuff submitted by my team. Not all these are in public. We work currently on Zephyr's LwM2M client.

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/development_process/project_roles.html#tsc-project-roles>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-permission-levels-for-an-organization>

SeppoTakalo, 2022-02-07:
CC: @rlubos and @carlescufi

carlescufi, 2022-02-07:
@SeppoTakalo I've reworded this so that you can become  a collaborator first.

SeppoTakalo, 2022-02-07:
Thanks @carlescufi 
Yes, the plan was to propose me as a collaborator, not maintainer. (I need a cup of coffee now)

rlubos, 2022-02-07:
Thanks, from my side, I'd really appreciate a collaborator for LwM2M to help with the reviewing process.

PR #42537 (2022-02-07): MAINTAINERS: Add SeppoTakalo as a collaborator for LwM2M
Add SeppoTakalo as a collaborator for LwM2M subsystem and
the client sample.

Nomination ticket:
Closes #42536
closed 2022-02-07 by nashif