zephyrproject-rtos/zephyr #42555: mimxrt1050_evk: samples/subsys/task_wdt is failing with thread

yerabolu, 2022-02-07:
**Describe the bug**
samples/subsys/task_wdt test is failing with below log.

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p mimxrt1050_evk --device-testing --device-serial /dev/ttyACM1 -T samples/subsys/task_wdt/ -vv
2. see error as shown in log
3.  Ran with west as well and see the same error. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs and console output**
```
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
Task watchdog sample application.
task wdt init failure: -22
[00:00:00.008,331] [1;31m<err> wdt_mcux_wdog: Invalid timeoutValue, valid (0.5s - 128.0s)[0m
[00:00:00.008,331] [1;31m<err> task_wdt: hw_wdt install timeout failed: -22[0m
Control thread started.
[00:00:01.007,354] [1;31m<err> wdt_mcux_wdog: No valid timeouts installed[0m
[00:00:01.007,354] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.007,354] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.057,617] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.107,910] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.158,203] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.208,496] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.258,789] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.309,082] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.359,375] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.409,667] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.459,960] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.510,253] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.560,546] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.610,839] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.661,132] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.711,425] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.761,718] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.812,011] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.862,304] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.912,597] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:01.962,890] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.013,183] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.063,476] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.113,769] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.164,062] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.214,355] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.264,648] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.314,941] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.365,234] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.415,527] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.465,820] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.516,113] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.566,406] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.616,699] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.666,992] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.717,285] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.767,578] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.817,871] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.868,164] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.918,457] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:02.968,750] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:03.019,042] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:03.069,335] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:03.119,628] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:03.169,921] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:03.220,214] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:03.270,507] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
Control thread getting stuck...
[00:00:03.320,800] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:03.371,093] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:03.421,386] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m
[00:00:03.471,679] [1;31m<err> wdt_mcux_wdog: Invalid channel id[0m

```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain SDK 13.2
 - Commit SHA: a775606ae955

LixinGuoX, 2022-02-08:
The same with https://github.com/zephyrproject-rtos/zephyr/issues/40153, which one should we keep?

yerabolu, 2022-02-08:
Will close this one as a duplicate.

closed 2022-02-08 by yerabolu