zephyrproject-rtos/zephyr #42556: frd_k64f: samples/subsys/modbus are failing with out a timeout.

yerabolu, 2022-02-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. twister  -p frdm_k64f --device-testing --device-serial /dev/ttyACM0 -T samples/subsys/modbus
2. see error as shown in the log below and timwouts.
3. Same result when run with west.

**Logs and console output**
```
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[... 1146 lines cut ...]
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
*** Booting Zephyr OS build v3.0.0-rc1-307-ga775606ae955  ***
[00:00:00.005,000] [0m<inf> modbus_serial: RTU timeout 2005 us[0m
```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain SDk 132.
 - Commit SHA: a775606ae955

LixinGuoX, 2022-02-08:
Shall we change title name to frdm_k64f?

jfischer-no, 2022-02-08:
You need a test fixture https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/subsys/modbus/src/test_modbus.h#L22-L26

LixinGuoX, 2022-02-11:
@jfischer-no Hi, we can not find the pins(PTB16/PTB17)  which are connected to the openSDAv2 according to the SCH of k64f, could you share a photo of how to connect those pins on real board?
[image]

jfischer-no, 2022-02-11:
PTB16/PTB17 should be PTC16/PTC17, sorry my mistake, have mixed it up with UART0, I prepare PR to fix description.

Zhaoningx, 2022-02-11:
Hi @jfischer-no, we try to connect PTC17 <-> PTD2 and PTC16 <-> PTD3, but it didn't work.  Have you tested or do need we to add some configures else? I'm not sure UART3 and UART2 are both enabled in zephyr dts directory.

jfischer-no, 2022-02-11:
> PTC17 <-> PTD2 and PTC16 <-> PTD3

yes, that is correct text fixture for _tests/subsys/modbus_ and FRDM-K64F board

LixinGuoX, 2022-02-14:
We still could not get _tests/subsys/modbus_ passed after connecting `PTC17 <-> PTD2 and PTC16 <-> PTD3` , could you help to check again?
log as below:
```
DEBUG - Building /home/zn/zephyrproject/zephyr/tests/subsys/modbus for frdm_k64f
DEBUG - run test: frdm_k64f/tests/subsys/modbus/subsys.modbus.rtu
DEBUG - Waiting for device frdm_k64f to become available
DEBUG - Waiting for device frdm_k64f to become available
DEBUG - Waiting for device frdm_k64f to become available
DEBUG - Waiting for device frdm_k64f to become available
DEBUG - Waiting for device frdm_k64f to become available
...
```

Zhaoningx, 2022-02-14:
Hi @jfischer-no, the wiring diagram is as below:
[image]

It didn't work about the case tests/subsys/modbus, so I would like to double confirm with you if it is right? If yes, could help to check it in your local environment?

jfischer-no, 2022-02-14:
> he wiring diagram is as below:

It looks right.

> DEBUG - Waiting for device frdm_k64f to become available
> DEBUG - Waiting for device frdm_k64f to become available
> DEBUG - Waiting for device frdm_k64f to become available

Flash it manually to the board and it will pass the test, looks like a twister bug, and has nothing to do with MODBUS.

LixinGuoX, 2022-02-14:
Yes, verified by running manually via west. And after defining a hardware map file, twister can work now. Thanks! @jfischer-no

PR #42592 (2022-02-10): modbus: minor fixes for v3.0.0
Minor MODBUS fixes for v3.0.0

~~Based/depends~~ follow up on #42550

Fixes: #42556
closed 2022-02-14 by LixinGuoX