zephyrproject-rtos/zephyr #42584: LWM2M: store and restore update state in permanent storage

petterssonandreas, 2022-02-08:
**Is your enhancement proposal related to a problem? Please describe.**
When I do a firmware update over the air via my LWM2M server, I need to reboot my device to apply the firmware update. After the reboot, I can verify the new image, eg. by running some tests. Then I want to signal to my LWM2M server that the update passed. When I do, I get a error log (not breaking) that the current state is unexpected, as it was reset to IDLE when I rebooted the device.

**Describe the solution you'd like**
There is a TODO in `lwm2m_obj_firmware.c` regarding restoring the update state from permanent storage.

https://github.com/zephyrproject-rtos/zephyr/blob/179fe06a808398b4899492f3e70a1549c79ba07d/subsys/net/lib/lwm2m/lwm2m_obj_firmware.c#L431-L434

That would be ideal, is there any plans to implement that?

**Describe alternatives you've considered**
I could store the state myself, and set it at startup, but would of course be nice if it was automatic.

**Additional context**
My current main commit where I inspected the code: 179fe06a808398b4899492f3e70a1549c79ba07d

carlescufi, 2022-02-08:
I would suggest using the settings subsystem for this purpose.

rlubos, 2024-02-09:
LwM2M FW update image processing (and thus FW update state) are application responsibility, and as for now there are no plans to change that.

closed 2024-02-09 by rlubos