zephyrproject-rtos/zephyr #42588: lsm6dso

RomainArtifeel, 2022-02-08:
Compilation issue : 

In zephyr/drivers/sensor/lsm6dso_trigger.c line 45 in function _lsm6dso_enable_t_int_, the variable int2_route is not defined and has been replaced by int2_ctrl (struct type lsm6dso_int2_ctrl_t ) line 31.

static int lsm6dso_enable_t_int(const struct device *dev, int enable)
{
	const struct lsm6dso_config *cfg = dev->config;
	stmdev_ctx_t *ctx = (stmdev_ctx_t *)&cfg->ctx;
	struct lsm6dso_data *lsm6dso = dev->data;
	lsm6dso_int2_ctrl_t int2_ctrl;

	if (enable) {
		int16_t buf;

		/* dummy read: re-trigger interrupt */
		lsm6dso_temperature_raw_get(ctx, &buf);
	}

	/* set interrupt (TEMP DRDY interrupt is only on INT2) */
	if (cfg->int_pin == 1)
		return -EIO;

	lsm6dso_read_reg(ctx, LSM6DSO_INT2_CTRL, (uint8_t *)&int2_ctrl, 1);
	int2_route.int2_ctrl.int2_drdy_temp = enable;
	return lsm6dso_write_reg(ctx, LSM6DSO_INT2_CTRL,
				 (uint8_t *)&int2_ctrl, 1);
}
#endif

The correction is to replace the line 45 by : int2_ctrl.int2_drdy_temp = enable;

gmarull, 2022-02-08:
Would you mind sending a PR?

RomainArtifeel, 2022-02-08:
PR added

avisconti, 2022-02-10:
LGTM, but you need to create a PR on zephyr mentioning #42588, so that we can review/approve.

gmarull, 2022-02-10:
https://github.com/zephyrproject-rtos/zephyr/pull/42589

petterssonandreas, 2022-04-14:
Since this issue and related PR was closed for some reason, without a fix, I made a new PR: https://github.com/zephyrproject-rtos/zephyr/pull/44873

PR #44873 (2022-04-20): drivers: sensor: lsm6dso: Remove usage of undefined variable
Remove usage of undefined variable, and remove unused variable.

Fixes the same thing as https://github.com/zephyrproject-rtos/zephyr/pull/42589 which for some unknown reason was closed by the author. Fixes the issue https://github.com/zephyrproject-rtos/zephyr/issues/42588 which was closed by the author without a fix.

Fixes #42588
closed 2022-04-20 by MaureenHelm