zephyrproject-rtos/zephyr #42627: Hardfault regression on 90 tests on CM0+ STM32 boards introduced by #39963 Cortex-R mpu fix  on 90 tests

ABOSTM, 2022-02-09:
**Describe the bug**
Regression, Hardfault, introduced by PR `Cortex-R mpu fix ` #39963.
Especially commit `arch: arm: core: aarch32: Fix the syscall design for Cortex-R`
sha1: 2a357e5dfd749b9d7aa5bc5ff5ed8a3ef5e5a498

2 boards are concerned: 
* nucleo_g071rb
* nucleo_l073rz

Note1: both are Cortex-M0-Plus (Armv6-M)
Note:2 nucleo_f091rc (Cortex-M0) is not impacted

90 tests are concerned:
   * samples/userspace/hello_world_user/sample.helloworld, nucleo_g071rb (Regression)
   * samples/userspace/hello_world_user/sample.helloworld, nucleo_l073rz (Regression)
   * samples/userspace/prod_consumer/sample.userspace.prod_consumer, nucleo_g071rb (Regression)
   * samples/userspace/shared_mem/sample.kernel.memory_protection.shared_mem, nucleo_g071rb (Regression)
   * samples/userspace/shared_mem/sample.kernel.memory_protection.shared_mem, nucleo_l073rz (Regression)
   * tests/arch/arm/arm_interrupt/arch.interrupt.arm, nucleo_g071rb (Regression)
   * tests/arch/arm/arm_interrupt/arch.interrupt.arm, nucleo_l073rz (Regression)
   * tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info, nucleo_g071rb (Regression)
   * tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info, nucleo_l073rz (Regression)
   * tests/arch/arm/arm_interrupt/arch.interrupt.no_optimizations, nucleo_l073rz (Regression)
   * tests/arch/arm/arm_thread_swap/arch.arm.swap.common, nucleo_g071rb (Regression)
   * tests/arch/arm/arm_thread_swap/arch.arm.swap.common, nucleo_l073rz (Regression)
   * tests/arch/arm/arm_thread_swap/arch.arm.swap.common.no_optimizations, nucleo_l073rz (Regression)
   * tests/drivers/adc/adc_api/drivers.adc, nucleo_g071rb (Regression)
   * tests/drivers/counter/counter_basic_api/drivers.counter, nucleo_g071rb (Regression)
   * tests/drivers/dac/dac_api/drivers.dac, nucleo_l073rz (Regression)
   * tests/drivers/pwm/pwm_api/drivers.pwm, nucleo_g071rb (Regression)
   * tests/drivers/uart/uart_async_api/drivers.uart.uart_async_api, nucleo_g071rb (Regression)
   * tests/kernel/common/kernel.common, nucleo_g071rb (Regression)
   * tests/kernel/common/kernel.common.misra, nucleo_g071rb (Regression)
   * tests/kernel/common/kernel.common.nano32, nucleo_g071rb (Regression)
   * tests/kernel/common/kernel.common.nano64, nucleo_g071rb (Regression)
   * tests/kernel/common/kernel.common.tls, nucleo_g071rb (Regression)
   * tests/kernel/condvar/condvar_api/kernel.condvar, nucleo_g071rb (Regression)
   * tests/kernel/device/kernel.device, nucleo_g071rb (Regression)
   * tests/kernel/device/kernel.device, nucleo_l073rz (Regression)
   * tests/kernel/device/kernel.device.pm, nucleo_g071rb (Regression)
   * tests/kernel/device/kernel.device.pm, nucleo_l073rz (Regression)
   * tests/kernel/mem_protect/futex/kernel.futex, nucleo_g071rb (Regression)
   * tests/kernel/mem_protect/stackprot/kernel.memory_protection.stackprot, nucleo_g071rb (Regression)
   * tests/kernel/mem_protect/stackprot/kernel.memory_protection.stackprot, nucleo_l073rz (Regression)
   * tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem, nucleo_g071rb (Regression)
   * tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem, nucleo_l073rz (Regression)
   * tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls, nucleo_g071rb (Regression)
   * tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls, nucleo_l073rz (Regression)
   * tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace, nucleo_g071rb (Regression)
   * tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace, nucleo_l073rz (Regression)
   * tests/kernel/msgq/msgq_api/kernel.message_queue, nucleo_g071rb (Regression)
   * tests/kernel/msgq/msgq_api/kernel.message_queue, nucleo_l073rz (Regression)
   * tests/kernel/mutex/mutex_api/kernel.mutex, nucleo_g071rb (Regression)
   * tests/kernel/mutex/mutex_api/kernel.mutex, nucleo_l073rz (Regression)
   * tests/kernel/mutex/mutex_error_case/kernel.mutex_error_case, nucleo_g071rb (Regression)
   * tests/kernel/mutex/mutex_error_case/kernel.mutex_error_case, nucleo_l073rz (Regression)
   * tests/kernel/mutex/sys_mutex/system.mutex, nucleo_g071rb (Regression)
   * tests/kernel/pipe/pipe/kernel.pipe, nucleo_g071rb (Regression)
   * tests/kernel/pipe/pipe/kernel.pipe, nucleo_l073rz (Regression)
   * tests/kernel/pipe/pipe_api/kernel.pipe.api, nucleo_g071rb (Regression)
   * tests/kernel/pipe/pipe_api/kernel.pipe.api, nucleo_l073rz (Regression)
   * tests/kernel/poll/kernel.poll, nucleo_g071rb (Regression)
   * tests/kernel/poll/kernel.poll, nucleo_l073rz (Regression)
   * tests/kernel/queue/kernel.queue, nucleo_g071rb (Regression)
   * tests/kernel/semaphore/semaphore/kernel.semaphore, nucleo_g071rb (Regression)
   * tests/kernel/semaphore/sys_sem/kernel.semaphore.usage, nucleo_g071rb (Regression)
   * tests/kernel/sleep/kernel.common.timing, nucleo_g071rb (Regression)
   * tests/kernel/sleep/kernel.common.timing, nucleo_l073rz (Regression)
   * tests/kernel/stack/stack/kernel.stack.usage, nucleo_g071rb (Regression)
   * tests/kernel/stack/stack/kernel.stack.usage, nucleo_l073rz (Regression)
   * tests/kernel/threads/dynamic_thread/kernel.threads.dynamic, nucleo_g071rb (Regression)
   * tests/kernel/threads/dynamic_thread/kernel.threads.dynamic, nucleo_l073rz (Regression)
   * tests/kernel/threads/thread_apis/kernel.threads.apis, nucleo_g071rb (Regression)
   * tests/kernel/threads/thread_error_case/kernel.threads.error.case, nucleo_g071rb (Regression)
   * tests/kernel/threads/thread_error_case/kernel.threads.error.case, nucleo_l073rz (Regression)
   * tests/kernel/threads/thread_init/kernel.threads.init, nucleo_g071rb (Regression)
   * tests/kernel/threads/thread_init/kernel.threads.init, nucleo_l073rz (Regression)
   * tests/kernel/threads/thread_stack/kernel.threads.thread_stack, nucleo_g071rb (Regression)
   * tests/kernel/threads/thread_stack/kernel.threads.thread_stack, nucleo_l073rz (Regression)
   * tests/kernel/threads/tls/kernel.threads.tls.userspace, nucleo_g071rb (Regression)
   * tests/kernel/threads/tls/kernel.threads.tls.userspace, nucleo_l073rz (Regression)
   * tests/kernel/timer/timer_api/kernel.timer, nucleo_g071rb (Regression)
   * tests/kernel/timer/timer_api/kernel.timer, nucleo_l073rz (Regression)
   * tests/kernel/timer/timer_error_case/kernel.timer.error_case, nucleo_g071rb (Regression)
   * tests/kernel/timer/timer_error_case/kernel.timer.error_case, nucleo_l073rz (Regression)
   * tests/kernel/workq/user_work/kernel.work.user, nucleo_g071rb (Regression)
   * tests/kernel/workq/user_work/kernel.work.user, nucleo_l073rz (Regression)
   * tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc, nucleo_g071rb (Regression)
   * tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc, nucleo_l073rz (Regression)
   * tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc_negative_testing, nucleo_g071rb (Regression)
   * tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc_negative_testing, nucleo_l073rz (Regression)
   * tests/lib/mem_alloc/libraries.libc.newlib.mem_alloc, nucleo_g071rb (Regression)
   * tests/lib/mem_alloc/libraries.libc.newlib.mem_alloc, nucleo_l073rz (Regression)
   * tests/lib/mem_alloc/libraries.libc.newlib_nano.mem_alloc, nucleo_g071rb (Regression)
   * tests/lib/mem_alloc/libraries.libc.newlib_nano.mem_alloc, nucleo_l073rz (Regression)
   * tests/lib/sprintf/libraries.libc.sprintf, nucleo_g071rb (Regression)
   * tests/lib/sprintf/libraries.libc.sprintf, nucleo_l073rz (Regression)
   * tests/lib/sprintf/libraries.libc.sprintf_new, nucleo_g071rb (Regression)
   * tests/lib/sprintf/libraries.libc.sprintf_new, nucleo_l073rz (Regression)
   * tests/subsys/logging/log_core_additional/logging.add.log1_user, nucleo_g071rb (Regression)
   * tests/subsys/logging/log_core_additional/logging.add.log1_user, nucleo_l073rz (Regression)
   * tests/ztest/error_hook/testing.ztest.error_hook, nucleo_g071rb (Regression)
   * tests/ztest/error_hook/testing.ztest.error_hook, nucleo_l073rz (Regression)

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_l073rz tests/kernel/mutex/mutex_api/
2. west flash
3. See error

**Logs and console output**
Log of nucleo_l073rz,  tests/kernel/mutex/mutex_api/
```
*** Booting Zephyr OS build v3.0.0-rc1-340-g2a357e5dfd74  ***
Running test suite mutex_api
===================================================================
START - test_mutex_lock_unlock
E: ***** HARD FAULT *****
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000047 r14/lr:  0x00000004
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x0000000a
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000328 (test_mutex_lock_unlock)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux and Windows
 - Toolchain SDK
 - Commit SHA 2a357e5dfd749b9d7aa5bc5ff5ed8a3ef5e5a498

ABOSTM, 2022-02-09:
@bbolen, I assigned this issue to you, as you created the PR that generates the regression.
@cfriedt, I set label `Release Blocker` , due to the high number of regression introduced.

bbolen, 2022-02-09:
looking...

PR #42637 (2022-02-10): arch: arm: core: aarch32: Fix Cortex-M userspace regression
This was introduced when trying to fix a previous merge conflict.  It
broke userspace tests on nucleo_l073rz.

This fixes #39963

Fixes #42627
closed 2022-02-10 by nashif