zephyrproject-rtos/zephyr #42629: stm32g0: Device hang/hard fault with AT45 + `CONFIG_PM_DEVICE`

ycsin, 2022-02-09:
**Describe the bug**
I'm trying to rebase to the latest main from v2.7.0 RC3, but I encountered hard fault with my custom application on init:

```
I: Jumping to the first image slot

[00:00:00.004,000] <err> os: r0/a1:  0x00000004  r1/a2:  0x000000ba  r2/a3:  0x2000954e
[00:00:00.004,000] <err> os: r3/a4:  0x0802e875 r12/ip:  0x00000000 r14/lr:  0x0802e87d
[00:00:00.004,000] <err> os:  xpsr:  0x21000000
[00:00:00.004,000] <err> os: Faulting instruction address (r15/pc): 0x08035498
[00:00:00.004,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.004,000] <err> os: Current thread: 0x20003110 (idle 00)
[00:00:00.055,000] <err> os: Halting system
```

[image]

I'm using a STM32G0B1RE custom board with `CONFIG_SPI_FLASH_AT45` enabled in the `_defconfig` by default.

I tried to compile the vanilla (with bootloader) hello_world & AT45 sample, they both work fine. However if I enable `CONFIG_PM` & `CONFIG_PM_DEVICE` in the `hello_world` example, then the device will hard fault and halt inside [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/spi_flash_at45.c#L601) (Zephyr didn't print anything).

I tried to increase the IDLE thread stack size to 1024, and now PM is suspending and resuming the [AT45 driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/spi_flash_at45.c#L593) repeatedly.

**To Reproduce**
Steps to reproduce the behavior:
1. Build `hello_world` sample (with bootloader & AT45 driver)
2. See error

**Environment (please complete the following information):**
 - OS: Windows 11
 - GNUARMEMB
 - Latest main (3.0.0 RC2)

ABOSTM, 2022-02-10:
Hi @ycsin,
It looks similar to #42627
Can you check the proposed fix: #42637

ABOSTM, 2022-04-07:
@ycsin 
Is issue solved ?

ycsin, 2022-06-06:
> @ycsin Is issue solved ?

Hi, sorry for the delay in response. We had decided stay on the 2.7LTS version instead on the latest main branch for the sake of stability, I can't really test this anymore.

closed 2022-06-06 by ycsin