zephyrproject-rtos/zephyr #42653: STM32 Hangs sending TCP packets

mdkf, 2022-02-09:
**Describe the bug**
An issue with the TCP ACK packet causes the STM32 to throw ERRNO 12 "Not enough core" when sending TCP packets. This does not occur on the FRDM-K64F

Please also mention any information which could help others to understand
the problem you're facing:
 - NUCLEO-F429ZI
 - Inserting a 2.3 mS delay between packets mitigates the issue, but reduces throughput to 0.4 Mbps

**To Reproduce**
Steps to reproduce the behavior:
Code used to test: https://github.com/mdkf/ZephyrTCPSlow

**Expected behavior**
While the 2Mbps I achieved with this same code (minus the delay) on the FRDM-K64F was painfully slow (see https://github.com/zephyrproject-rtos/zephyr/issues/23302), I should be able to get at least that on the STM32.

**Impact**
TCP over Ethernet is too slow and unreliable to be useable

**Logs**
Here is a Wireshark capture when the cod was run without a delay:
[Zephyr STM32.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/8034979/Zephyr.STM32.pcapng.gz)

**Environment (please complete the following information):**
 - Zephyr 2.7.1

mdkf, 2022-03-02:
@dkalowsk The code is now completely broken on 3.0.99. It gives the same fualt on the FRDM-K64, IMXRT1050-EVKB, and the NUCLEO-F746ZG

uart:~$ *** Booting Zephyr OS build zephyr-v3.0.0-337-g52bee41bba11  ***
[00:00:03.556,000] <inf> net_telnet_sample: Starting Telnet sample nwe

[00:00:03.556,000] <inf> net_telnet_sample: Running dhcpv4 client...
[00:00:03.855,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
uart:~$ IPv4 address: 172.16.0.211out of loop

[00:00:05.902,000] <inf> net_dhcpv4: Received: 172.16.0.211
[00:00:05.904,000] <inf> net_telnet_sample: Lease time: 86400 seconds
[00:00:05.904,000] <inf> net_telnet_sample: Subnet: 255.255.255.0
[00:00:05.904,000] <inf> net_telnet_sample: Router: 172.16.0.1
[00:00:05.909,000] <err> os: ***** MPU FAULT *****
[00:00:05.909,000] <err> os:   Stacking error (context area might be not valid)
[00:00:05.909,000] <err> os:   Data Access Violation
[00:00:05.909,000] <err> os:   MMFAR Address: 0x20013fc8
[00:00:05.909,000] <err> os: r0/a1:  0xb75e30c9  r1/a2:  0x1522f7fc  r2/a3:  0x47f47bfe
[00:00:05.909,000] <err> os: r3/a4:  0x63e2131f r12/ip:  0xebb8eb51 r14/lr:  0xad670aea
[00:00:05.909,000] <err> os:  xpsr:  0xe000ec00
[00:00:05.910,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:05.910,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:05.910,000] <err> os: Current thread: 0x200116c8 (main)
[00:00:06.045,000] <err> os: Halting system
uart:~$

erwango, 2022-03-03:
> @dkalowsk The code is now completely broken on 3.0.99. It gives the same fualt on the FRDM-K64, IMXRT1050-EVKB, and the NUCLEO-F746ZG

@mdkf Thanks for reporting this. To avoid confusion and allow quick action, would you mind raise a new issue for that particular point?

cfriedt, 2022-03-03:
@mdkf - can you reproduce the issue with HEAD from `v2.7-branch`? There were some bugfixes that went in for STM32 and networking recently.

mdkf, 2022-03-03:
@cfriedt How do I change branches with West?

cfriedt, 2022-03-03:
> @cfriedt How do I change branches with West?

```
git fetch origin
git checkout -b v2.7-branch
west update
```

erwango, 2022-05-19:
@mdkf Did you finally manage to reproduce your issue (app broken) on mainline ?

About the performance issue reported on STM32:
Note that there is a new Ethernet HAL API that has been developed which should overcome limitations of the previous one (cf https://github.com/zephyrproject-rtos/zephyr/pull/45770).
I can't say it will fix the performance issue you reported (even though I hope so), but there won't be significant efforts made on the legacy API at this point. If the perf issue is still present on the next version, we'll dig into it.

cfriedt, 2022-08-16:
@mdkf, @erwango - please try to get this sorted out soon, as v2.7.3 is scheduled to be cut on 2022-08-22

ssharks, 2022-09-11:
@mdkf There have been a lot of fixes in TCP between  March and July. Both reliability and throughput improved a lot. Could you give is a try with the current Zephyr version?

ssharks, 2022-10-23:
No reaction received, I think the issue is no longer applicable, closing.

closed 2022-10-23 by ssharks