zephyrproject-rtos/zephyr #42672: [backport v2.7-branch] backport of #38603 failed

zephyrbot, 2022-02-10:
This issue was created automatically because the backport of #38603 to `v2.7-branch` failed.

The backport to `v2.7-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v2.7-branch v2.7-branch
# Navigate to the new working tree
cd .worktrees/backport-v2.7-branch
# Create a new branch
git switch --create backport-38603-to-v2.7-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick 2ae9a54a09265b705542deaeeeccf5dc11dc0c6d~7..2ae9a54a09265b705542deaeeeccf5dc11dc0c6d
# Push it to GitHub
git push --set-upstream origin backport-38603-to-v2.7-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v2.7-branch
```
Then, create a pull request where the `base` branch is `v2.7-branch` and the `compare`/`head` branch is `backport-38603-to-v2.7-branch`.

cfriedt, 2022-04-12:
@rlubos - can you take a look at this one and see if can be manually applied to v2.7?

rlubos, 2022-04-13:
@cfriedt I've opened a PR with a manual backport (https://github.com/zephyrproject-rtos/zephyr/pull/44820).

PR #44820 (2022-04-13): [backport v2.7-branch] TCP fixes
Original PR was #38603
Fixes #42672 (automatic backport failed)
Fixes https://github.com/zephyrproject-rtos/zephyr/issues/38576

From Original PR:
2 important fixes:
* Fix local TCP connections (currently leads to non functioning connection and memory leak)
* Fix TCP disconnection initiated on Zephyr side (currently leads to memory leak)
closed 2022-04-19 by cfriedt