zephyrproject-rtos/zephyr #42687: HID is not connecting to intel 7265 Bluetooth Module

mabbas22, 2022-02-10:
**Describe the bug**
The HID (peripheral_hids) of Zephyr 1.13.0 is not connecting to Windows Tablet with Intel 7265 Bluetooth Module on it.
The peripheral connects and disconnects with reason (Remote User Terminated Connection - 0x13).

**To Reproduce**
Steps to reproduce the behavior:
1. Use the peripheral_hids of the Zephyr 1.13.0.

**Expected behavior**
The connection should be established

**Impact**
It is blocking us to provide solution to our customer.

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Tool chain (Zephyr 1.13.0)

carlescufi, 2022-02-10:
> 1. Use the peripheral_hids sample of the Zephyr 1.13.0.

Zephyr 1.13.0 is several years old. Are you sure about the version?

cvinayak, 2022-02-11:
It should not be very difficult to migrate to https://github.com/zephyrproject-rtos/zephyr/releases/tag/v1.14.3 which has many fixes.

mabbas22, 2022-02-14:
> > 1. Use the peripheral_hids sample of the Zephyr 1.13.0.
> 
> Zephyr 1.13.0 is several years old. Are you sure about the version?

Yes we are use this old version.

mabbas22, 2022-02-14:
I sniffed the BLE packets with wire shark and came to know that

From Slave  -> **_Control Opcode: LL_LENGTH_REQ (0x14)_**
From Master -> **_Control Opcode: LL_REJECT_IND
Error Code: Invalid LMP/LL Parameters (0x1e)_**

See in the attached image at below link.

https://ibb.co/P9HQ38r

Do you some specific changes that can fix the issue? Right now its not possible for use to migrate to 1.14.3.

cvinayak, 2022-02-15:
There are no PDU details in the screenshot, please attach the sniffer trace for further analysis.

Closest related changes found from v1.14.3 release could be https://github.com/zephyrproject-rtos/zephyr/commit/3f8c2fd39ceacf2dee2e555755d847162c43bd9f, but it will not apply clean onto v1.13.0, and will need many other dependent commits from the v1.14.3 source code.

mabbas22, 2022-02-15:
Hi @cvinayak,

Attached you can find the wire shark BLE sniffer logs. 

- windows_10_connection_failed_hid_sample_zephyr_1_13_0.pcapng -> windows 10 connection sample HID with zephyr 1.13.0
- windows_10_connection_failed_with_HID_sample_1_14_3.pcapng -> windows 10 connection sample HID with  logs with zephyr 1.14.3
- android_connection_with_HID_sample_zephyr_1_14_3.pcapng -> Android connection sample HID with  logs with zephyr 1.14.3

[wire_shark_sniffer_logs.zip](https://github.com/zephyrproject-rtos/zephyr/files/8067327/wire_shark_sniffer_logs.zip)

We have tried with zephyr 1.14.3 while trying to connect with the windows 10. The HID was connecting and disconnecting. There was LL_REJECT_IND_EXT as you can see in the attached image. 
[image]

If you need any further information please let me know.

cvinayak, 2022-02-15:
`windows_10_connection_failed_hid_sample_zephyr_1_13_0.pcapng`  disconnection initiated by central device after 30 seconds timeout due to peripheral not replying to the GATT MTU exchange response.

[image]

`windows_10_connection_failed_with_HID_sample_1_14_3.pcapng` disconnection is due to use of incorrect Long Term Key by the central, disconnections are due to Pin or Key missing reason. The issue should resolve if both peripheral and central start a fresh pairing procedure with old bond/keys removed.

[image]

`android_connection_with_HID_sample_zephyr_1_14_3.pcapng` I cant see anything wrong, the connection is alive until the end of the log frame no. 12017.

Please retake the logs with `CONFIG_BT_USE_DEBUG_KEYS=y` added to your application, so that the sniffer is able to decrypt the traffic after encryption is setup.

The data length issue (use of 2120 us as tx time) observed with v1.13.0 is resolved by https://github.com/zephyrproject-rtos/zephyr/commit/3f8c2fd39ceacf2dee2e555755d847162c43bd9f present in v1.14.3.

mabbas22, 2022-02-16:
Hi @cvinayak,
Building the application with `CONFIG_BT_CTLR_PHY_CODED=n` solve the connection issue with the windows Tablet.
What would be the impact of disabling PHY_CODED on the other devices that we have already worked on?

If it would create issues with other devices, then do you think [3f8c2fd](https://github.com/zephyrproject-rtos/zephyr/commit/3f8c2fd39ceacf2dee2e555755d847162c43bd9f) will resolve the issue?

What are other that dependent commit that we need to bring in **_zephyr 1.13.0_** as the [3f8c2fd](https://github.com/zephyrproject-rtos/zephyr/commit/3f8c2fd39ceacf2dee2e555755d847162c43bd9f) don,t apply clean to zephyr 1.13.0?

cvinayak, 2022-02-16:
> What would be the impact of disabling PHY_CODED on the other devices that we have already worked on?

There is should be no impact, in fact, `CONFIG_BT_CTLR_PHY_CODED=n` for nRF52840 on v1.13.0 is the correct thing to do.

> If it would create issues with other devices, then do you think [3f8c2fd](https://github.com/zephyrproject-rtos/zephyr/commit/3f8c2fd39ceacf2dee2e555755d847162c43bd9f) will resolve the issue?
Yes.

> What are other that dependent commit that we need to bring in zephyr 1.13.0 as the [3f8c2fd](https://github.com/zephyrproject-rtos/zephyr/commit/3f8c2fd39ceacf2dee2e555755d847162c43bd9f) don,t apply clean to zephyr 1.13.0?

You have to take all changes from v1.14.3 of the subsys/bluetooth/controller subsystem tree and some more out of the subsystem tree dependencies maybe required, it is not straight forward to list here.

mabbas22, 2022-02-16:
Thanks a lot @cvinayak .

closed 2022-02-16 by mabbas22