zephyrproject-rtos/zephyr #42705: sam_e70b_xplained: tests in kernel/mem_protect are failing.

yerabolu, 2022-02-10:
**Describe the bug**
kernel.memory_protection.kernel.memory_protection.alloc_kobjects
kernel.memory_protection.kernel.memory_protection.thread_alloc_out_of_idx
kernel.memory_protection.kernel.memory_protection.kobj_create_out_of_memory
kernel.memory_protection.kernel.memory_protection.kobject_perm_error
kernel.memory_protection.kernel.memory_protection.kobject_free_error
kernel.memory_protection.kernel.memory_protection.all_kobjects_str

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p sam_e70b_xplained --device-testing --device-serial /dev/ttyACM0 -T tests/kernel/mem_protect/ 
2. see error

**Logs and console output**
```
*** Booting Zephyr OS build v3.0.0-rc2-22-g32c4bd3b684a  ***
Running test suite memory_protection_test_suite
===================================================================
START - test_permission_inheritance
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x20401454
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000001
E: r3/a4:  0x20401454 r12/ip:  0x0040ca8f r14/lr:  0x004006ad
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x0040072a
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20400c10 (unknown)
Caught system error -- reason 0 1
fatal error expected as part of test case
PASS - test_permission_inheritance in 0.40 seconds
===================================================================
START - test_inherit_resource_pool
PASS - test_inherit_resource_pool in 0.1 seconds
===================================================================
START - test_mem_domain_setup
PASS - test_mem_domain_setup in 0.1 seconds
===================================================================
START - test_mem_domain_valid_access
PASS - test_mem_domain_valid_access in 0.1 seconds
===================================================================
START - test_mem_domain_invalid_access
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x204008e0
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x0000000a r12/ip:  0x0040cf61 r14/lr:  0x0040cf6d
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x004020ac
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20401098 (child_thread)
Caught system error -- reason 0 1
fatal error expected as part of test case
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x204008c0
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x0040cf61 r14/lr:  0x0040cf6d
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x00402110
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20401098 (child_thread)
Caught system error -- reason 0 1
fatal error expected as part of test case
PASS - test_mem_domain_invalid_access in 0.81 seconds
===================================================================
START - test_mem_domain_no_writes_to_ro
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x204008c0
[... 433 lines cut ...]
===================================================================
START - test_kobject_init_error
E: bad object type 255 requested
E: bad object type 55 requested
E: forbidden object type 'k_futex' requested
PASS - test_kobject_init_error in 0.10 seconds
===================================================================
START - test_alloc_kobjects
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x1d0ef3c
E: r0/a1:  0x01d0ef3c  r1/a2:  0x0040d307  r2/a3:  0x00000000
E: r3/a4:  0x00000020 r12/ip:  0x20402bd0 r14/lr:  0x0040b875
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x004067e4
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20401150 (test_alloc_kobjects)
Caught system error -- reason 0 0
fatal error was unexpected, aborting
```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain - SDK 13.2
 - Commit SHA: 32c4bd3b684a

dleach02, 2022-02-15:
@yerabolu, I am not able to reproduce this on the frdm_k64f. How repeatable is this for you?

yerabolu, 2022-02-15:
@dleach02 : i am able to reproduce it, but the reproducibility rate is very low. Failure happens randomly. So far i have run it 10 times and test failed 2 times.

dleach02, 2022-02-15:
@hakehuang, this is called out against an Atmel part but the reporter also noted it occurring on k64f.

dleach02, 2022-02-15:
Ok. I'm seeing the failure(s)

dleach02, 2022-02-16:
yeah, I see the same one

hakehuang, 2022-02-16:
@dleach02 mimxrt595_evk_cm33 also fails at this test and with different log.
[log.txt](https://github.com/zephyrproject-rtos/zephyr/files/8077193/log.txt)

hakehuang, 2022-02-16:
> @dleach02 mimxrt595_evk_cm33 also fails at this test and with different log. [log.txt](https://github.com/zephyrproject-rtos/zephyr/files/8077193/log.txt)
and it is ok at my frdm_k64 board log as linked
[frdm_k64f_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/8077555/frdm_k64f_log.txt)

my board is using openocd debugger, and the board revision is 
70028163 rev E
SCH-28163 REV F

dleach02, 2022-02-17:
This is randomly failing because of uninitialized data in an allocated K_OBJ_MSGQ. The test cases are doing a k_object_alloc(K_OBJ_MSGQ) but not following up the alloc with a k_msgq_init() to ensure that the fields are initialized appropriately. 

The test cases here are expecting fatal errors so after a fatal error it kills the thread which will cause the logic to walk through the k_objects to remove the thread ID references to any objects. In unref_check(), if the kernel object has no threads referencing it AND it is an allocated object, the code will free the object back to its respective heap.  In the case of a k_msg object, the k_msgq_cleanup() function will further check msgq->flags to determine if the buffer_start pointer points to an allocated buffer and if it does it will k_free() buffer_start back to its heap. 

Problem with uninitialized data is that the msgq->flags and msgq->buffer_start will have random RAM values and randomly it may attempt to free the buffer_start pointer to invalid memory or unaligned memory so the k_free() function will sometimes fault when doing this clean up.

The fix for this is likely to change the test suite to ensure allocated kernel objects are initialized. Per the documentation of k_object_alloc():

> The object will be returned in an uninitialized state, with the calling thread being granted permission on it

A kludge fix to verify this was to memset() the kernel object to zero in the allocation function. Not sure we want to incur that overhead since the documentation says the object returned is uninitialized.

dleach02, 2022-02-17:
@yerabolu can you test my PR #42913 on your system. 

I figured out the root issue and can see that it is a random type of failure.

dleach02, 2022-02-21:
> Found the same issue on reel_board.

Yeah, this will randomly occur on any board. Uninitialized memory has a way of doing this ;-)

PR #42913 (2022-02-22): tests: mem_protect: avoid allocating K_OBJ_MSGQ in userspace.
The K_OBJ_MSGQ object is unitialized so when the thread cleanup occurs after
an expected fault for invalid access the test case can randomly fault
again because the cleanup of the thread will sometimes attempt to free
invalid buffer_start pointer in the msgq object.

Fixes #42705
commit a0737e68 (2022-02-22): tests: mem_protect: avoid allocating K_OBJ_MSGQ in userspace.

The K_OBJ_MSGQ object is unitialized so when the thread cleanup occurs
after an expected fault for invalid access the test case can randomly
fault again because the cleanup of the thread will sometimes attempt
to free invalid buffer_start pointer in the msgq object.

Fixes #42705
commit d05c3bdf (2023-01-11): tests: mem_protect: avoid allocating K_OBJ_MSGQ in userspace.

The K_OBJ_MSGQ object is unitialized so when the thread cleanup occurs
after an expected fault for invalid access the test case can randomly
fault again because the cleanup of the thread will sometimes attempt
to free invalid buffer_start pointer in the msgq object.

Fixes #42705

(cherry picked from commit a0737e687cff8600d87f95f3daa2861fc0b13529)
closed 2022-02-22 by nashif