zephyrproject-rtos/zephyr #42746: echo_server and echo_client sample code builds fail for native_posix_64

sagarshah-leg, 2022-02-11:
Building echo_server and echo_client sample code fails for the native_posix_64 board with the following error:

zephyr/subsys/net/ip/tcp2.c:17:10: fatal error: mbedtls/md5.h: No such file or directory
   17 | #include <mbedtls/md5.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.

I have narrowed it down to CONFIG_NET_TCP config setting in prj.conf. 
CONFIG_NET_TCP=y causes the error. Setting it to "n" avoids the error.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b native_posix_64 -p always zephyr/samples/net/sockets/echo_server/
2. see build error
3. west build -b native_posix_64 -p always zephyr/samples/net/sockets/echo_client/
4. see build error

**Expected behavior**
The builds should compile without errors.

**Impact**
Medium impact, potential showstopper.

**Logs and console output**

**Environment (please complete the following information):**
 - OS:  Linux
 - Zephyr SDK v 2.7.99
 - Commit SHA or Version used

**Additional context**

sagarshah-leg, 2022-02-14:
Hi @rlubos I've already performed west update and it didn't solve the issue. For my info, where is it supposed to get the md5.h file from? Zephyr's zephyr/include/ directory doesn't seem to have that mbedtls/ dir.

closed 2022-02-14 by sagarshah-leg

rlubos, 2022-02-15:
> Zephyr's zephyr/include/ directory doesn't seem to have that mbedtls/ dir.

It's shipped with mbed TLS, it's in `modules/crypto/mbedtls/include/` (looking from the west root directory).