zephyrproject-rtos/zephyr #42774: pinctrl-0 issue in device tree building

zzzh, 2022-02-14:
**Describe the bug**
During the west build process, it says: 
devicetree error: 'pinctrl-0' is marked as required in 'properties:' in /home/zhu/zephyrproject/zephyr/dts/bindings/serial/st,stm32-usart.yaml, but does not appear in <Node /soc/serial@40004400 in '/home/zhu/zephyrproject/zephyr/misc/empty_file.c'>

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - Arm
 - What have you tried to diagnose or workaround this issue?
 - Changed usart1 part in dts file, form:
 - &usart1 {
    current-speed = <115200>;
    status = "okay";
};
To:
&usart1 {
    current-speed = <115200>;
    pinctrl-names = "default";
    pinctrl-0 = <&usart1_tx_pa9 &usart1_rx_pa10>;
    status = "okay";
};
got same result.

**To Reproduce**
Steps to reproduce the behavior:
1. the dts file works in zephyr-2.3.
2. but not on zephyr-2.7 or zephyr3.0-rc3
3. west build would bring this error
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Pass the building.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
Don't work.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)
devicetree error: 'pinctrl-0' is marked as required in 'properties:' in /home/zhu/zephyrproject/zephyr/dts/bindings/serial/st,stm32-usart.yaml, but does not appear in <Node /soc/serial@40004400 in '/home/zhu/zephyrproject/zephyr/misc/empty_file.c'>

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)   Linux
 - Toolchain (e.g Zephyr SDK, ...)   Zephyr SDK
 - Commit SHA or Version used   zephyr-2.7, zephyr-3.0-rc3

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

gmarull, 2022-02-14:
Are you including the file where `&usart1_tx_pa9 &usart1_rx_pa10` are defined?

FRASTM, 2022-02-14:
@zzzh  Depending on the board and the soc of that board there is a  (ex of nucleo_l476rg)
  `#include <st/l4/stm32l476r(c-e-g)tx-pinctrl.dtsi>`
This dtsi refers to the ../modules/hal/stm32/dts/st/l4/stm32l476r(c-e-g)tx-pinctrl.dtsi
that contains the pin definition usart1_tx_pa9 and usart1_rx_pa10
the pins are not defined. 
Second they must be free (not conflicting with another pin function) on the board. For example, in the same board config, pa10 is also the one  for  tim1_ch3_pa10

zzzh, 2022-02-15:
> @zzzh Depending on the board and the soc of that board there is a (ex of nucleo_l476rg) `#include <st/l4/stm32l476r(c-e-g)tx-pinctrl.dtsi>` This dtsi refers to the [...]

@FRASTM For me, it's "#include <st/f0/stm32f030c8tx-pinctrl.dtsi>", and I'm sure they are free (not conflicting with another pin function) on the board.

zzzh, 2022-02-15:
@gmarull @FRASTM :

Can it because they are already defined in pinmux.c?  maybe pinmux.c way had been obsoleted?

FRASTM, 2022-02-15:
The pinctrl had changes since the zephyr 2.3 
In the current zephyr version (Zephyr version: 3.0.0-rc3) the nucleo_f030r8.dts already includes the pins for usart1 :
```
&usart1 {
	pinctrl-0 = <&usart1_tx_pa9 &usart1_rx_pa10>;
	pinctrl-names = "default";
	current-speed = <115200>;
	status = "okay";
};
```
 That's the way to define the node.

zzzh, 2022-02-16:
> The pinctrl had changes since the zephyr 2.3 In the current zephyr version (Zephyr version: 3.0.0-rc3) the nucleo_f030r8.dts already includes the pins for usart1 :
>
> ```
> &usart1 {
> 	pinctrl-0 = [...]

@FRASTM  Thank you.
So, is pinmux.c still useful?  what about CONFIG_PINMUX?

zzzh, 2022-02-16:
> The pinctrl had changes since the zephyr 2.3 In the current zephyr version (Zephyr version: 3.0.0-rc3) the nucleo_f030r8.dts already includes the pins for usart1 :
>
> ```
> &usart1 {
> 	pinctrl-0 = [...]

Is there any document about the change from using pinmux.c to pinctrl way?

carlescufi, 2022-03-08:
> Is there any document about the change from using pinmux.c to pinctrl way?

https://docs.zephyrproject.org/latest/guides/pinctrl/index.html

closed 2022-03-08 by carlescufi