zephyrproject-rtos/zephyr #42800: gptp_mi neighbor_prop_delay is not included in sync_receipt_time calculation due cast from double to uint64_t

bibibibi89, 2022-02-14:
**Describe the bug**
in gptp_mi_clk_slave_sync_compute function, both pss->rate_ratio and nport_ds->neighbor_rate_ratio are double type but sync_receipt_time is uint64_t.

If pss->rate_ratio is less than 1 or port_ds->neighbor_rate_ratio is less than 1, sync_receipt_time will be 0 after first 2 statements. Then port_ds->neighbor_prop_delay won't be included in sync_receipt_time. 

sync_receipt_time = pss->rate_ratio;
sync_receipt_time /= port_ds->neighbor_rate_ratio;
sync_receipt_time *= port_ds->neighbor_prop_delay;

https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/ethernet/gptp/gptp_mi.c#L709-L710

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
global_ds->sync_receipt_time does not include neighbor_prop_delay but global_ds->sync_receipt_local_time does include neighbor_prop_delay properly then calculated time difference between GM clock and local clock is not correct. Sometimes, we observed time difference equals to neighbor_prop_delay.

jukkar, 2022-02-15:
Hi @bibibibi89, would you be able to send a fix for this issue?

bibibibi89, 2022-02-16:
I will creat a fix.

bibibibi89, 2022-03-02:
Will do. Now I know the PR process better.

rlubos, 2022-03-04:
Thank you @bibibibi89 for the PR :+1:

PR #43421 (2022-03-07): net: gptp: Fix sync_receipt_time calculation in gptp_mi
Fix #42800
Both pss->rate_ratio and port_ds->neighbor_rate_ratio are double type but sync_receipt_time is uint64_t. If pss->rate_ratio is less than 1 or sync_receipt_time * port_ds->neighbor_rate_ratio is less than 1, sync_receipt_time becomes 0 due to double to uint64_t cast.
commit 818d90ef (2022-03-07): net: gptp: Fix sync_receipt_time calculation in gptp_mi

Fix #42800
Both pss->rate_ratio and port_ds->neighbor_rate_ratio are double type
but sync_receipt_time is uint64_t. If pss->rate_ratio is less than 1
or sync_receipt_time * port_ds->neighbor_rate_ratio is less than 1,
sync_receipt_time becomes 0 due to double to uint64_t cast.
Assign port_ds->neighbor_prop_delay to sync_receipt_time first to fix
this issue.
closed 2022-03-07 by carlescufi