zephyrproject-rtos/zephyr #42805: nrfx uart/uarte driver: allow non-standard baudrates?

jonnykl, 2022-02-15:
**Is your enhancement proposal related to a problem? Please describe.**
I have to use non-standard baudrates like 10400. This is not supported by the nrfx uart/uarte drivers.

**Describe the solution you'd like**
The driver implements setting the baudrate based on the values given in the datasheet for the nRF52840 SOC. But in https://devzone.nordicsemi.com/f/nordic-q-a/391/uart-baudrate-register-values a simplified formula to calculate the baudrate register value for non-standard baudrates is given. The values calculated by the formula are not exact and can differ from the values given in the datasheet (e.g. baud 14400: datasheet value = `0x0x003AF000`, formula value = `0x003AF000`). The function `baudrate_set` of the driver could calculcate the baudrate based on the formula only if a non-standard baudrate is given (i.e. in the `default` case, instead of returning `-EINVAL`).

spam-receiver, 2023-03-04:
You can try to add your own baudrate by inserting new parameters in the folowing files:

zephyr/drivers/serial/uart_nrfx_uart.c
zephyr/drivers/serial/uart_nrfx_uarte.c

Into the function «static int baudrate_set(const struct device *dev, uint32_t baudrate)»:

```
case 10400:
	/* value not supported by Nordic HAL */
	nrf_baudrate = 0x002AA000;
	break;
```
		
P.S.:  Actual baudrate will be 10406.494140625

spam-receiver, 2023-03-04:
The manual contains inaccurate values of the BAUDRATE register, because there is no formula for calculating the baudrate directly in the manual itself. Apparently the employees made mistakes when calculating manually. See for yourself what's going on in the «nRF51 Series Reference Manual Version 3.0.1»:

```
Value Id    Value       Description
Baud1200    0x0004F000  1200 baud (actual rate: 1205)
Baud2400    0x0009D000  2400 baud (actual rate: 2396)
Baud4800    0x0013B000  4800 baud (actual rate: 4808)
Baud9600    0x00275000  9600 baud (actual rate: 9598)
Baud14400   0x003B0000  14400 baud (actual rate: 14414)
Baud19200   0x004EA000  19200 baud (actual rate: 19208)
Baud28800   0x0075F000  28800 baud (actual rate: 28829)
Baud31250   0x00800000  31250 baud
Baud38400   0x009D5000  38400 baud (actual rate: 38462)
Baud56000   0x00E50000  56000 baud (actual rate: 55944)
Baud57600   0x00EBF000  57600 baud (actual rate: 57602)
Baud76800   0x013A9000  76800 baud (actual rate: 76797)
Baud115200  0x01D7E000  115200 baud (actual rate: 115204)
Baud230400  0x03AFB000  230400 baud (actual rate: 230393)
Baud250000  0x04000000  250000 baud
Baud460800  0x075F7000  460800 baud
Baud921600  0x0EBEDFA4  921600 baud (actual rate: 921585)
Baud1M      0x10000000  1Mega baud
```

spam-receiver, 2023-03-04:
I think the correct formula is:

baudrate_reg_val = (baudrate * 2^32 / 16000000) & 0xFFFFF000

Then check actual baudrate:

baudrate = baudrate_reg_val * 16000000 / 2^32

Add 0x1000 to baudrate_reg_val and check actual baudrate again. Use the closest value.