zephyrproject-rtos/zephyr #42817: ADC on ST Nucleo H743ZI board with DMA

sachinjairam, 2022-02-15:
Hai, 
I am working with ST Nucleo H743ZI board for analog-to-digital conversion without CPU intervention between samples, by using DMA in Zephyr platform. Is there any ready-made example for the above, that I can refer?

Thank you.
Sachin

FRASTM, 2022-02-15:
Unfortunately not. 
The DMA transfer is used for SPI or UART on the stm32 mcus  running zephyr OS.

FRASTM, 2022-02-15:
The only source could be a stm32CUBE example   [ADC_DMA_Transfer](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/NUCLEO-H743ZI/Examples/ADC/ADC_DMA_Transfer) running on the nucleo_h743zi, using the HAL layer.

sachinjairam, 2022-02-16:
Hai @FRASTM thanks for your reply.

sachinjairam, 2022-02-16:
Hai @FRASTM , thanks for your reply. But how can I use HAL layer code on zephyr?, what change I need to make on files like CMakefiles.txt, prj.conf etc.?

FRASTM, 2022-02-16:
@pdietl said on [discord]( https://discord.com/channels/720317445772017664/883445358821249084/943263618047283240 ) that he   "is curently implementing the stm32 ADC with DMA"

pdietl, 2022-02-16:
Yeah let me push what I have this far.

sachinjairam, 2022-02-18:
> The only source could be a stm32CUBE example [ADC_DMA_Transfer](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/NUCLEO-H743ZI/Examples/ADC/ADC_DMA_Transfer) running on the nucleo_h743zi, using the HAL layer.

Hai @FRASTM , I have tried HAL layer code with zephyr and is built successfully. But the conversion is not happening. The functions `HAL_ADCEx_Calibration_Start` and `HAL_ADC_Start_DMA` returns error.
I can't figure out any solution for that. 
Is there any tutorial or something like that, I can refer about using HAL layer code with Zephyr?

Thank you.
Sachin

sachinjairam, 2022-02-22:
@pdietl Is it possible to implement STM32 ADC with DMA on zephyr? If so, is there any sample code that I can refer?

Thank you.
Sachin

erwango, 2022-02-23:
> Hai @FRASTM , thanks for your reply. But how can I use HAL layer code on zephyr?, what change I need to make on files like CMakefiles.txt, prj.conf etc.?

Cf https://github.com/zephyrproject-rtos/hal_stm32#use-stm32cube-in-your-application

> @pdietl Is it possible to implement STM32 ADC with DMA on zephyr? If so, is there any sample code that I can refer?

This is not available today. @pdietl is working on it but availability date of the code upstream is unknown so far.

sachinjairam, 2022-03-02:
@erwango Thanks for your reply. 
Let me try.

closed 2022-03-02 by sachinjairam