zephyrproject-rtos/zephyr #42825: MQTT client disconnection (EAGAIN) on publish with big payload

nvincent-vossloh, 2022-02-15:
I have an MQTT thread handling all incoming and outgoing messages. When other threads want  to publish a message they store a `struct mqtt_publish_param` element into a `msgq` which is read by the mqtt thread and then publish with `mqtt_publish`.

For small payloads sent once (4kB), everything is fine. If the same payload is re-sent immediately the client is disconnected during `mqtt_publish` with the error `-11, EAGAIN`.
If I send a big payload directy (8kB) the client is disconnected directly with the same error.
However, if I bypass the MQTT thread and call directly `mqtt_publish` with 32kB of payload from another thread I do not have this disconnection.

I have been able to modify `samples/net/mqtt_publisher` and reproduce the issue.
I am using `nucleo_h743zi` board.

**To Reproduce**
1. Download [mqtt_disconnect.diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/8069262/mqtt_disconnect.diff.txt) into zephyr tree
2. git apply mqtt_disconnect.diff.txt
3. west build -p auto -b nucleo_h743zi zephyr/samples/net/mqtt_publisher/
4. west flash 
5. See disconnection in logs

**Expected behavior**
No `EAGAIN ` error no matter the size given the fact that when calling directly `mqtt_publish` I do not have any disconnection.

**Impact**
MQTT client disconnection

**Logs and console output**
```
*** Booting Zephyr OS build v3.0.0-rc3-7-g10da718fd7f2  ***
[00:00:00.007,000] <inf> net_config: Initializing network
[00:00:00.007,000] <inf> net_config: Waiting interface 1 (0x24000bcc) to be up...
[00:00:01.501,000] <inf> net_config: Interface 1 (0x24000bcc) coming up
[00:00:01.501,000] <inf> net_config: IPv4 address: 10.42.0.2
[00:00:01.505,000] <inf> net_mqtt_publisher_sample: attempting to connect:
[00:00:01.509,000] <inf> net_mqtt_publisher_sample: MQTT client connected!
[00:00:01.509,000] <inf> net_mqtt_publisher_sample: try_to_connect: 0 <OK>
[00:00:01.509,000] <inf> net_mqtt_publisher_sample: mqtt_ping: 0 <OK>
[00:00:01.511,000] <inf> net_mqtt_publisher_sample: PINGRESP packet
[00:00:01.606,000] <inf> net_mqtt_publisher_sample: Send 8192 bytes to mqtt thread
[00:00:01.606,000] <inf> net_mqtt_publisher_sample: Bypass mqtt thread for 32768 bytes
[00:00:01.641,000] <inf> net_mqtt_publisher_sample: Bypass mqtt thread for 32768 bytes
[00:00:01.676,000] <inf> net_mqtt_publisher_sample: Send 2048 bytes to mqtt thread
[00:00:01.676,000] <inf> net_mqtt_publisher_sample: Send 2048 bytes to mqtt thread
[00:00:01.676,000] <inf> net_mqtt_publisher_sample: Send 2048 bytes to mqtt thread
[00:00:01.676,000] <inf> net_mqtt_publisher_sample: Send 4096 bytes to mqtt thread
[00:00:01.676,000] <inf> net_mqtt_publisher_sample: Bypass mqtt thread for 32768 bytes
[00:00:02.009,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:02.510,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:02.511,000] <inf> net_mqtt_publisher_sample: PUBACK packet id: 59784
[00:00:03.011,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:03.013,000] <inf> net_mqtt_publisher_sample: PUBREC packet id: 1590
[00:00:03.014,000] <inf> net_mqtt_publisher_sample: PUBCOMP packet id: 1590
[00:00:03.511,000] <inf> net_mqtt_publisher_sample: Publish on foo/bar 8192 bytes
[00:00:03.515,000] <inf> net_mqtt_publisher_sample: MQTT client disconnected -11
[00:00:03.515,000] <err> net_mqtt_publisher_sample: Failed to publish, error -11
[00:00:03.515,000] <inf> net_mqtt_publisher_sample: mqtt_disconnect: -128 <ERROR>
[00:00:03.515,000] <inf> net_mqtt_publisher_sample: Bye!
[00:00:08.515,000] <inf> net_mqtt_publisher_sample: attempting to connect:
[00:00:08.518,000] <inf> net_mqtt_publisher_sample: MQTT client connected!
[00:00:08.519,000] <inf> net_mqtt_publisher_sample: try_to_connect: 0 <OK>
[00:00:08.519,000] <inf> net_mqtt_publisher_sample: mqtt_ping: 0 <OK>
[00:00:08.520,000] <inf> net_mqtt_publisher_sample: PINGRESP packet
[00:00:09.020,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:09.521,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:09.522,000] <inf> net_mqtt_publisher_sample: PUBACK packet id: 3357
[00:00:10.022,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:10.023,000] <inf> net_mqtt_publisher_sample: PUBREC packet id: 23608
[00:00:10.024,000] <inf> net_mqtt_publisher_sample: PUBCOMP packet id: 23608
[00:00:10.522,000] <inf> net_mqtt_publisher_sample: Publish on foo/bar 2048 bytes
[00:00:10.524,000] <inf> net_mqtt_publisher_sample: Publish on foo/bar 2048 bytes
[00:00:10.525,000] <inf> net_mqtt_publisher_sample: MQTT client disconnected -11
[00:00:10.525,000] <err> net_mqtt_publisher_sample: Failed to publish, error -11
[00:00:10.526,000] <inf> net_mqtt_publisher_sample: mqtt_disconnect: -128 <ERROR>
[00:00:10.526,000] <inf> net_mqtt_publisher_sample: Bye!
[00:00:15.526,000] <inf> net_mqtt_publisher_sample: attempting to connect:
[00:00:15.529,000] <inf> net_mqtt_publisher_sample: MQTT client connected!
[00:00:15.529,000] <inf> net_mqtt_publisher_sample: try_to_connect: 0 <OK>
[00:00:15.529,000] <inf> net_mqtt_publisher_sample: mqtt_ping: 0 <OK>
[00:00:15.530,000] <inf> net_mqtt_publisher_sample: PINGRESP packet
[00:00:16.030,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:16.531,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:16.532,000] <inf> net_mqtt_publisher_sample: PUBACK packet id: 19448
[00:00:17.032,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:17.033,000] <inf> net_mqtt_publisher_sample: PUBREC packet id: 27875
[00:00:17.035,000] <inf> net_mqtt_publisher_sample: PUBCOMP packet id: 27875
[00:00:17.532,000] <inf> net_mqtt_publisher_sample: Publish on foo/bar 2048 bytes
[00:00:17.535,000] <inf> net_mqtt_publisher_sample: Publish on foo/bar 4096 bytes
[00:00:17.536,000] <inf> net_mqtt_publisher_sample: MQTT client disconnected -11
[00:00:17.536,000] <err> net_mqtt_publisher_sample: Failed to publish, error -11
[00:00:17.536,000] <inf> net_mqtt_publisher_sample: mqtt_disconnect: -128 <ERROR>
[00:00:17.536,000] <inf> net_mqtt_publisher_sample: Bye!
```

**Environment:**
 - ubuntu 20.0.4.3 LTS
 - zephyr-sdk-0.13.1
 - zephyr version: 10da718fd7f2dbcd5f11a1d38bb3be7eb1fd05bd

rlubos, 2022-02-16:
@nvincent-vossloh Please check https://github.com/zephyrproject-rtos/zephyr/pull/42867, I've submitted a fix for the EAGAIN problem.

nvincent-vossloh, 2022-02-16:
@rlubos, Thanks, I managed to run your fix against the testcase submitted in this issue and the mqtt client is not disconnected anymore.
:+1:

PR #42867 (2022-03-15): net: sockets: Retry net_context_sendmsg if EAGAIN is reported
TCP module can report EAGAIN in case TX window is full. This should not
be forwarded to the application, as blocking socket is not supposed to
return EAGAIN.

Fix this for sendmsg by implementing the same mechanism for handling TX
errors as for regular send/sendto operations.

Fixes #42825