zephyrproject-rtos/zephyr #42829: GATT: bt_gatt_is_subscribed does not work as expected when called from bt_conn_cb->connected

MariuszSkamra, 2022-02-15:
**Describe the bug**
CONFIG_BT_SETTINGS_CCC_LAZY_LOADING is set. 
bt_gatt_is_subscribed will always return false when called from bt_conn_cb->connected callback.

It is because bt_gatt_connected is called after all other registered callbacks were called. The change that introduced such behavior: 6bb75a53d12d3c7afd9dad628dfab40a2d79a089

GATT as the stack layer shall be ready when the application is notified about connection.

**To Reproduce**
1. Use peer device to connect and subscribe for notifications/indications.
2. Disconnect
3. Connect again and check the output of bt_gatt_is_subscribed called from bt_conn_cb->connected callback context.

**Expected behavior**
bt_gatt_is_subscribed will return true.

**Impact**
App does not know whether the device is subscribed unless it stores the value itself (it's not preferred solution since the value is stored in BT_SETTINGS).

alwa-nordic, 2022-02-24:
I agree. Rationale:
 - `CONFIG_BT_SETTINGS_CCC_LAZY_LOADING` should not change API behavior. "Lazy" means *on-demand*.
 -  My intuition says `bt_gatt_is_subscribed` could be a function of `bt_addr_le_t` instead of `bt_conn`. The subscription exist across connections.

Possible fixes:
 - On-demand load CCC settings when the CCC database is queried.
 - Revert #13473. The rationale behind that change was that an application wants to know where it is asked to send notifications to when it receives a `BT_GATT_CCC` `_changed` event. But the unfinished discussion in the that issue says to me that this is not an use case of the API. The `_changed` event asks the application start or stop notifications that are periodically broadcasted. (I.e. it should use `bt_gatt_notify(NULL, ...)`.

carlescufi, 2022-03-10:
@MariuszSkamra are you planning to send a PR to fix this?

MariuszSkamra, 2022-03-21:
@alwa-nordic I would go with the revert, since the `_changed` callback does not even carry information about the `bt_conn` nor `bt_addr_le_t` so that it does not matter whether it's called before or after connection cb.

PR #44015 (2022-03-22): Revert "Bluetooth: gatt: ccc changed cb after connection cb"
This reverts commit 6bb75a53d12d3c7afd9dad628dfab40a2d79a089.
This fixes GATT that was uninitalized when application received
bt_conn_cb->connected callback. As the result, the bt_gatt_is_subscribed
was not working as expected when called from bt_conn_cb->connected.

The _bt_gatt_ccc->cfg_changed callback does not carry information about
the device that subscribed for notifications but rather is says the
app when it should start/stop broadcasting notifications.

This leads to the conclusion that the bt_gatt_ccc->cfg_changed can be
ca [... cut]
commit d0111f7f (2022-03-22): Revert "Bluetooth: gatt: ccc changed cb after connection cb"

This reverts commit 6bb75a53d12d3c7afd9dad628dfab40a2d79a089.
This fixes GATT that was uninitalized when application received
bt_conn_cb->connected callback. As the result, the bt_gatt_is_subscribed
was not working as expected when called from bt_conn_cb->connected.

The _bt_gatt_ccc->cfg_changed callback does not carry information about
the device that subscribed for notifications but rather is says the
app when it should start/stop broadcasting notifications.

This leads to the conclusion that the bt_gatt_ccc->cfg_changed can be
c [... cut]
closed 2022-03-22 by carlescufi