zephyrproject-rtos/zephyr #42831: Do the atomic* functions require protection from optimization?

scapri, 2022-02-15:
With an integer variable modified by both an ISR and normal thread code, using atomic functions for access is a useful method to provide concurrent access.  But I have concerns about the build system optimizing out accesses.  Using a volatile keyword has been a method for preventing this in the past.  I was hoping that just using the atomic* functions would guarantee that the code would not be optimized out, but this kernel.org web page is a bit of a concern:

https://www.kernel.org/doc/html/v4.12/core-api/atomic_ops.html 

Especially these 2 statements:
...users of atomic_t should treat atomic_read() and atomic_set() as simple C statements that may be reordered or optimized away entirely by the compiler or processor, and explicitly invoke the appropriate compiler and/or memory barrier for each use case

 The READ_ONCE() and WRITE_ONCE() macros should be used to prevent the compiler from using optimizations that might otherwise optimize accesses out of existence on the one hand, or that might create unsolicited accesses on the other.

I don't see a READ/WRITE_ONCE() available in Zephyr, so I tried adding a volatile qualifier to the variable, but then I get compiler warnings (understandable, since the atomic functions aren’t expecting a volatile parameter, but this makes me assume this is an incorrect strategy).

So the overall question is:
What is the proper way to define an atomic variable so that it will not be optimized away in Zephyr?

I have attached some pseudocode here for discussion:

    #define ORIG_VAL 100
    #define HALF_FULL (ORIG_VAL/2)
    #define MAX_TOP_UP 10

    volatile atomic_t new_data_len;

    static void gpio_glass_isr(const struct device *port,
                               struct gpio_callback *cb,
                               uint32_t pins)
    {
        int periph_data;

        periph_data = read_data_from_glass_peripheral();
        atomic_add(&new_data_len, periph_data);
    }

    void main(void)
    {
        int value, newvalue;
    
        // NOTE, here is where the device_get_binding, gpio_pin_configure,
        // gpio_pin_interrupt_configure, gpio_init_callback, and gpio_callback,
        // etc., occur.  I didn't add it here, and please assume it will result in
        // the running of the the gpio_glass_isr above.

        atomic_set(&new_data_len, ORIG_VAL);

        while(1)
        {
            value = atomic_get(&new_data_len);
            if (value <= HALF_FULL)
            {
                printf("Filling up the glass, value was %d\n", value);
                newvalue = atomic_add(&new_data_len, MAX_TOP_UP);
                if (value != newvalue)
                {
                    printf("change of %d since last read\n", value - newvalue);
                }
            }
            k_msleep(10 * 1000);
        }
    }

carlescufi, 2022-02-15:
@carlocaione @andyross @peter-mitsis @dcpleung @nordic-krch input welcome on this one

andyross, 2022-02-15:
So... indeed, compiler optimizers have a history of messing with this stuff, different OSes have picked different strategies to defeat that behavior, and.. yeah, Zephyr should probably be more rigorous here than it is.

But I don't see any possibility for misoptimization in the code you submitted.  Code gets optimized away when the compiler can prove with certainty that it will have no detectable effect at runtime.  Any write to a global variable (like new_data_len) or any action taken based on its value is non-removable by definition, simply because the compiler doesn't know what the existing memory contents are.  Even modification of a static variable is non-removable unless the compiler can prove that all sequences of changes to it result in a zero that matches its initialization condition (so if you have two entry points in your file that can both look at the value, it's protected).

Basically: you have to work very, very hard to produce a misoptimization bug, and when it happens it's generally because your code wasn't correctly synchronized to begin with.

The other stuff talked about in the linked Linux docs, whether or not atomic access involve memory barriers, is something we **don't** document or do correctly yet though.  We're only now getting our first implementation of a barrier API, having merged our first out of order architecture (arm64) very recently.

scapri, 2022-02-15:
I agree that in a path that modifies the variable, the read won't be removed.  I may have simplified the code sample too much, but I was trying to imply that there are paths in which the variable is never modified.  Consider the situation of a thread polling on a variable, sleeping (waiting on a kernel object) in between reads, never writing the variable.  I would like to know in that situation that I am guaranteed that the call to atomic_get() is never optimized out.

andyross, 2022-02-15:
>  Consider the situation of a thread polling on a variable, sleeping (waiting on a kernel object) in between reads, never writing the variable. I would like to know in that situation that I am guaranteed that the call to atomic_get() is never optimized out.

If the variable is potentially shared with other context, then the compiler will be unable to prove it's unused.  If it's literally a local stack variable never passed via pointer to another context and only used within one function, then maybe.  But then that would probably be correct as it wouldn't be providing any synchronization.

carlescufi, 2022-02-21:
I think this is a good candidate to discuss in the [dev-review meeting this week](https://lists.zephyrproject.org/g/devel/viewevent?repeatid=14985&eventid=1326093&calstart=2022-02-24).

MaureenHelm, 2022-02-23:
> I think this is a good candidate to discuss in the [dev-review meeting this week](https://lists.zephyrproject.org/g/devel/viewevent?repeatid=14985&eventid=1326093&calstart=2022-02-24).

I need to cancel this week's meeting due to a conflict, but I'll put it on the agenda for next week.

andyross, 2022-03-03:
Decision from the dev meeting was that we should put a formal specification about memory ordering and compiler/optimizer barriers into the atomics and arch subsystem documentation to call out explicitly that Zephyr atomics require memory serialization.  I'll write something up.

PR #43559 (2022-03-09): include/atomics: doc: Clarify memory barrier behavior
Zephyr is migrating into hardware where memory ordering is starting to
matter.  The existing gcc-based defaults have always been specifying
__ATOMIC_SEQ_CST, which is safe and correct.  And all the
arch-specific assembly we have currently are either safe or for
platforms where barriers aren't needed.

Discussion in #42831 made the case that it would be nice if we were to
formally promise this, and require it from future implementations.  So
let's do that.

Fixes: #42831
commit 24d4beca (2022-03-09): include/atomics: doc: Clarify memory barrier behavior

Zephyr is migrating into hardware where memory ordering is starting to
matter.  The existing gcc-based defaults have always been specifying
__ATOMIC_SEQ_CST, which is safe and correct.  And all the
arch-specific assembly we have currently are either safe or for
platforms where barriers aren't needed.

Discussion in #42831 made the case that it would be nice if we were to
formally promise this, and require it from future implementations.  So
let's do that.

Fixes: #42831
closed 2022-03-09 by nashif