zephyrproject-rtos/zephyr #42848: it8xxx2_evb: Test in /tests/subsys/cpp/libcxx is failing.

fuhaolei, 2022-02-16:
**Describe the bug**
cpp.libcxx : cpp.libcxx.newlib.exception

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyrproject
2. ./zephyr/scripts/twister -p it8xxx2_evb --device-testing -T zephyr/tests/subsys/cpp/libcxx--west-flash="../itetool/loader.sh" --device-serial=/dev/ttyUSB0
3. See error

**Expected behavior**
The test should pass, but it's stuck.

**Logs and console output**
> ***** delaying boot 1ms (per build configuration) *****
*** Booting Zephyr OS build v3.0.0-rc3-2-g8889114e3ca9  (delayed boot 1ms) ***
version 201703
Running test suite libcxx_tests
===================================================================
START - test_array
PASS - test_array in 0.1 seconds
===================================================================
START - test_vector
PASS - test_vector in 0.1 seconds
===================================================================
START - test_make_unique
PASS - test_make_unique in 0.1 seconds
===================================================================
START - test_exception
exit

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain zephyr-sdk-0.13.2
 - Commit 8889114e3ca9

Dino-Li, 2022-04-06:
@fuhaolei,
We can not build the test on the top of repository:
fatal error: array: No such file or directory
    7 | #include <array>
Can you build the test without error in your environment?

fuhaolei, 2022-04-06:
@Dino-Li ,We also meet this error. The reason is that some definitions are missing. It passed after these definitions were added.

fuhaolei, 2022-04-06:
This test passed on it81302 board. And because we no longer use the it8320 board. So this issue should be closed.