zephyrproject-rtos/zephyr #42851: it8xxx2_evb: Mutlitple tests in tests/kernel/contex are failing.

fuhaolei, 2022-02-16:
**Describe the bug**
kernel.common : kernel.common.kernel_cpu_idle_atomic
kernel.common : kernel.common.kernel_cpu_idle
kernel.common : kernel.common.k_yield
kernel.common : kernel.common.kernel_thread
kernel.common : kernel.common.kernel_timer_interrupts
kernel.common : kernel.common.stack_sentinel.fatal
kernel.common : kernel.common.timing.usleep

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyrproject
2. ./zephyr/scripts/twister -p it8xxx2_evb --device-testing -T zephyr/tests/kernel/context--west-flash="../itetool/loader.sh" --device-serial=/dev/ttyUSB0
3. See error

**Expected behavior**
The test should pass, but it's stuck.

**Logs and console output**
> *** Booting Zephyr OS build v3.0.0-rc3-6-gd517947de4e1  (delayed boot 1ms) ***
Running test suite context
===================================================================
START - test_kernel_interrupts
PASS - test_kernel_interrupts in 0.1 seconds
===================================================================
START - test_kernel_ctx_thread
Testing k_current_get() from an ISR and thread
Testing k_is_in_isr() from an ISR
Testing k_is_in_isr() from a preemptible thread
PASS - test_kernel_ctx_thread in 0.12 seconds
===================================================================
START - test_busy_wait
Thread busy waiting for 20000 usecs
Thread busy waiting completed
Thread busy waiting for 20000 usecs (irqs locked)
Thread busy waiting completed (irqs locked)
PASS - test_busy_wait in 0.55 seconds
===================================================================
START - test_k_sleep
thread sleeping for 50 milliseconds
thread back from sleep
Testing k_thread_create() without cancellation
thread (q order: 2, t/o: 500) is running
got thread (q order: 2, t/o: 500) as expected
thread (q order: 3, t/o: 750) is running
got thread (q order: 3, t/o: 750) as expected
thread (q order: 0, t/o: 1000) is running
got thread (q order: 0, t/o: 1000) as expected
thread (q order: 6, t/o: 1250) is running
got thread (q order: 6, t/o: 1250) as expected
thread (q order: 1, t/o: 1500) is running
got thread (q order: 1, t/o: 1500) as expected
thread (q order: 4, t/o: 1750) is running
got thread (q order: 4, t/o: 1750) as expected
thread (q order: 5, t/o: 2000) is running
got thread (q order: 5, t/o: 2000) as expected
Testing k_thread_create() with cancellations
cancelling [q order: 0, t/o: 1000, t/o order: 0]
thread (q order: 3, t/o: 750) is running
got (q order: 3, t/o: 750, t/o order 1) as expected
thread (q order: 0, t/o: 1000) is running
got (q order: 0, t/o: 1000, t/o order 2) as expected
cancelling [q order: 3, t/o: 750, t/o order: 3]
cancelling [q order: 4, t/o: 1750, t/o order: 4]
thread (q order: 4, t/o: 1750) is running
got (q order: 4, t/o: 1750, t/o order 5) as expected
cancelling [q order: 6, t/o: 1250, t/o order: 6]
PASS - test_k_sleep in 5.339 seconds
===================================================================
START - test_kernel_cpu_idle_atomic

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain zephyr-sdk-0.13.2
 - Commit 8889114e3ca9

Dino-Li, 2022-03-01:
@fuhaolei , @Zhaoningx , the PR fixed this issue.

fuhaolei, 2022-03-02:
@Dino-Li , the last test case (kernel.common : kernel.common.timing.usleep) still not work. And I find this test case is located in folder tests/kernel/sleep that different with others.

PR #43299 (2022-03-07): it8xxx2: timer: to unified method of enabling event timer.
This enable timer and also reset it.

fixes #42851
fixes #42953
commit d8c2c2cb (2022-03-07): it8xxx2: timer: to unified method of enabling event timer.

This enable timer and also reset it.

fixes #42851
fixes #42953
closed 2022-03-07 by carlescufi