zephyrproject-rtos/zephyr #42854: k_busy_wait() never returns when called - litex vexriscv soc and cpu on xilinx ac701 board

Aravindh-Swaminathan, 2022-02-16:
**Describe the bug**
k_busy_wait() never returns when called.

Hardware details
Board - Xilinx AC701
SoC - Litex
CPU - Vexriscv

**To Reproduce**
Steps to reproduce the behavior:
1) Build litex: python3 ./xilinx_ac701.py --build --csr-csv test/csr.csv --csr-json test/csr.json

2) Generate dts overlay: litex/litex/tools/litex_json2dts_zephyr.py --dts overlay.dts --config overlay.config csr.json

3) Build sample code: west build -b litex_vexriscv samples/synchronization --pristine -DDTC_OVERLAY_FILE=/Users/user/project/litex/litex-boards/litex_boards/targets/test/overlay.dts -DCONFIG_UART_LITEUART=y -DCONFIG_LITEX_TIMER=y -DCONFIG_ETH_LITEETH=n -DCONFIG_SPI_LITESPI=n -DCONFIG_I2C_LITEX=n

4) Load kernel onto platform: sudo lxterm /dev/ttyUSB2 --kernel zephyr_multi_thread.bin --kernel-adr 0x40000000

5) Load bitstream: sudo openFPGALoader -c digilent xilinx_ac701.bit

**Expected behavior**
Hello world printed alternatively on to console by thread a and thread b

**Impact**
High

**Logs and console output**
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs
Running on AC701 FPGA Development Board
Added Feature: Ethernet, SDCard

 BIOS CRC passed (ebbfb352)

 Migen git sha1: ac70301
 LiteX git sha1: dff33885

--=============== SoC ==================--
CPU:            VexRiscv @ 100MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            8-bit data
ROM:            62KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          1048576KiB 16-bit @ 800MT/s (CL-7 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00111111111111100000000000000000| delays: 08+-06
  m0, b04: |00000000000000000111111111111100| delays: 23+-06
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b04 delays: 23+-06
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |01111111111111100000000000000000| delays: 07+-07
  m1, b04: |00000000000000000111111111111110| delays: 24+-07
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b04 delays: 24+-07
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 27.2MiB/s
   Read speed: 27.3MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading zephyr_multi_thread.bin to 0x40000000 (114868 bytes)...
[LXTERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LXTERM] Upload complete (9.9KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--
*** Booting Zephyr OS build v2.7.99-2698-gf92542ecf3e4  ***
thread_a: Hello World from cpu 0 on litex_vexriscv!

```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr

**Additional context**
comment out the line number 67 k_busy_wait(1000); in the main.c file of zephyr/samples/synchronization/src/
rebuild the code and run the bin file on hardware again. The application works as expected. 

**Logs**

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs
Running on AC701 FPGA Development Board
Added Feature: Ethernet, SDCard

 BIOS CRC passed (ebbfb352)

 Migen git sha1: ac70301
 LiteX git sha1: dff33885

--=============== SoC ==================--
CPU:            VexRiscv @ 100MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            8-bit data
ROM:            62KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          1048576KiB 16-bit @ 800MT/s (CL-7 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00111111111111100000000000000000| delays: 08+-06
  m0, b04: |00000000000000000111111111111100| delays: 23+-06
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b04 delays: 23+-06
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |01111111111111100000000000000000| delays: 08+-07
  m1, b04: |00000000000000000111111111111110| delays: 24+-07
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b04 delays: 24+-07
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 27.2MiB/s
   Read speed: 27.3MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading zephyr_multi_thread.bin to 0x40000000 (114756 bytes)...
[LXTERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LXTERM] Upload complete (9.9KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--
*** Booting Zephyr OS build v2.7.99-2698-gf92542ecf3e4  ***
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
thread_b: Hello World from cpu 0 on litex_vexriscv!
thread_a: Hello World from cpu 0 on litex_vexriscv!
```

Aravindh-Swaminathan, 2022-02-16:
https://github.com/zephyrproject-rtos/zephyr/issues/23622

fkokosinski, 2022-04-28:
Hi! Thanks for reporting this issue, we're looking into it.

fkokosinski, 2022-05-18:
Closing this one as #45196 and #45198 which fix this issue had been merged.

closed 2022-05-18 by fkokosinski