zephyrproject-rtos/zephyr #42857: sam_e70b_xplained: failed to run test cases tests/net/npf and tests/net/bridge

Zhaoningx, 2022-02-16:
**Describe the bug**
Failed to run test cases tests/net/npf and tests/net/bridge on sam_e70b_xplained platform.

**To Reproduce**
twister -p sam_e70b_xplained --device-testing --device-serial /dev/ttyACM0 -T tests/net/ -v
see error:
**tests/net/bridge/net.eth_bridge.ip/handler.log**
I: Device I2C_2 initialized
I: Device I2C_0 initialized
E: transfer timedout MDIO
E: transfer timedout MDIO
E: transfer timedout MDIO
E: transfer timedout MDIO
I: MAC: fc:c2:3d:0d:5d:8b
E: Failed to reserve data net buffers
E: Unable to initialize ETH queue0
W: You have 1 IPv6 net_if addresses but 4 network interfaces
W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.
W: You have 1 IPv4 net_if addresses but 4 network interfaces
W: Consider increasing CONFIG_NET_IF_MAX_IPV4_COUNT value.
I: PHY (0) Link speed 100 Mb, full duplex
**tests/net/npf/net.pkt_filter/handler.log**
I: Device I2C_2 initialized
I: Device I2C_0 initialized
E: transfer timedout MDIO
E: transfer timedout MDIO
E: transfer timedout MDIO
E: transfer timedout MDIO
I: MAC: fc:c2:3d:0d:5d:8b
E: Failed to reserve data net buffers
E: Unable to initialize ETH queue0
E: Iface 0x20400cac driver API init NULL
E: Iface 0x20400cb4 driver API init NULL
W: You have 1 IPv6 net_if addresses but 3 network interfaces
W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.
I: PHY (0) Link speed 100 Mb, full duplex

**Environment**
zephyr SDK: 13.2
commit: ccf51e3cd3ae

nandojve, 2022-02-16:
CC @arvinf

yerabolu, 2022-03-15:
@nandojve : Any updates on this? still seeing the issue on commit:c31cb0075ad1

nandojve, 2022-03-17:
Hi @Zhaoningx , @yerabolu ,

Looking at this log I can see two important messages:

> E: Failed to reserve data net buffers
> ...
> W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.

Did you increased that values?

Maybe you can try with:

```diff
- CONFIG_NET_PKT_TX_COUNT=10
- CONFIG_NET_PKT_RX_COUNT=10
- CONFIG_NET_BUF_RX_COUNT=10
- CONFIG_NET_BUF_TX_COUNT=10
+ CONFIG_NET_PKT_TX_COUNT=20
+ CONFIG_NET_PKT_RX_COUNT=20
+ CONFIG_NET_BUF_RX_COUNT=20
+ CONFIG_NET_BUF_TX_COUNT=20
```

Zhaoningx, 2022-03-25:
Hi @nandojve, sorry for delay, after testing, the changes you mentioned above can fix this issue.

nandojve, 2022-03-25:
HI @Zhaoningx ,

Thank you for your feedback. Can you send a fix for those platforms?

Zhaoningx, 2022-03-28:
Hi @nandojve, sure, I will push a PR to fix this issue.

nandojve, 2022-03-30:
> Hi @nandojve, sure, I will push a PR to fix this issue. And could you tell me why it works when changing the number from 10 to 20? Thanks.

The memory buffers for ETH are statically allocated and not necessarily different platform will uses the same values. This adjust is normal at network subsystem in Zephyr.

commit 1bc003ad (2022-05-26): tests: net: Adjusting the memory buffer for ETH

The net cases failed to run on sam_e70b_xplained
platform because failed to reserve data net buffers,so
adjusting the memory buffer to fix this.

Fixes #42857
closed 2022-05-26 by mbolivar-nordic