zephyrproject-rtos/zephyr #42860: Bluetooth: LE-Audio: Codec config data and meta data storage refactor

Casper-Bonde-Bose, 2022-02-16:
**Is your enhancement proposal related to a problem? Please describe.**
The current implementation of codec config and meta data is based on copying the LTV encoded data into an array of type bt_codec_data. This approach was appropriate at the time where all defined data types were 4 bytes or less in size as the overhead of having pre-allocated memory for all possible values were not huge.
But as more and more metadata types and potentially vendor specific codec config values are added and needs to be processed this approach is no longer suitable. One example is the metadata type 0x03 (program_info) this is a UTF-8 string - hence even for a small supported size of say 20 bytes the overhead of re-configuring the bt_codec_data type to be able to hold this amount of data is huge as all entries have the same size - hence you would also use a 20-byte entry to store a single byte value.

**Describe the solution you'd like**
I'll suggest to keep the LTV encoded data in a flat memory buffer and use utility functions to extract data as needed (you would need this anyway if using bt_codec_data[]).
One solution could be to treat CC-data and meta-data as adv-data is handled today. The data is provided to the application while still in the net_buf in which it was received and lives only while processing the call-backs after which it is deleted from the stack.
A link from the stream to the codec config and metadata could be done in several ways:

1. the application could have sole control of it - and the stack can never access it. (this could complicate future integration into an audio-subsystem.
2. The application could save a copy of relevant TLV encoded CC- and meta-data and the stack could have an api to register this data to a stream (like bt_audio_capability_register() does for PAC records). I would prefer a storage type that do not force the application to store the data in RAM as is the case for PAC records due to the presence of sys_snode_t node; in the data type.)

**Describe alternatives you've considered**
I considered having a flat shared block of memory in the stream structure and use that for CC- and meta-data directly. That way you would not be limited by a size limit for the largest element, but rather the total amount of data - hence if you have many small entries and one large memory usage would be much less. But I'm not sure there is a good MAX_SIZE for this - it could of cause be user-configurable as it is today, but it would prevent an application to use dynamic memory allocation - which allows to share memory between multiple streams that might have different amount of metadata - for example the larger program_info Utf-8 string only applies to broadcast streams - hence no need to allocate memory for it for all unicast streams.

bssc2g, 2022-10-24:
Is this issue still being looked at? The codec configuration is not compliant with the BAP spec. Subgroup codec configs (Level 2) need to be inherited into BIS-specific codec configs (Level 3). I agree that it is probably necessary and best to use the flat LTV array to encode information, but the utility functions, as well as the BASE parsing logic for codec info will need to verify consistency and compliance with the BAP spec, wrt codec config inheritance. 

An alternate approach would be to duplicate a codec struct across each BIS and "externally" ensure consistency between all BISs within a subgroup.

Thalley, 2022-10-24:
> Is this issue still being looked at? The codec configuration is not compliant with the BAP spec. Subgroup codec configs (Level 2) need to be inherited into BIS-specific codec configs (Level 3). I [...]

Hi @bssc2g, I don't think this issue is being looked at by anyone. W.r.t. your comment about subgroups and the codec configurations for the broadcast sink and source, we have the following PR awaiting review: https://github.com/zephyrproject-rtos/zephyr/pull/49946

bssc2g, 2022-10-24:
Thanks for the update @Thalley, glad to see it's in progress! Great work on the existing framework, btw. This was only an issue when we tested against a non-Zephyr broadcaster that was putting additional channel config info in the BIS codec config.

Thalley, 2023-04-14:
As part of this refactoring, we should also consider splitting codec configuration and codec capabilities into 2 separate structs, as the current design is confusing for users.