zephyrproject-rtos/zephyr #42865: openocd configurations missing for stm32mp157c_dk2 board

KalpeshKadam-eaton, 2022-02-16:
Board = stm32mp157c_dk2
Path = https://github.com/zephyrproject-rtos/zephyr/tree/main/boards/arm/stm32mp157c_dk2/support
File = openocd.cfg

Above openocd.cfg file does not have any configuration steps.

**To Reproduce**
Steps to reproduce the behavior:
1. open https://github.com/zephyrproject-rtos/zephyr/tree/main/boards/arm/stm32mp157c_dk2/support/openocd.cfg 

**Expected behavior**
openOCD Configuration steps should be mentioned here. We should able to debug M4 core in engineering mode using this configuration.

erwango, 2022-02-21:
@KalpeshKadam-eaton Please have a look at debug instructions provided here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/stm32mp157c_dk2/doc/stm32mp157_dk2.rst#debugging

arnopo, 2022-02-21:
Hello @KalpeshKadam-eaton 

What do you mean by **rpmsg_virtio_return_buffer**?  Is it the mode supported in stm32mp1Cube distribution to run the cortex-M4 is standalone?

KalpeshKadam-eaton, 2022-02-22:
@arnopo 
I did not mentioned around rpmsg_virtio_return_buffer. Is this question for me

closed 2022-02-22 by KalpeshKadam-eaton

arnopo, 2022-02-22:
> @arnopo I did not mentioned around rpmsg_virtio_return_buffer. Is this question for me

sorry copy past issue...
My question was "What do you mean by **engineering mode**? Is it the mode supported in stm32mp1Cube distribution to run the cortex-M4 is standalone?