zephyrproject-rtos/zephyr #42870: Build error due to minimal libc qsort callback cast

daor-oti, 2022-02-16:
**Describe the bug**
Some toolchains experience problems when casting a function pointer to another function pointer with different number of parameters. This is especially a problem when the cast is from a function pointer with fewer parameters than the function pointer being cast to. As is the case in the minimal libc stdlib.h function qsort.

Following error comes from arm-none-eabi-gcc version 10.3.1 20210824
> stdlib.h:71:29: error: cast between incompatible function types from 'int (*)(const void *, const void )' to 'int ()(const void *, const void *, void *)'

The error above will probably be very similar to what a proprietary toolchain could output.

A fix is proposed in [PR42384](https://github.com/zephyrproject-rtos/zephyr/pull/42384)

**To Reproduce**
The above error was from a zephyr decoupled build of the minimal libc, with the gnu arm compiler. This is only to show that gcc also does not like the cast.

**Expected behavior**
No error during build, even with proprietary toolchains, and thus no casts of this type should be present in the minimal libc.

**Impact**
This could potentially be a showstopper for those not using the zephyr sdk, and thus probably proprietary toolchains.

**Logs and console output**
None

**Environment (please complete the following information):**
 - Linux
 - Toolchain proprietary, but shown with gnu arm gcc version 10.3.1 20210824

**Additional context**
None

cfriedt, 2022-02-27:
Using the latest GNU ARM Embedded toolchain.
```
$ md5sum gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2
2383e4eb4ea23f248d33adc70dc3227e  gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2
```

I have not tested with earlier versions of the above toolchain, but given that 

In addition to the regular build options, I added `CONFIG_COMPILER_OPT="-Wall -Wextra"` to `tests/lib/c_lib/prj.conf`.

I'm unable to coerce the GNU ARM Embedded toolchain into producing a warning let alone an error with qsort. Using qsort quite regularly with a proprietary toolchain on riscv64 myself as well.

@daor-oti - are you able to provide the error message from the other, proprietary toolchain that you reference (sanitized if necessary)?

```
west build -p auto -b qemu_cortex_m3 -t run tests/lib/c_lib 2>&1 | grep -i qsort
Including boilerplate (Zephyr base): /home/cfriedt/workspace/zephyrproject/zephyr/cmake/app/boilerplate.cmake
[58/142] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/qsort.c.obj
[121/142] Building C object CMakeFiles/app.dir/src/test_qsort.c.obj
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling
START - test_qsort
 PASS - test_qsort in 0.6 seconds
 ```

daor-oti, 2022-03-01:
@cfriedt Thank you for your test, but unfortunately I cannot, and there is no need, as it is essentially the same as the one from gcc, which I listed in the description above. I think you are getting build warnings in you above example, so if you put -Werror, then you will probably see the same. I will look at your comment in the PR and rework asap, so that we can get the cast removed.

cfriedt, 2022-03-01:
> @cfriedt Thank you for your test, but unfortunately I cannot, and there is no need, as it is essentially the same as the one from gcc, which I listed in the description above.

I cannot reproduce the error message you have provided at all using the same toolchain or with the Zephyr SDK toolchain.

> I think you are getting build warnings in you above example, so if you put -Werror, then you will probably see the same. I will look at your comment in the PR and rework asap, so that we can get the cast removed.

There is not a single warning associated with `qsort` actually. There are warnings from `uart.h` and elsewhere. But no, adding `-Werror` does not cause additional warnings in `qsort.h`.

In any case, I've made some additional comments on your PR, which should help to solve your issue.

cfriedt, 2022-03-02:
Since this issue cannot be reproduced and is only reportedly failing using an undisclosed & unsupported proprietary toolchain, I'm changing this to an enhancement rather than a bug.

daor-oti, 2022-03-03:
@cfriedt Thank you for taking the time to see if you could reproduce the error, and I am not sure why you are not able to. But as the PR will resolve our issue, and perhaps even others, I hope that we can have it merged as soon as possible.

PR #42384 (2022-03-09): libc: minimal: qsort remove callback cast in call of qsort_r
This removes the cast of the two parameter compare function used by qsort, to
the three parameter callback function used by qsort_r, in order to
ensure compatibility with other toolchains, even those off-tree.

Fixes #42870
commit 845a200c (2022-03-09): libc: minimal: qsort remove callback cast in call of qsort_r

Remove the cast of the two parameter compare function used by qsort, to
the three parameter callback function used by qsort_r, in order to
ensure compatibility with other toolchains, even those off-tree.

Fixes #42870
closed 2022-03-09 by cfriedt