zephyrproject-rtos/zephyr #42877: [POSIX] k_cycle_get_32 returns 0 on start-up

asemjonovs, 2022-02-16:
**Describe the bug**
k_cycle_get_32 always returns 0 on start-up.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? POSIX
 - What have you tried to diagnose or workaround this issue? Ran in gdb and observed hw_models_top.c `simu_time==0`

**To Reproduce**
Steps to reproduce the behavior:
1. Enable CONFIG_ZTEST_DO_THE_SHUFFLE
2. ./scripts/twister -v -c -T tests/ztest/base
3. Observe log "Let's shuffle! seeding: 0"

**Expected behavior**
I'd expect a different non-zero value returned each time on start-up

**Impact**
It prevents a random shuffling of test cases being executed.  This is needed to identify any dependencies between test cases.

**Logs and console output**
```
*** Booting Zephyr OS build v3.0.0-rc1-43-g479bf856e78a  ***
Let's shuffle! seeding: 0
Running test suite rules_tests
===================================================================
START - test_rules_before_after
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/pull/42330

**Additional context**
none

yperess, 2022-02-22:
Added the medium priority label as this affects tests.

asemjonovs, 2022-02-22:
@aescolar - would it make sense to add a kconfig option that either provides deterministic or non-deterministic results?  This way we can satisfy both needs.

aescolar, 2022-02-23:
>@aescolar - would it make sense to add a kconfig option that either provides deterministic or non-deterministic results? This way we can satisfy both needs.

@asemjonovs I don't think it would be a too good idea to do that. Meaning, you would not get what you are hoping for.
In a regression system you do not want uncontrolled randomness. Say you do have it, and your test fails just 1 out of 1000 times, what can you do? how do you debug the failing instance when re-triggering the failure takes you 1000 trials?
The idea is that you can have pseudo-randomness, but it must be well controller (thru a random seed, which you would print out in the test log, so when things fail, you can rerun just that test exactly as when it failed). That is why you can enable pseudo-random behavior but you must provide the seed.

In any case, in this particular issue you seem to be after random boot timer counts. But many real embedded platforms will provide you very/fully deterministic boot times. So that is not a source of randomness you should use for any purpose anyhow. (And even if we were to add some more "randomness" to the native_posix target, we would not sprinkle that randomness around in the code)

Note the aim of native_posix: https://docs.zephyrproject.org/2.6.0/boards/posix/native_posix/doc/index.html#rationale-for-this-port

carlescufi, 2022-03-08:
Closing this due to inactivity, feel free to reopen @asemjonovs if you have additional questions.

closed 2022-03-08 by carlescufi