zephyrproject-rtos/zephyr #42881: Arduino due missing 'arduino_i2c' alias.

zerkerX, 2022-02-16:
Hello folks. I have recently been trying out the Zephyr project with my Arduino Due and ran into some issues trying to run the SSD1306 128x64 shield with some examples. It appears the Arduino Due is missing the `arduino_i2c` alias needed to interface with the `ssd1306_128x64` shield.

I've attempted to add the following to the bottom of `arduino_due.dts`:

```
arduino_i2c: &twi1 { };
```

This does allow the example to build and flash, but the result does not function. There is no i2C data sent on the bus when I hooked up my scope. I have tested my wiring with the Arduino IDE's example, and double-checked from the schematic that the pins are indeed `twi1`.

Obviously, something else needs to be done for this to operate correctly. I will probably spend some time digging into the shield and i2c code, but any advice would be appreciated!

zerkerX, 2022-02-17:
Well. I guess I figured it out. The shield had the wrong default I2C address compared to my specific board. I would have thought *something* would have still appeared on the bus for my scope to see, but I guess not.

Anyways, I'll be submitting a really small pull request to add the alias shortly, since I know it works now :smile:.

As an aside, is there any good way to deal with I2C address mismatch when running examples besides just hacking the shield's overlay file? If it was my project, I could update in my own overlay, but I'm drawing a blank for the samples.

zerkerX, 2022-02-18:
In my case, it's the silkscreen on the board which has `0x3D` printed on it. I am specifically using the [Adafruit version of the board](https://www.adafruit.com/product/326). It's a pretty popular generic display, so I imagine there are a couple versions of it that might be configured for different I2C addresses.

jfischer-no, 2022-02-18:
> I am specifically using the [Adafruit version of the board](https://www.adafruit.com/product/326)

"This board/chip uses I2C 7-bit address between 0x3C-0x3D, selectable with jumpers"

jfischer-no, 2022-02-18:
> Anyways, if we want to do anything here to make it easier for 0x3d use, we should probably split this out into its own issue.

Documentation for generic SSD1306 shield can be updated to mention that the default address is `0x3c`.

nandojve, 2022-02-18:
> Documentation for generic SSD1306 shield can be updated to mention that the default address is 0x3c.

+1

I was wondering if could be valid add an specific overlay like adafruit_ssd1306_128x64.overlay which uses 0x3D address with proper documentation. This could complement @jfischer-no suggestion.

zerkerX, 2022-02-19:
I raised #42992 to continue the discussion specific to the SSD1306 shield.

PR #42948 (2022-03-08): boards: arduino_due: Add arduino_i2c alias
The arduino_i2c alias is needed to use the Arduino Due with some of the
examples. This is assigned to the twi1 interface, which is labelled as
SDA and SCL (20 and 21) on the board silkscreen and is the default
interface using the Arduino IDE.

Resolves #42881.
commit 5ce17a96 (2022-03-08): boards: arduino_due: Add arduino_i2c alias

The arduino_i2c alias is needed to use the Arduino Due with some of the
examples. This is assigned to the twi1 interface, which is labelled as
SDA and SCL (20 and 21) on the board silkscreen and is the default
interface using the Arduino IDE.

Resolves #42881.
closed 2022-03-08 by MaureenHelm