zephyrproject-rtos/zephyr #42904: Bluetooth: Controller: CIG: LL Reject: Local initiated being rejected

cvinayak, 2022-02-17:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

mtpr-ot, 2022-11-21:
Supposedly implemented in #52147. Will need testing to verify correct behavior.

cvinayak, 2023-03-16:
Manually tested using central_iso sample to try create second CIS  to peripheral_iso sample which rejects it in `iso_accept` callback. The behavior is as expected.

closed 2023-03-17 by cvinayak