zephyrproject-rtos/zephyr #42961: periodic_sync sample never executes .recv callback

GatCode, 2022-02-18:
I've tried to execute the periodic_adv and periodic_sync samples on 5340dk's and 52840dk's but the periodic_sync board never reaches the .recv callback altough the sync gets established.

For testing purposes, I've already stripped parts of the original samples to make it more clear. Basically line 50 (printk("Hello From recv_cb\n");) on the receiver side never gets executed. Here you can find the simple example: [https://gist.github.com/GatCode/ab1513545154fb53e1dd390a5ac2d30c ](https://gist.github.com/GatCode/ab1513545154fb53e1dd390a5ac2d30c)

For the 52s, I used the correct config from the boards folder and with the 53s, I enabled the correct settings on the netcore and I'm using [ncs v1.9.0](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/releases/release-notes-1.9.0.html#zephyr) - this version is based on Zephyr 45ef0d2.

carlescufi, 2022-02-18:
@GatCode could you please run your simple example with the latest upstream Zephyr (v3.0.0-rc3 right now) and see if the issue is there as well?

ppryga, 2022-02-18:
I've run periodic_scync sample on nRF5340DK and periodic_adv on nRF52833 and below is console log.
It synchronizes with advertiser.
I've used code from https://github.com/zephyrproject-rtos/zephyr/commit/45ef0d2aef9bbba92460e19129b35f0da9729610

```
*** Booting Zephyr OS build v3.0.0-rc1  ***
Starting Periodic Advertising Synchronization Demo
Checking LED device...done.
Configuring GPIO pin...done.
Scan callbacks register...success.
Periodic Advertising callbacks register...Success.
Start scanning...success.
Start blinking LED...
Waiting for periodic advertising...
[DEVICE]: 07:E8:86:CF:8A:88 (random), AD evt type 3, Tx Pwr: 127, RSSI -41  C:0 S:0 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 ms)TPwr: 127, RSSI -88  C:0 S:0 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 ms), SID: 255
[DEVICE]: 4F:EB:D1:3E:CB:3D (random), AD evt type 3, Tx Pwr: 127, RSSI -82  C:0 S:0 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 ms), S type 5, Tx Pwr: 127, RSSI -69 Test Periodic Advertising C:0 S:0 D:0 SR:0 E:1 Prim: LE 1M, Secn: LE 2M, Interval: 0x03c0 (1200 ms), SID: 0
Fouodic Advertising Sync...success.
Waiting for periodic sync...
[DEVICE]: 76:27:9B:B3:A4:89 (random), AD evt type 0, Tx Pwr: 127, RSSI -96  C:1 S:1 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 ([DEVICE]: 1C:31:83:BC:4C:9A (random), AD evt type 3, Tx Pwr: 127, RSSI -99  C:0 S:0 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 [DEVICE]: BC:14:85:D0:EF:EF (public), AD evt type 3, Tx Pwr: 127, RSSI -76  C:0 S:0 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (random) synced, Interval 0x03c0 (1200 ms), PHY LE 2M
PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (Periodic sync established.
Stop blinking LED.
Waiting for periodic sync lost...
[DEVICE]: B8:8E:82:3F:FA:E6 (public), AD evt type 0, Tx Pwr: 127, RSSI -97  C:1 S:1 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 msevt type 4, Tx Pwr: 127, RSSI -97 HUAWEI Band 6-AE6 C:1 S:1 D:0 SR:1 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 ms), SID: 255
[DEVICE]: 6D:53:B1:D7:03:15 (random), AD evt type 0, Tx Pwr: 127, RSSI -101  C:1 S:1 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 mandom), AD evt type 0, Tx Pwr: 127, RSSI -98  C:1 S:1 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 ms), SID: 255
[DEVICE]: E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 ms), SID: 255
PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (random), tx_power 127, RSSI -58, CTE 0, data length 5, data: 04ffffff08
PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (random), tx_power 127, RSSI -60, CTE 0, data length 5, data: 04ffffff08
PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (random), tx_power 127, RSSI -54, CTE 0, data length 5, data: 04ffffff08
PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (random), tx_power 127, RSSI -56, CTE 0, data length 5, data: 04ffffff08
PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (random), tx_power 127, RSSI -58, CTE 0, data length 5, data: 04ffffff08
[DEVICE]: 40:16:3B:08:E7:AB (public), AD evt type 0, Tx Pwr: 127, RSSI -102  C:1 S:1 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 ms)PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (random), tx_power 127, RSSI -53, CTE 0, data length 5, data: 04ffffff09
PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (random), tx_power 127, RSSI -57, CTE 0, data length 5, data: 04ffffff09
[DEVICE]: 04:21:44:BD:14:F9 (public), AD evt type 0, Tx Pwr: 127, RSSI -90 JBL Charge 4 C:1 S:1 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 ms), SID: 255
[DEVICE]: 04:21:44:BD:14:F9 (public), AD evt type 4, Tx Pwr: 127, RSSI -90  C:1 S:1 D:0 SR:1 E:0 Prim: LE 1M, Secl: 0x0000 (0 ms), SID: 255
PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (random), tx_power 127, RSSI -59, CTE 0, data length 5, data: 04ffffff09
PER_ADV_SYNC[0]: [DEVICE]: 3C:5F:27:41:F4:A0 (random), tx_power 127, RSSI -64, CTE 0, data length 5, data: 04ffffff09
[DEVICE]: 3C:BD:3E:56:7E:CD (public), AD evt type 0, Tx Pwr: 127, RSSI -100  C:1 S:1 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (06:7E:CD (public), AD evt type 4, Tx Pwr: 127, RSSI -99  C:1 S:1 D:0 SR:1 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 ms), SID: 255
```

GatCode, 2022-02-18:
[@carlescufi](https://github.com/carlescufi) can you please tell me how I can checkout this revision inside the ncs v1.9.0? I thought this is easily possible with a west update ... command but this seems not to be the case?

ppryga, 2022-02-18:
@GatCode go to Zephyr repository and call `git checkout v3.0.0-rc2`.

GatCode, 2022-02-19:
Thanks to both of you! Yes the issue is resolved in v3.0.0-rc3 and for all the future readers who struggle like me - these links might be helpful:

- https://devzone.nordicsemi.com/f/nordic-q-a/76244/nrfconnect-sdk-nrf-modules-trusted-firmware-m-kconfig-and-kconfig-tfm-nordic-sdk-not-compatible-with-zephyr-upstream/
- https://github.com/maksimdrachov/zephyr-rtos-tutorial

closed 2022-02-19 by GatCode