zephyrproject-rtos/zephyr #42973: Zephyr-sdkConfig.cmake file not found

pcotret, 2022-02-18:
**Describe the bug**
Followed instructions of the getting started guide available at: https://docs.zephyrproject.org/latest/getting_started/index.html
Got an error while compiling the Hello world example

**To Reproduce**
Steps to reproduce the behavior:
```bash
$ west build -p auto -b qemu_riscv32 samples/hello_world  
-- Zephyr version: 3.0.0-rc3 (/home/pascal/Documents/ensta/work/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: qemu_riscv32
-- Cache files will be written to: /home/pascal/.cache/zephyr
CMake Error at /home/pascal/zephyr/cmake/verify-toolchain.cmake:70 (find_package):
  Could not find a package configuration file provided by "Zephyr-sdk"
  (requested version 0.13.1) with any of the following names:

    Zephyr-sdkConfig.cmake
    zephyr-sdk-config.cmake

  Add the installation prefix of "Zephyr-sdk" to CMAKE_PREFIX_PATH or set
  "Zephyr-sdk_DIR" to a directory containing one of the above files.  If
  "Zephyr-sdk" provides a separate development package or SDK, be sure it has
  been installed.
```

pcotret, 2022-02-18:
Was working on an older Zephyr release. Issue closed.

closed 2022-02-18 by pcotret

joelk18, 2023-04-04:
I have the same problem, please someone could help me ?