zephyrproject-rtos/zephyr #42988: Specify and standardize undefined behavior on empty response from server for http_client

glarsennordic, 2022-02-18:
**Is your enhancement proposal related to a problem? Please describe.**
[The documentation for `http_client`](https://docs.zephyrproject.org/apidoc/latest/group__http__client.html) currently does not specify what the intended behavior is if the contacted server gives an empty response when calling `http_client_req`.

By this, I mean a graceful closure of the underlying TCP socket by remote, with zero-length response. In other words, a FIN-ACK frame with no payload. This is distinct from, for instance, a valid HTTP response that merely has an empty body.

To be clear, a truly blank response is not a valid HTTP response, AFAIK. This is technically an error condition.

Currently, `http_client_req` will call the response callback with all attributes left in their zeroed-out state. This behavior is not documented at all, and it is not clear if it is the intended behavior. This makes it difficult for developers who may need to handle this type of response gracefully to know exactly how to detect it reliably and without fear that future changes to the library could invalidate their solution.

**Describe the solution you'd like**
I would like to discuss what the ideal behavior in this scenario should be, and work toward implementing that behavior and updating documentation so that said behavior is made explicit. In other words.

In other words, I would like the intended behavior of `http_client_req` upon empty response to no longer be undefined/undocumented.

**Possible Specific Solutions**
I see two distinct practices we could adopt. We could either raise an error (return negative error code), or specify a unique state for the response struct that represents this behavior.

Currently, the latter of these two approaches is (possibly completely by accident) implemented. When an empty response is received from remote, the response callback is called with `final_data == HTTP_DATA_FINAL`, and the passed response information struct has both `http_status_code == 0` and `message_complete == 0`. This would allow an end user to detect an empty response by checking either `http_status_code ==0` or  `message_complete == 0` when the callback is called for the last time.

However, at the moment, it is dangerous to rely on this behavior, due to the fact that this scenario is not documented, and thus the behavior is not guaranteed. I would like for this specific scenario to be documented, and for an official behavior to be adopted.

Alternatively, we should consider adding another struct flag (perhaps `message_empty`) or a special `http_status_code` value that is properly defined in the header files. Anything, really, as long as it allows this scenario to be uniquely identified.

I am in favor of creating a well-defined response struct state, rather than simply returning an error, since technically the request was successfully sent (no errors with the socket, it was a graceful close), and technically an empty, invalid response is still a response.

Once a solution has been decided, I am available to do the work updating documentation or implementing it if need be.

jukkar, 2022-02-22:
> Currently, `http_client_req` will call the response callback with all attributes left in their zeroed-out state. This behavior is not documented at all, and it is not clear if it is the intended behavior.

I can confirm that this case was not intended but was just missed during the implementation and testing.

> Currently, the latter of these two approaches is (possibly completely by accident) implemented. When an empty response is received from remote, the response callback is called with `final_data == [...]

IMHO this sounds like a reasonable solution for this.

> Once a solution has been decided, I am available to do the work updating documentation or implementing it if need be.

Thanks for volunteering :+1:

rlubos, 2022-02-22:
> Alternatively, we should consider adding another struct flag (perhaps message_empty) or a special http_status_code value that is properly defined in the header files. Anything, really, as long as it allows this scenario to be uniquely identified.

Perhaps, we could convert the `http_status_code` member to be a signed value, and allow to write a negative `errno` value in case of errors? It should be pretty straightforward to document that a negative status value indicate that no valid response has been received, for any reason (`ECONNRESET` for instance in case peer closes the connection before sending reply).

jukkar, 2022-02-22:
> Perhaps, we could convert the `http_status_code` member to be a signed value, and allow to write a negative `errno` value in case of errors? It should be pretty straightforward to document that a [...]

This sounds like a good alternative too. Then we would have only one variable to check in the application.

glarsennordic, 2022-02-22:
I think either could work!

My one concern with allowing errno values as http_status values is the potential for confusion for an end user -- Since `http_client_req` can also return errnos, the two different sources of errno could have different meanings for the same return code.

For instance, `ECONNRESET` returned directly by http_client_req indicates that the request flat out failed due to some kind of RST signal received on the underlying socket, whereas, hypothetically, the same code in the HTTP status field would indicate a graceful closure with no response. 

Neither of those is technically normal HTTP server behavior, but it may be important for an end user to distinguish between them, and having that kind of error code overlap could potentially be confusing.

In this particular case, `http_status = 0` is fairly elegant (since 0 is not a valid HTTP status code), and IMO evocative of the concept of an empty response.

Honestly, though, I'd rather the final decision be in both of your hands. If you think errno in status is a better solution, I'm down to do the work to implement.

Are there any other potential reasons an errno could end up on the status field?

rlubos, 2022-02-23:
> Are there any other potential reasons an errno could end up on the status field?

Generally we could forward any error code returned at the socket level.

> In this particular case, http_status = 0 is IMO fairly elegant (since 0 is not a valid HTTP status code), and evocative of the concept of an empty response.

No strong feelings towards the errno approach, we can stick to `http_status = 0` approach if you think it's cleaner from the application POV.

jukkar, 2022-02-23:
I am fine with either solution.

glarsennordic, 2022-02-25:
Sorry for going missing for the past couple of days.

If you are leaving it up to me, I think I would go with `http_status = 0` as the solution, which I think would only require doc changes and maybe some notes thrown into the code.

I'll put in a PR with the changes some time next week or the week after most likely.

PR #43453 (2022-03-07): net: http_client: Add official null response behavior
Adds an official behavior in response to null response from HTTP
endpoint.

Fixes #42988
commit 303c42e7 (2022-03-07): net: http_client: Add official null response behavior

Adds an official behavior in response to null response from HTTP
endpoint.

Fixes #42988
closed 2022-03-07 by MaureenHelm