zephyrproject-rtos/zephyr #42997: periodic advertisement with larger ad broken from v3.0.0-rc2 onwards

GatCode, 2022-02-19:
Hi everyone,
I've discovered that the periodic advertisement implementation is broken on v3.0.0-rc2 and rc3 - in more detail it breaks as soon as you set the periodic advertisement data to more than 30 bytes on the sender side. As you can see in [this video](https://youtu.be/-3Fxn8-4SgI), I basically executed the samples (periodic_adv and periodic_sync) on two nrf52840dk's and only changed the advertisement data size and commented out some print statements on the receiver side. If you now look at the received data (right side on the video), you can see that in rc1, everything looks good and comes in at a certain pace but as soon as I flash the rc2, the periodic advertisements sometimes come in and sometimes not. Also in rc2, the periodic advertisements stop completely after some time.

carlescufi, 2022-02-19:
@GatCode thanks for the report. Could you try bisecting to find the exact commit that introduced the regression please?

GatCode, 2022-02-22:
@carlescufi, sure - the last working commit was 93d6d1e1142434925d490f80794d2798ee45ed49 - therefore faa89c779cbee1b35a4c9323effa82232451ace6 introduced the bug

cvinayak, 2022-02-23:
@GatCode Please set the correct value for  `CONFIG_BT_CTLR_SCAN_DATA_LEN_MAX` in your application if you want longer than the Bluetooth minimum AD data size of 31 bytes to be received.