zephyrproject-rtos/zephyr #42998: Should board.dts enable peripherals by default?

yonsch, 2022-02-19:
As discussed in #42513

Many board.dts files in the Zephyr tree enable peripherals by default. For example, [nucleo_f091rc.dts](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nucleo_f091rc/nucleo_f091rc.dts) which defines:
```
&i2c1 {
	pinctrl-0 = <&i2c1_scl_pb8 &i2c1_sda_pb9>;
	pinctrl-names = "default";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
};

&i2c2 {
	pinctrl-0 = <&i2c2_scl_pa11 &i2c2_sda_pa12>;
	pinctrl-names = "default";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
};
```
This does not immediately enable those peripherals, since they depend on `CONFIG_I2C=y`. The problem with this approach is that enabling `CONFIG_I2C` will enable **BOTH** peripherals. This is problematic for two reasons I can think of:
1. Enabling an extra peripheral wastes power
2. Enabling an extra peripheral can cause pin collisions with other peripherals (e.g. the default i2c pin happens to also be the uart console tx pin).

Ideally, all peripherals should be disabled by default, and enabled by a project's DTS overlay. Realistically, doing so at this point would probably break some stuff, including samples and tests enabling peripherals with Kconfig without having an overlay (thanks @gmarull for pointing this out).

Given that it isn't possible to disable peripherals in DTS, other solutions to alleviate this problem may be:
1. The pinctrl driver can assert and warn/crash if multiple peripherals are configured to use the same pin.
2. The build system can print or log to a file the enabled peripherals. When optimizing for power consumption, this info can be used to contrive a board overlay that filters all unused peripherals.
3. (not sure this is a good idea but worth mentioning) An assertion to check if a peripheral (device) is enabled but is never accessed through `device_get_binding` or similar.

Any better ideas are of course welcomed.

jfischer-no, 2022-02-22:
Generally board's DTS describes the available hardware configuration of the board. If the devicetree description of your nucleo board is not correct then open a PR to fix it. In general, I see no justification for this issue.

erwango, 2022-02-22:
> Ideally, all peripherals should be disabled by default, and enabled by a project's DTS overlay.

So, for any sample or test, we'd have an overlay for each applicable board ?
I understand the limitations of current way of doing, but I'm not convinced this solution would scale.

henrikbrixandersen, 2022-11-02:
I cannot see how this would be more correct nor how it would scale to the amount of boards, we support in-tree. I suggest closing.

erwango, 2022-11-07:
> I cannot see how this would be more correct nor how it would scale to the amount of boards, we support in-tree. I suggest closing.

Fully agree. Closing

closed 2022-11-07 by erwango