zephyrproject-rtos/zephyr #43016: Self inc/dec works incorrectly with logging API.

smrtos, 2022-02-21:
**Describe the bug**
When using logging API with self inc/dec operation, the increased/decreased value is incorrect.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the code from here: https://github.com/smrtos/zephyr_rtos_bug_repro/tree/main/SelfIncInLoggingApi
2. Build it on qemu_x86_64.
3. Run the app in qemu.

**Expected behavior**
100 logs should be printed. Like this:

```
[00:00:00.049,635] <inf> main: hello, main()

[00:00:00.052,334] <inf> main: hello, logger 0

[00:00:00.053,610] <inf> main: hello, logger 1

[00:00:00.054,293] <inf> main: hello, logger 2

....

```

**Actual behavior**

```
[00:00:00.049,154] <inf> main: hello, main()

[00:00:00.051,420] <inf> main: hello, logger 2 // <============= the value is increased too much

[00:00:00.052,509] <inf> main: hello, logger 5

[00:00:00.053,200] <inf> main: hello, logger 8

[00:00:00.053,872] <inf> main: hello, logger 11

...
```

**Impact**
Customer cannot use self inc/dec in logging API.

**Investigation**
I checked the preprocessed file and debugged through it. It  showed that the macro expansion in logging infrastructure introduced some extra/unwanted execution of the self inc/dec statement.

stephanosio, 2022-02-21:
```
// bad one
LOG_INF("hello, logger %d\n", ii++);
```

Generally, placing increment and decrement operators within a complex expression is a bad practice and is even prohibited by various coding standards, including the Zephyr Coding Guidelines and MISRA C (Rule 13.3).

Unless it is technically feasible to eliminate multiple evaluation of the parameter expressions in the logging subsystem, this should be marked "won't fix" and "known issue."

We could potentially address this by generating errors in the CI when such malformed expressions are used in the code.

nordic-krch, 2022-02-21:
It is fixed by #42686

PR #42686 (2022-02-22): logging: Prevent multiple arguments evaluation (take 2)
This is another attempt to fix #42050. First attempt (#42165) was reverted (#42609) due to #42608. Test was failing because of increased stack usage after changes in logging. Increased stack usage has been seen in `CONFIG_NO_OPTIMIZATIONS=y` case.

In this take, logging macro path has been changed so that runtime packaging (the one without preprocessor magic and multiple arguments evaluation) is used always when `CONFIG_NO_OPTIMIZATIONS=y` or `CONFIG_LOG_MODE_IMMEDIATE=y`. In both cases static packaging is not needed since static packagin [... cut]
closed 2022-02-22 by carlescufi