zephyrproject-rtos/zephyr #43020: samples/subsys/fs/littlefs does not work with native_posix board

ivan2angelov, 2022-02-21:
samples/subsys/fs/littlefs is not working with native_posix board.
file system mounting fails with error -19.

branch main at 

Build environment: WSL2 with Ubuntu 20.04

**To Reproduce**
west build -p auto samples/subsys/fs/littlefs -b native_posix

that build produces zephyr.elf.
Running zephyr.elf in WSL2 produces following output:

 ./build/zephyr/zephyr.elf
Failed to get status of flash device file flash.bin: Value too large for defined data type
*** Booting Zephyr OS build v3.0.0-rc3-48-gd353e5516079  ***
Area 4 at 0xfc000 on flash_ctrl for 16384 bytes
[00:00:00.000,000] <inf> littlefs: LittleFS version 2.4, disk version 2.0
[00:00:00.000,000] <err> littlefs: can't get flash device: flash_ctrl
[00:00:00.000,000] <err> fs: fs mount error (-19)
FAIL: mount id 4 at /lfs: -19

**Expected behavior**
I expected application example to run to completion without any errors (mounting fs, open dir, close dir, unmount fs).

**Impact**
Not sure what is the correct way to configure and use littlefs on native_posix.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) - Win 10 + WSL2
 - Toolchain (e.g Zephyr SDK, ...) zephyr SDK
 - Commit SHA or Version used: d353e551607926612ff258dadbc13d028d58b54e

**Additional context**
samples/subsys/fs/fat_fs/ does not work with native_posix - it produces different error code on mounting though.

Laczen, 2022-02-21:
@ivan2angelov, I just tested this and cannot reproduce it. Could you delete the build folder and try again ?

ivan2angelov, 2022-02-22:
Sure. I did that.
I am at commit d353e551607926612ff258dadbc13d028d58b54e

git status reports no "dirty" files.

Here is build output:
`
zephyrproject/zephyr (main) $ west build -p always -b native_posix samples/subsys/fs/littlefs
-- west build: generating a build system
Including boilerplate (Zephyr base): /mnt/c/work/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /mnt/c/work/zephyrproject/zephyr/samples/subsys/fs/littlefs
-- Zephyr version: 3.0.0-rc3 (/mnt/c/work/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: native_posix
-- Cache files will be written to: /home/iangelov/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /mnt/c/work/zephyrproject/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /mnt/c/work/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /mnt/c/work/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /mnt/c/work/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /mnt/c/work/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /mnt/c/work/zephyrproject/zephyr/samples/subsys/fs/littlefs/Kconfig
Loaded configuration '/mnt/c/work/zephyrproject/zephyr/boards/posix/native_posix/native_posix_defconfig'
Merged configuration '/mnt/c/work/zephyrproject/zephyr/samples/subsys/fs/littlefs/prj.conf'
Configuration saved to '/mnt/c/work/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/mnt/c/work/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/work/zephyrproject/zephyr/build
-- west build: building application
[1/118] Preparing syscall dependency handling

[2/118] Generating include/generated/version.h
-- Zephyr version: 3.0.0-rc3 (/mnt/c/work/zephyrproject/zephyr), build: v3.0.0-rc3-48-gd353e5516079
[114/118] Linking C executable zephyr/zephyr_pre0.elf

[118/118] Linking C executable zephyr/zephyr.elf
`

And here is the output of executing the binary:
` 
./build/zephyr/zephyr.elf
Failed to get status of flash device file flash.bin: Value too large for defined data type
*** Booting Zephyr OS build v3.0.0-rc3-48-gd353e5516079  ***
Area 4 at 0xfc000 on flash_ctrl for 16384 bytes
[00:00:00.000,000] <inf> littlefs: LittleFS version 2.4, disk version 2.0
[00:00:00.000,000] <err> littlefs: can't get flash device: flash_ctrl
[00:00:00.000,000] <err> fs: fs mount error (-19)
FAIL: mount id 4 at /lfs: -19
`

Laczen, 2022-02-22:
@ivan2angelov, it seems like wsl2 has a problem with the created flash.bin file to support flash simulation on native-posix. Have you changed something in the native-posix dts file? Normally a 2MB file (2048kB) is created for the flash emulation, but wsl is reporting that this file is to large.

ivan2angelov, 2022-02-22:
@Laczen thank you for the hints.
this is what I have:
zephyrproject/zephyr (main) $ find . -name flash.bin
./flash.bin
####@### /mnt/c/work/zephyrproject/zephyr (main) $ ll ./flash.bin
-rw------- 1 ##### ####  0 Feb 22 09:38 ./flash.bin

it seems like flash.bin is created in zephyr folder and it has zero bytes size.

Laczen, 2022-02-22:
@ivan2angelov, I think this is because you are trying to create the flash.bin file on `/mnt/c` and it might be that this doesn't work. Could you try the same thing in your wsl2 home folder. I don't have this problem but I am using zephyr on wsl2 and installed zephyr on the ubuntu disk. There might be a wsl2 problem with calling `stat` on windows folders.

ivan2angelov, 2022-02-22:
@Laczen Great thank you for the tip! 

Indeed, it was "user error". 

I created flash.bin under my linux home directory and pass that to the example with -flash=<path-to-linux-folder/flash.bin>.
After that the example "formatted" flash.bin and second run mounted/unmounted without problems:

./build/zephyr/zephyr.elf -flash=/home/######/zephyr_flash/flash.bin
*** Booting Zephyr OS build v3.0.0-rc3-48-gd353e5516079  ***
Area 4 at 0xfc000 on flash_ctrl for 16384 bytes
[00:00:00.000,000] <inf> littlefs: LittleFS version 2.4, disk version 2.0
[00:00:00.000,000] <inf> littlefs: FS at flash_ctrl:0xfc000 is 4 0x1000-byte blocks with 512 cycle
[00:00:00.000,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.000,000] <inf> littlefs: /lfs mounted
/lfs mount: 0
/lfs: bsize = 16 ; frsize = 4096 ; blocks = 4 ; bfree = 2
/lfs/boot_count stat: 0
        fn 'boot_count' size 4
/lfs/boot_count read count 1: 4
/lfs/boot_count seek start: 0
/lfs/boot_count write new boot count 2: 4
/lfs/boot_count close: 0
/lfs opendir: 0
  F 4 boot_count
End of files
[00:00:00.000,000] <inf> littlefs: /lfs unmounted
/lfs unmount: 0

closed 2022-02-22 by ivan2angelov